pyxem/pyxem #539: indexer.correlate() not working when loading .cif file.

EirikOpheim, 2020-02-03:
**Describe the bug**
When using the following .cif file:  
[9016521.txt](https://github.com/pyxem/pyxem/files/4149044/9016521.txt) (converted to .txt format to allow upload to pyxem). I am allowed to load the file and generate simulated data and a template library, but when I use indexer.correlate() to match simulated data with the template library I get an error message (Will be posted below).

**To Reproduce**
I have written a MWE, but it is a bit extensive

import numpy as np
import math
import hyperspy.api as hs
import pyxem as pxm
import diffpy.structure
from matplotlib import pyplot as plt

#Pyxem sub-libraries
from diffsims.libraries.structure_library import StructureLibrary
from diffsims.generators.diffraction_generator import DiffractionGenerator
from diffsims.generators.library_generator import DiffractionLibraryGenerator
from pyxem.generators.indexation_generator import IndexationGenerator
from pyxem.utils.sim_utils import sim_as_signal

# Load structure
Structure = diffpy.structure.loadStructure("./Cif/9016521.cif")

# Set simulation Parameters
pattern_size = 256  # pixels
half_pattern_size = pattern_size // 2
reciprocal_radius = 4 #How far out in reciprocal space we look for reflections
calibration = reciprocal_radius / half_pattern_size #Scaling of diffraction pattern
beam_energy = 300.0 #keV
ediff = DiffractionGenerator(beam_energy, 0.025)  # keV and relrod length (1/Å)

# Simulate diffraction pattern
phase_name = ['Monoclinic'] 
Angles = [] 
discreteAngleValues = np.linspace(-60.,60.,25)
lengthOfAnglesVek = len(discreteAngleValues)
for i in range (lengthOfAnglesVek):
    for j in range (lengthOfAnglesVek):
        Angles.append((float(discreteAngleValues[j]),float(discreteAngleValues[i]),float(0)))

sample_lib = StructureLibrary(phase_name, [Structure], [Angles])

diff_gen = DiffractionLibraryGenerator(ediff)
library = diff_gen.get_diffraction_library(sample_lib,
                                           calibration=calibration,
                                           reciprocal_radius=reciprocal_radius,
                                           half_shape=(half_pattern_size, half_pattern_size),
                                           with_direct_beam=False)

data_au = []

for angles in Angles:
    
pattern = sim_as_signal(library.get_library_entry(phase='Monoclinic', angle=(angles))['Sim'],
                            pattern_size, 0.05, reciprocal_radius)
    data_au.append(pattern)
        
data = [x.data for x in data_au] 

test_data = pxm.ElectronDiffraction2D(np.asarray(data).reshape(lengthOfAnglesVek,lengthOfAnglesVek, pattern_size, pattern_size))
test_data.set_diffraction_calibration(calibration) 

# Create template library (Syntax of updated diffsims)
structure_library = StructureLibrary.from_orientation_lists(['Monoclinic'],[Structure],[Angles])
diff_gen = DiffractionLibraryGenerator(ediff)
template_library = diff_gen.get_diffraction_library(structure_library,
                                                    calibration=calibration,
                                                    reciprocal_radius=reciprocal_radius,
                                                    half_shape=(half_pattern_size, half_pattern_size),
                                                    with_direct_beam=False)

# Run indexer.correlate 
indexer = IndexationGenerator(test_data, template_library)
match_results = indexer.correlate(n_largest=3)

--> Error message:
IndexError                                Traceback (most recent call last)
<ipython-input-8-4a874069a0b3> in <module>
      1 indexer = IndexationGenerator(test_data, template_library)
----> 2 match_results = indexer.correlate(n_largest=3)

~/anaconda3/envs/pyxem/lib/python3.7/site-packages/pyxem/generators/indexation_generator.py in correlate(self, n_largest, mask, *args, **kwargs)
    111                                  mask=mask,
    112                                  inplace=False,
--> 113                                  **kwargs)
    114 
    115         matching_results = TemplateMatchingResults(matches)

~/anaconda3/envs/pyxem/lib/python3.7/site-packages/hyperspy/signal.py in map(self, function, show_progressbar, parallel, inplace, ragged, **kwargs)
   3939                                     parallel=parallel, inplace=inplace,
   3940                                     ragged=ragged,
-> 3941                                     **kwargs)
   3942         if inplace:
   3943             self.events.data_changed.trigger(obj=self)

~/anaconda3/envs/pyxem/lib/python3.7/site-packages/hyperspy/signal.py in _map_iterate(self, function, iterating_kwargs, show_progressbar, parallel, ragged, inplace, **kwargs)
   4057                            show_progressbar)
   4058         for ind, res in zip(range(res_data.size),
-> 4059                             thismap(func, zip(*iterators))):
   4060             # In what follows we assume that res is a numpy scalar or array
   4061             # The following line guarantees that that's the case.

~/anaconda3/envs/pyxem/lib/python3.7/concurrent/futures/_base.py in result_iterator()
    584                     # Careful not to keep a reference to the popped future
    585                     if timeout is None:
--> 586                         yield fs.pop().result()
    587                     else:
    588                         yield fs.pop().result(end_time - time.monotonic())

~/anaconda3/envs/pyxem/lib/python3.7/concurrent/futures/_base.py in result(self, timeout)
    423                 raise CancelledError()
    424             elif self._state == FINISHED:
--> 425                 return self.__get_result()
    426 
    427             self._condition.wait(timeout)

~/anaconda3/envs/pyxem/lib/python3.7/concurrent/futures/_base.py in __get_result(self)
    382     def __get_result(self):
    383         if self._exception:
--> 384             raise self._exception
    385         else:
    386             return self._result

~/anaconda3/envs/pyxem/lib/python3.7/concurrent/futures/thread.py in run(self)
     55 
     56         try:
---> 57             result = self.fn(*self.args, **self.kwargs)
     58         except BaseException as exc:
     59             self.future.set_exception(exc)

~/anaconda3/envs/pyxem/lib/python3.7/site-packages/hyperspy/misc/utils.py in func(*args)
    972     def func(*args):
    973         dat, these_kwargs = figure_out_kwargs(*args)
--> 974         return function(dat, **these_kwargs)
    975 
    976     return func, iterators

~/anaconda3/envs/pyxem/lib/python3.7/site-packages/pyxem/utils/indexation_utils.py in correlate_library(image, library, n_largest, mask)
    110                 # TODO: Factorise out the generation of corr_local to a method='mthd' section
    111                 # Extract experimental intensities from the diffraction image
--> 112                 image_intensities = image[px_local[:, 1], px_local[:, 0]]
    113                 corr_local = np.sum(np.multiply(image_intensities, int_local)) / \
    114                     pn_local  # Correlation is the partially normalized dot product

IndexError: index 256 is out of bounds for axis 1 with size 256

pc494, 2020-02-03:
First pass suggests;

EirikOpheim, 2020-02-04:
You are right! :)

closed 2020-02-04 by EirikOpheim