pyxem/pyxem #542: Improving IndexationResults & CrystallographicMap

pc494, 2020-02-13:
**Is your feature request related to a problem? Please describe.**
1) Currently `.save()` doesn't work for `IndexationResults`, which can often take a long time to create
2) We now have `orix` to work with rotations, making some functionality in `CrystallographicMap` redundant.

**Describe the solution you'd like**
I think turning indexation results into (internally) a single numpy array with known rows kills both birds with one stone.
```
phase, alpha, beta, gamma, correlation
phase, alpha, beta, gamma, correlation 
etc
```
Because (1) is resolved as this type of object is 'like' a diffraction pattern in that it's just a numpy array with extra info. Then when we parse this down to CrystallographicMap we can very easily extract the relevant subarray `.data[1:4]` to then convet and send in with `orix`'s `.from_euler()`

**Describe alternatives you've considered**
Help welcome.

**Additional context**
Given that `orix` is designed specifically with the orientation mapping purpose in mind now seems as good a time as any to pull this section of code towards this workflow. Also it helps us remove rotation code from pyxem, which I'm also keen to do.

dnjohnstone, 2020-03-26:
This issue will now be addressed by removing these classes and replacing the corresponding function using orix.CrystalMap - see https://github.com/pyxem/orix/pull/49

dnjohnstone, 2020-05-21:
I'm anticipating that pyxem-0.13.0 will be the first version to depend on orix-0.3.0 and therefore to use the new CrystalMap class, so adding this to the appropriate milestone.

dnjohnstone, 2020-07-21:
This will be dealt with in #615 and is now a 0.12.0 issue.

PR #654 (2020-11-22): Rebuilding the orientation mapping workflow to make use of orix.CrystalMap
---
name: Rebuilding the orientation mapping workflow to make use of orix.CrystalMap
about: A large change to the orientation mapping workflow structure to make use of new tech and experience.

---
closes #542 and closes #615 
**What does this PR do?**
- We remove the class, "`CrystallographicMap`" from pyxem, instead storing the results of an indexation as the orix object type, Crystal Map.
- We split `IndexationGenerator` into two classes, `TemplateIndexationGenerator` and `VectorIndexationGenerator`
- Remove code tha [... cut]
closed 2020-11-22 by dnjohnstone