pyxem/pyxem #547: Binning signal and navigation

dnjohnstone, 2020-02-22:
It would be nice to be able to bin the signal or navigation axes by a factor of 'n' - this may be possible with hyperspy functionality. But @M0hsend had written the following functions to do it. 

This is a memory prompt to either find the "correct" hyperspy way to do this or to implement some kind of more helpful utility for this.

def bin_sig(d, bin_fact):
    """
    bins the reshaped 4DSTEMhs object by bin_fact on signal (diffraction) plane
    Parameters:
    ------------
    d: hyperspy.signals.Signal2D
        reshaped to scanX, scanY | DetX, DetY
        This needs to be computed, i.e. not lazy, to work. If lazy, and binning 
        not aligned with dask chunks raises ValueError 
    Returns:
    -------------
    d_sigbin: binned d in the signal plane
    """
    # figuring out how many pixles to crop before binning
    # we assume the Detx and DetY dimensions are the same
    to_crop = d.axes_manager[-1].size % bin_fact
    d_crop = d.isig[to_crop:,to_crop:]
    try:
        d_sigbin = d_crop.rebin(scale=(1,1,bin_fact,bin_fact))
    except ValueError:
        print('Rebinning does not align with data dask chunks. Pass non-lazy signal before binning.')
        return
    return d_sigbin

def bin_nav(d, bin_fact):
    """
    bins the reshaped 4DSTEMhs object by bin_fact on navigation (probe scan) plane
    Parameters:
    ------------
    d: hyperspy.signals.Signal2D
        reshaped to scanX, scanY | DetX, DetY
        This needs to be computed, i.e. not lazy, to work. If lazy, and binning 
        not aligned with dask chunks raises ValueError 
    Returns:
    -------------
    d_navbin: binned d in the signal plane
    """
    # figuring out how many pixles to crop before binning
    # we assume the Detx and DetY dimensions are the same
    to_cropx = d.axes_manager[0].size % bin_fact
    to_cropy = d.axes_manager[1].size % bin_fact
    d_crop = d.inav[to_cropx:,to_cropy:]
    try:
        d_navbin = d_crop.rebin(scale=(bin_fact,bin_fact,1,1))
    except ValueError:
        print('Rebinning does not align with data dask chunks. Pass non-lazy signal before binning.')
        return
    return d_navbin

dnjohnstone, 2020-02-22:
This can be done with existing hyperspy functionality: 
E.g. to bin by 4x in each of the navigation dimensions use
dp.rebin(scale=(4,4,1,1))

closed 2020-02-22 by dnjohnstone

M0hsend, 2020-02-22:
There was a reason I wrote those. 
I think I wanted to be able to rebin lazy signals too. You would get an error if bin factor not integer / divisor of size in that case (so I crop first and then rebin):
`NotImplementedError: Lazy rebin requires scale to be integer and divisor of the original signal shape`

dnjohnstone, 2020-02-23:
> There was a reason I wrote those.
> I think I wanted to be able to rebin lazy signals too. You would get an error if bin factor not integer / divisor of size in that case (so I crop first and then [...]

Thanks @M0hsend - yeah I learnt this the hard way by making the script I was working on crash for that reason.

There is another advantage of your approach too, which is that it is more memory efficient because the integer divisor binning method is more efficient in that regard than the non-integer one - that also caused me an issue in trying to remove this.

Anyway, overall I think this is one that we should keep in workflow (i.e. in the script) in the short term and longer term look to upgrade the main hyperspy rebin method rather than putting this work around in to pyxem for example.