pyxem/pyxem #555: Dealing with Non-Crystalline Datasets

CSSFrancis, 2020-03-17:
**Is your feature request related to a problem? Please describe.**
I have been independently developing a hyperspy [add on](https://github.com/CSSFrancis/empyer) which is focused on analyzing amorphous materials using Fluctuation Electron Microscopy(FEM) which measures variance in diffraction patterns[1], angular correlation measurements which measure symmetry in diffraction patterns[2] and Election Correlation Microscopy which measure image correlations as a function of time (ECM)[3].

There is significant overlap for some of the basic functions i.e. polar reproduction, distortion correction, masking of zero beam/ beamstop etc.

I would be interested in moving a large part of that project into pyxem if you are interested as well.

**Describe the solution you'd like**
I can see this going a couple of different ways.

- Adding methods for analyzing amorphous materials to the Diffraction2D and  PolarDiffraction2D classes.
    - Calculating variance (using different FEM methods) and angular correlations
    - Creating a new Corrleation2D Class
- Creating a pyxem-am extension 
    - Create an AmorphousDiffraction2D class which inherits from the Diffraction2D class. As well as an AmorphousPolarDiffraction2D
    - Create AmorphousCorrelation2D class
    - Create ECM2D class

I like the idea of an pyxem-am extension as I don't know if the ECM datasets which are all time resolved fit in with some of diffraction based experiments. 

**Additional context**
For reference to the techniques:
[1] Daulton, T. L., Bondi, K. S. & Kelton, K. F. Nanobeam diffraction fluctuation electron microscopy technique for structural characterization of disordered materials-Application to Al88-xY7Fe5Tix metallic glasses. Ultramicroscopy 110, 1279–1289 (2010).

[2]Im, S. et al. Direct Determination of Structural Heterogeneity in Metallic Glasses Using Four-Dimensional Scanning Transmission Electron Microscopy. Ultramicroscopy 195, 189–193 (2018).

[3]Zhang, P., Maldonis, J. J., Liu, Z., Schroers, J. & Voyles, P. M. Spatially heterogeneous dynamics in a metallic glass forming liquid imaged by electron correlation microscopy. Nat. Commun. 9, 1–7 (2018).

dnjohnstone, 2020-03-17:
@CSSFrancis fantastic to hear from you - it's been on my list to contact you about this.

We'd be delighted to have your contributions in pyxem.

As you may have notices we have a VarianceGenerator and a PDFGenerator for FEM and PDF type things, although this part of the code has been a bit rudimentary so far because we haven't been actively doing a huge amount of reseearch in that area. I dabbled in FEM for a very short period when Mike Treacy was on sabbatical in our group, but never really did anything with it.

Personally I would much rather see this come in to pyxem than create a pyxem-am hyperspy-extension-extension because I think it can be useful in crystalline cases and as above we already want to support non-crystalline stuff anyway.

I think the first thing to do would be to finish this PR https://github.com/pyxem/pyxem/pull/524 to get the PolarDiffraction classes in.

Really we should be agnostic with respect to navigation dimensions so I think time dependence should "be the same as" spatial dependence and the commonality is in the signal dimensions.

dnjohnstone, 2020-03-17:
PS... we'd also be delighted to have some of your forward model stuff in https://github.com/pyxem/diffsims if you were interested in that

CSSFrancis, 2020-03-18:
Sounds like a good plan!  I need to clean up some of the modeling stuff but once I do I would love to add it.  I can get starting working on the PolarDiffraction2D class in the next couple of days if you would like? I started looking through what has already been written and it is fairly similar to what I have implemented.

dnjohnstone, 2020-03-19:
> Sounds like a good plan! I need to clean up some of the modeling stuff but once I do I would love to add it. I can get starting working on the PolarDiffraction2D class in the next couple of days if [...]

Sounds very good to me.

dnjohnstone, 2020-04-14:
As discussed in #569 with @AndrewHerzing - I've removed pixstem FEM code from that PR with the intention of restoring it in a generalized and consolidated framework.

The following should go back in in that framework

`def fem_calc(s, centre_x=None, centre_y=None, show_progressbar=True):
    """Perform analysis of fluctuation electron microscopy (FEM) data
    as outlined in:

    T. L. Daulton, et al., Ultramicroscopy 110 (2010) 1279-1289.
    doi:10.1016/j.ultramic.2010.05.010

    Parameters
    ----------
    s : PixelatedSTEM
        Signal on which FEM analysis was performed
    centre_x, centre_y : int, optional
        All the diffraction patterns assumed to have the same
        centre position.

    show_progressbar : bool
        Default True

    Returns
    -------
    results : Python dictionary
        Results of FEM data analysis, including the normalized variance
        of the annular mean (V-Omegak), mean of normalized variances of
        rings (V-rk), normalized variance of ring ensemble (Vrek),
        the normalized variance image (Omega-Vi), and annular mean of
        the variance image (Omega-Vk).

    Examples
    --------
    >>> import pixstem.dummy_data as dd
    >>> import pixstem.fem_tools as femt
    >>> s = dd.get_fem_signal()
    >>> fem_results = femt.fem_calc(
    ...     s,
    ...     centre_x=128,
    ...     centre_y=128,
    ...     show_progressbar=False)
    >>> fem_results['V-Omegak'].plot()

    """
    offset = False

    if centre_x is None:
        centre_x = np.int(s.axes_manager.signal_shape[0] / 2)

    if centre_y is None:
        centre_y = np.int(s.axes_manager.signal_shape[1] / 2)

    if s.data.min() == 0:
        s.data += 1  # To avoid division by 0
        offset = True
    results = dict()

    results["RadialInt"] = s.radial_average(
        centre_x=centre_x,
        centre_y=centre_y,
        normalize=False,
        show_progressbar=show_progressbar,
    )

    radialavgs = s.radial_average(
        centre_x=centre_x,
        centre_y=centre_y,
        normalize=True,
        show_progressbar=show_progressbar,
    )
    if radialavgs.data.min() == 0:
        radialavgs.data += 1

    results["V-Omegak"] = ((radialavgs ** 2).mean() / (radialavgs.mean()) ** 2) - 1
    results["RadialAvg"] = radialavgs.mean()

    if s._lazy:
        results["Omega-Vi"] = ((s ** 2).mean() / (s.mean()) ** 2) - 1
        results["Omega-Vi"].compute(progressbar=show_progressbar)
        results["Omega-Vi"] = pixstem.pixelated_stem_class.PixelatedSTEM(
            results["Omega-Vi"]
        )

        results["Omega-Vk"] = results["Omega-Vi"].radial_average(
            centre_x=centre_x,
            centre_y=centre_y,
            normalize=True,
            show_progressbar=show_progressbar,
        )

        oldshape = None
        if len(s.data.shape) == 4:
            oldshape = s.data.shape
            s.data = s.data.reshape(
                s.data.shape[0] * s.data.shape[1], s.data.shape[2], s.data.shape[3]
            )
        y, x = np.indices(s.data.shape[-2:])
        r = np.sqrt((x - centre_x) ** 2 + (y - centre_y) ** 2)
        r = r.astype(np.int)

        nr = np.bincount(r.ravel())
        Vrklist = []
        Vreklist = []

        for k in tqdm(range(0, len(nr)), disable=(not show_progressbar)):
            locs = np.where(r == k)
            vals = s.data.vindex[:, locs[0], locs[1]].T
            Vrklist.append(np.mean((np.mean(vals ** 2, 1) / np.mean(vals, 1) ** 2) - 1))
            Vreklist.append(np.mean(vals.ravel() ** 2) / np.mean(vals.ravel()) ** 2 - 1)

        Vrkdask = delayed(Vrklist)
        Vrekdask = delayed(Vreklist)

        results["Vrk"] = hs.signals.Signal1D(
            Vrkdask.compute(progressbar=show_progressbar)
        )
        results["Vrek"] = hs.signals.Signal1D(
            Vrekdask.compute(progressbar=show_progressbar)
        )
    else:
        results["Omega-Vi"] = ((s ** 2).mean() / (s.mean()) ** 2) - 1
        results["Omega-Vk"] = results["Omega-Vi"].radial_average(
            centre_x=centre_x,
            centre_y=centre_y,
            normalize=True,
            show_progressbar=show_progressbar,
        )
        oldshape = None
        if len(s.data.shape) == 4:
            oldshape = s.data.shape
            s.data = s.data.reshape(
                s.data.shape[0] * s.data.shape[1], s.data.shape[2], s.data.shape[3]
            )
        y, x = np.indices(s.data.shape[-2:])
        r = np.sqrt((x - centre_x) ** 2 + (y - centre_y) ** 2)
        r = r.astype(np.int)

        nr = np.bincount(r.ravel())
        results["Vrk"] = np.zeros(len(nr))
        results["Vrek"] = np.zeros(len(nr))

        for k in tqdm(range(0, len(nr)), disable=(not show_progressbar)):
            locs = np.where(r == k)
            vals = s.data[:, locs[0], locs[1]]
            results["Vrk"][k] = np.mean(
                (np.mean(vals ** 2, 1) / np.mean(vals, 1) ** 2) - 1
            )
            results["Vrek"][k] = (
                np.mean(vals.ravel() ** 2) / np.mean(vals.ravel()) ** 2 - 1
            )

        results["Vrk"] = hs.signals.Signal1D(results["Vrk"])
        results["Vrek"] = hs.signals.Signal1D(results["Vrek"])

    if oldshape:
        s.data = s.data.reshape(oldshape)
    if offset:
        s.data -= 1  # Undo previous addition of 1 to input data
    return results`

CSSFrancis, 2020-04-21:
@dnjohnstone That sounds like a good plan.  Is the idea here that this would replace the generator in favor of a more direct diffraction_signal.get_fem() method?

I have started a pull which formalizes the use of masks in hyperspy https://github.com/hyperspy/hyperspy/pull/2375 and I might finish that up before I dive into this. 

That pull request will help simplify some of the different FEM methods because I can use the nan.mean method on both lazy and non lazy signals without having to do some weird slicing with the lazy signal.

dnjohnstone, 2020-04-21:
> @dnjohnstone That sounds like a good plan. Is the idea here that this would replace the generator in favor of a more direct diffraction_signal.get_fem() method?
> 
> I have started a pull which formalizes the use of masks in hyperspy [hyperspy/hyperspy#2375](https://github.com/hyperspy/hyperspy/pull/2375) and I might finish that up before I dive into this.
> 
> That pull request will help simplify some of the different FEM methods because I can use the nan.mean method on both lazy and non lazy signals without having to do some weird slicing with the lazy signal.

I don't have any strong opinion about this - very open to suggestions for architecture on this.

I would however rather have get_variance() or other descriptive method names rather than get_fem() because I feel like a full FEM analysis is unlikely (perhaps not actually possible) from a single 4D-SED data.

What do you think would be best?

Thanks for the hyperspy PR

dnjohnstone, 2020-07-31:
@CSSFrancis are you still interested in doing this?

CSSFrancis, 2020-08-01:
@dnjohnstone Yea I started working on it the other day but it kind of spiraled into a bunch of other things. I can start a pull request on Monday once I clean things up and we can go from there.

PR #618 (2020-12-15): FEM-Redesign
---
name: **FEM-Redesign**
about: _This pull requests redesigns the FEM analysis code allowing the user to have more flexibility in how they perform the analysis.  The naming formalism follows the standard set in [1].

---

**Release Notes**
> new feature / improvement

Summary: Extends the functionality and flexibility of FEM measurements while speeding up some of the techniques using the new azimuthal integration. As I work on this I might consider breaking it apart into a couple of different pull requests as it is starting to grow. 

**What does this PR do? Please describe and/ [... cut]
closed 2020-12-15 by dnjohnstone