pyxem/pyxem #556: Possible bug in CalibrationGenerator

dnjohnstone, 2020-03-20:
@tinabe and @affaniqbal have reported issues with CalibrationGenerator via e-mail (please can we start doing this with code examples on Github ;) )

calibration_generator.py [1] line 293 and 294 read

au_pre = db - (self.ring_params[0] / 1.437)
au_post = db + (self.ring_params[0] / 1.437)

but I think that should be

au_pre = db - 1/(self.ring_params[0] * 1.437)
au_post = db + 1/(self.ring_params[0] * 1.437)

Just writing here so as not to lose track.

affaniqbal, 2020-03-22:
Thanks @dnjohnstone, I don't think this is the issue here, I think its correct.

Conversely, by making the changes, as 1/(self.ring_params[0] * 1.437) is quite small it gives au_pre and au_post to be quite close to db.

dnjohnstone, 2020-03-23:
#558 makes me notice that there are some other places in calibration_generator.py where we should also generalise from fixed numbers as @affaniqbal  did there. That should now be done in a new PR since I merged some things to avoid too many black formatting conflicts!

pc494, 2020-03-27:
I think what is currently in the code is correct. It seems the `self.ring_params` has dimensions [L]^{-1} and the LHS is clearly a distance in reciprocal angstroms. I can only presume the constant factor is dimensionless.

dnjohnstone, 2020-03-27:
Ok - I think this at least illustrates that this needs better comments in the code - I'll do that.

closed 2020-03-27 by dnjohnstone

tinabe, 2022-04-26:
I am looking into this again and have some comments - please let me know if you agree and/or if I have missed something.

1. I think that the misunderstanding I had previously (as reported above) is rooted in the `scale` parameter (`scale=self.ring_params[0]`) which is said in the documentation to be ‘An initial guess for the diffraction calibration in 1/Angstrom units’ (see Ref. 1). Here I would expect the calibration value to have units [px-1 Å-1]. However, the way it is used, it seems that it has units [px Å] (see Ref. 2), so that the ring radii in `rings = np.multiply(rings, scale)` have units: [Å-1][px Å]=[px]. Would it be possible to either describe this more clearly in the documentation, or change this code line so that the units of `scale` match the documentation? I think the code would then instead be: `rings = np.divide(rings, scale)` with units [Å-1] / [px-1 Å-1]=[px]. 
2. Also, it would be best it if a reference could be added for the inverse d-spacings and amplitude values listed for Au within this function (Ref. 2). 

3. Further, an issue that I have faced is related to `mask_length` which the user should provide in `get_diffraction_calibration`. If the parameter is too short (or too long, but that is more obvious), the peak finding method will not work as intended and the resulting calibration will be wrong. Would it be possible to describe this parameter better in demo03 or the documentation, and/or to let a safe value be set automatically if the user does not provide a value? This can for instance be done by finding the minima on either side of the direct beam, e.g. add the following code after line 353 (giving `mask_length = 34` for the demo03 data). It might be that the problem alternatively can be resolved by adjusting the input parameters to the peak finder, but I have not investigated that.
```
from scipy.signal import argrelextrema
minis = argrelextrema(trace.data, np.less)[0]
minis = minis[np.argsort(np.absolute([minis-db]))[0]][:2]
mask_length = abs(int((minis[0]-minis[1])/2))
```

4. Further, it could be useful to have the option to plot reference circles for all rings used for fitting, and not only one ring, as is currently done in `plot_corrected_diffraction_pattern(reference_circle=True)`. 

- Ref. 1: pyxem/generators/calibration_generator.py in get_elliptical_distortion, lines 126-128, and diffsims/utils/ring_pattern_utils.py in ring_pattern lines 67-69. 
- Ref. 2: diffsims/utils/ring_pattern_utils.py in ring_pattern lines 92-94.

tinabe, 2022-04-27:
That sounds like a good idea!