pyxem/pyxem #559: Grid-like features in background after apply_affine_transformation

affaniqbal, 2020-03-23:
**Describe the bug**

- Grid like features appear in the background in dp.mean() when using apply_affine_transformation. 

- The correction matrix I am using includes both distortion and rotation correction and is obtained from the reference calibration notebook in pyxem-demos.

**To Reproduce**
1. Load a 4D dataset (dp = pxm.load_hspy), see [link](https://drive.google.com/open?id=1ikpX59XgU7peYvEMgoU8uiC75AbTScC_) for an example file (16mb file size)

2. Run dp = pxm.ElectronDiffraction2D(dp)

3. Run dp.apply_affine_transformation(np.array([[ 0.21725534, -0.94230027,  0.        ],
                                                                              [ 0.97510544,  0.22137543,  0.        ],
                                                                              [ 0.        ,  0.        ,  1.        ]]),
                                                               keep_dtype=True, inplace=True)
4.Run dp.mean().plot()

**Expected behaviour**

- apply_affine_transform should correct for distortion and apply rotation of 76 degrees. This is expected and is observed.

- Alongside this, the background of the dp changes to a grid-like pattern as opposed to a uniform background.

**Screenshots**

- Screenshot 1 is dp.mean() of file as loaded

- Screenshot 2 is dp.mean() of file after apply_affine_transform

- Screenshot 3 is the code used

[image]
[image]
[image]

**Additional context**
- The appearance of grid-like features is seen to occur **only when apply_affine_transform includes rotation correction.**

dnjohnstone, 2020-03-23:
Thanks for the report - can you try:
1. setting keep_dtype=False
2. setting order=5

pc494, 2020-03-23:
Excellent bug report @affaniqbal, if duncan's suggestions don't work let me know and I'll have a look. Have a couple of vague ideas I am happy to try out.

affaniqbal, 2020-03-23:
Thanks @dnjohnstone & @pc494. 

1. I tried keep_dtype = False (with order = 3)
- This fixes the current issue (i.e. gives smooth background)
- However it leads to a 8x increase in filesize (dp.data.nbytes), this may become an issue with large mib files and/or processing scripts in cluster.
- It also leads to a drop in intensity but this can be solved by using preserve_range  = True

2. I tried setting order = 5 (and trying both dtype = True and False)

- This leads to same grid-like features but to a lesser extent

dnjohnstone, 2020-03-24:
Thanks Affan. I think then there is not really a bug in the code - but we do probably need to implement more of our processing in a more RAM efficient way. Do you think you could have a look in to potential solutions to that? E.g. re-writing this in a way that can be done "lazily" perhaps using dask?

hakonanes, 2020-03-24:
Just want to chime in that pixstem (@magnunor ++) uses dask in a smart way, e.g. the `template_match_with_binary_image()` method:
1. a dask array is always created from the signal array (if not already dask) (https://gitlab.com/pixstem/pixstem/-/blob/master/pixstem/pixelated_stem_class.py#L736)
2. the dask array passed to a `template_match_binary_image_chunk()` routine operating on individual chunks (https://gitlab.com/pixstem/pixstem/-/blob/master/pixstem/dask_tools.py#L164)
3. images in individual chunks are passed to `template_match_binary_image_single_frame()` (https://gitlab.com/pixstem/pixstem/-/blob/master/pixstem/dask_tools.py#L135)

I've adopted this approach for all pattern processing in KikuchiPy, which allows me to cast 8-bit patterns to 32-bit for operations, but still keep memory usage at an acceptable level.

dnjohnstone, 2020-03-24:
Yeah - I've been speaking to @magnunor about hopefully merging pixSTEM in with pyxem, which I hope will happen in the next couple of months.

magnunor, 2020-03-24:
Indeed! If you're able to keep all the processing using `dask`, and within each chunk, you can do some very, very memory and time efficient processing. It takes a little while to get used to this, but I've managed to implement in a pretty nice fashion in the `dask_tools` package.

pc494, 2023-01-16:
Too many changes have happened to the API since this was raised for me to know if it's valid - closing.

closed 2023-01-16 by pc494