pyxem/pyxem #563: "typing" in pyxem family packages.

dnjohnstone, 2020-03-23:
In a recent PR to orix https://github.com/pyxem/orix/pull/49 - the topic of "typing" came up, because that PR had lots of try/except statements to check the type of variables passed to some methods.

I think we should just take a moment here to agree a policy on dealing with type in pyxem/orix/diffsims.

The approach we have taken until now is I think summarised as - "Python is a dynamically typed language, we only type check when a function/method can take a variable in more than one type, and we declare the type in the docstring." Users are therefore expected to be able to read the docstring and to understand/be able to determine what type their variable is when they pass it to a function/method.

I think that this is fine, maybe even "correct", except that this "basic" Python literacy regarding typing turns out to be something that we cannot necessarily assume amongst our user base. Because the primary way to interact with pyxem is by writing scripts, often in Jupyter notebooks, users do have to take care of type and this can cause issues. Further, the error message that you get when type is incorrect is often somewhat obtuse because things fail wherever the type is an issue and the Error message usually doesn't say "the type of variable x is wrong" - this is what @hakonanes seemed to want to make happen.

Using try/except statements to check the type of variables and raise relatively easy to understand Error messages when wrong doesn't seem like a very good, or very pythonic, solution to me.

We could consider using annotation as introduced in PEP 484 - see: https://docs.python.org/3/library/typing.html - but I don't think this actually checks type, only declares it? which we arguable already do in our docstrings...

So there are 3 questions:
1. Do we all agree that as a rule try/except statements should not be used for type checking.
2. Do we want to use annotation?
3. Should we add something to an initial user guide to tell people that they need to understand type?

dnjohnstone, 2020-03-23:
@hakonanes and @pc494

pc494, 2020-03-23:
I'm pretty firm in my belief that type-checking in python is a bad idea. I think some clear information that says "if something goes wrong, the first thing to do is check all the types of your inputs" would go a long way to fixing this problem. I am not that keen on annotation either.

hakonanes, 2020-03-23:
Thanks for bringing this up, @dnjohnstone.

1. To quote myself from https://github.com/pyxem/orix/pull/49:
> I become a better programmer mainly by reading error messages... I've therefore included many try/except which raises ValueError when input is not correct or IndexError in __getitem__() when indexing key is invalid. orix is a library. I think it should raise clear error messages immediately when unexpected input was passed, instead of numpy or matplotlib raising errors later when the initial context is lost.

I don't know how many, if any, share this view. But based upon the arguments you're laying out, and the fact that it is arguably easier to maintain and read code without these checks, you've changed my mind, and I agree with your sentiment.

2. In the top of PEP 484:

> Note: The Python runtime does not enforce function and variable type annotations. They can be used by third party tools such as type checkers, IDEs, linters, etc. 

So I think you're right, Python itself won't raise an error message when an argument does not have the indicated type in the function header. I think the benefit from this would be that my PyCharm editor would tell me that, no, that variable which is an integer cannot be iterated over, for example. I think it is not worth it to adopt typing just for these editor hints.

3. Sort of yes, but that might scare users as well... I think the best approach would be to state two things clearly:
   i. That all necessary information can be found in docstrings (via Shift + Tab + Tab in notebooks)
   ii. Where the user can find help if something behaves unexpectedly, which you already do in notebooks with pointing to github.

dnjohnstone, 2020-03-23:
Ok, great - I think we all agree on this in the end then. Please can you update the orix PR accordingly and we'll deal with the documentation in due course.

closed 2020-03-23 by dnjohnstone