pyxem/pyxem #568: pyXem development plan

dnjohnstone, 2020-03-24:
It might be helpful for developers new and old if we write down a development plan for pyxem - I have been thinking about this for the past few days while reviewing many open topics and discussions both on github and elsewhere. I'm also aware that ongoing global events mean that many people may either have more time or inclination or both to pursue code based projects.

Based on these discussions there are a few fairly substantial plans for pyxem in the near term, which are all very exciting. To document these plans in one place, we intend to:

1. Consolidate existing functionality and improve code quality.

2. Refactor parts of the code dealing with the results of phase and orientation mapping to orix (https://github.com/pyxem/orix/pull/49) and therefore remove VectorMachingResults, TemplateMatchingResults and CrystallographicMap classes from pyxem, with @hakonanes 

3. Work with @magnunor to merge pixSTEM (https://gitlab.com/pixstem/pixstem) into pyxem - this follows offline discussions with @chrisallen80 and @M0hsend

4. Work with @CSSFrancis to incorporate improved methods for non-crystals into pyxem - based on empyer (https://github.com/CSSFrancis/empyer/tree/master/empyer) - see #555 

 5. Consider where the code for VDF segmentation introduce in #515 should be.

Based on this I anticipate that we will have a few minor releases in the coming few weeks and months to do primarily the following:

v0.11.0 --> Consolidate existing functionality and add PolarDiffraction classes (I have labelled open PRs and issues that I expect to go here in the corresponding milestone. (due ca. 1st Apr.).

v0.12.0 --> The initial release in which we merge pixSTEM and pyxem. (due mid to late Apr.)

v0.13.0 --> Incorporation of new methods and further consolidation. (due mid to late May.).

There are quite many stakeholders in this plan now - so please all comment if you agree with this plan or would prefer us to work in a different way.

I am considering organising semi-regular online developer meetings in order to chat through this plan and track progress over the next few months - if you would be interested in that please either send me an e-mail or comment here and we'll see if there is sufficient demand for this?

hakonanes, 2020-03-24:
Interested in participating in online developer meetings (hwaanes@gmail.com), mainly because if at some point in the future (not within the plan above, I assume) there are discussions of introducing EBSD functionality in pyxem, I would like KikuchiPy to be part of that discussion. I'm also interested in the dead/hot pixel corrections in pixstem... But that can be a discussion for later!

I would want to participate in the refactoring in point 2 after https://github.com/pyxem/orix/pull/49 is merged. Apart from that, I don't have any further comments.

pc494, 2020-03-24:
This looks good to me and I am all set up to do a lot of reviewing in the near future! :)

CSSFrancis, 2020-03-24:
This looks good to me. I would be interested in participating in online developer meetings as well. 

I think that with these expansions it might be worth it to flesh out some of the documentation. While the Jupyter Notebook demos are a good resource to learn how implement something it might be good to take some of the outputs from those files and build something like [Hyperspy's User Guide](http://hyperspy.org/hyperspy-doc/current/user_guide/index.html) which might be more accessible to the average user who is just starting out.

magnunor, 2020-03-24:
With the big merge, I'm also interested the dev meeting.

> I think that with these expansions it might be worth it to flesh out some of the documentation. While the Jupyter Notebook demos are a good resource to learn how implement a certain technique using pyXEM it might be good to take some of the outputs from those files and build something like Hyperspy's User Guide which might be more accessible to the average user who is just starting out.

My really big pull request also contains fairly extensive documentation for `pixstem`'s functionality, so it shouldn't be too hard to refactor and extend that documentation. I do agree having a nice website (with nice images!) is really useful for people to understand what the software actually does.

dnjohnstone, 2020-03-24:
Ok, great - I think that is sufficient interest that we should go ahead with developer meetings. I will e-mail those who have indicated an interest here with meeting details.

Is everyone ok with the first meeting being Wednesday 1st April 2020 @ 15.00 UK time?

I think that balances working days in various relevant time zones reasonably well...

Regarding documentation - we've so far been inspired in our approach by The Agile Manifesto:
https://en.wikipedia.org/wiki/Agile_software_development#The_Agile_Manifesto

(incidentally I'd quite like to re-write that manifesto using words more appropriate to the open-source scientific software community)

Part of our take on that has been to focus on delivering end-to-end workflows for scientists, which we document by example in Notebooks, rather than writing some kind of comprehensive user guide.

One reason we took that approach is because we initially did write documentation very similar to the hyperspy user guide and got to a point where we were wasting time re-writing it as we modified the code architecture quite rapidly to build better workflows for people.

Another noteable advantage was that we update the notebooks as part of every release cycle and release a working version with every version of pyxem - this ensures that the demos (as obtained from zenodo) should all run correctly with the corresponding release version. It was a real pain to try to guarantee that the documentation was similarly correct in a rapidly developing dynamic code base with quite many active contributors.

We also didn't really like having all the website/documentation code in the main repository.

So upshot is - we do have a sphinx based website: https://pyxem.github.io/pyxem-website/

That website currently only documents basic things like axis conventions, basic installation, and how to get help etc.

We also have a "Documentation" section built of of the docstrings for every function/method in the code.

https://pyxem.github.io/pyxem-website/getting_started.html#documentation

So my personal preference would be if we do the following:

1. Keep the documentation in a separate repository i.e. https://github.com/pyxem/pyxem-website
2. Continue to prioritize notebook demonstrations that are checked as part of the release cycle - including creating new demos for all significant new functionality.
3. Hold off attempting comprehensive documentation until we're in a rather more steady state for pyxem.

Would everyone be ok with this?

M0hsend, 2020-03-24:
All sounds very exciting! Definitely interested in joining the developers meeting.

hakonanes, 2020-03-25:
The meeting time is OK for me.

Regarding holding off on writing docs, I think you've made good arguments against introducing it at this point, especially based upon your experience. While I would want to see the documentation website via http://pyxem.org be updated (e.g. the "Docstring" link in the description under "Documentation" points to "Indices and tables", which has a "pyXem 0.9.0" in the navbar, with the "pyXem" home going nowhere), I agree that the focus should be on notebooks.

dnjohnstone, 2020-05-18:
Just in case anyone didn't notice - we've now released pyxem-0.11.0.

Owing to some release issues (which we seem to have nearly every time at the moment, see #601) there will likely be a v0.11.1 release quite promptly. I hope that this will include #590 because that's pretty close too.

I also want to accelerate #569 by removing the changes to lazy peak finding from that PR because those are really quite substantial changes that I think are best addressed in separate PRs and it would be good to have people using the DPC stuff from pixSTEM ASAP.

Do let me know if you have any comments/suggestions/requests.

pc494, 2021-01-13:
I'm closing this, as all but 1 of the points in the OP are dealt with in the `0.13.0` version. Overhauling diffraction vectors has (yet again) been pushed back.

closed 2021-01-13 by pc494