pyxem/pyxem #580: Performance of peak_finders

tjof2, 2020-04-03:
First contribution here, so be nice :)

I've been taking a quick look at the peak-finding methods implemented in `pyxem/utils/peakfinders2D.py` - in particular, the following two:

- `find_peaks_zaefferer()`
- `find_peaks_stat()`

### `find_peaks_zaefferer()`
With a couple of small Python-only tweaks, I'm able to improve the performance of `find_peaks_zaefferer()` by ~20-30% for the case of large images and/or lots of peaks.

[image]

### `find_peaks_stat()`
This is a bit more work, though not much. I was looking into the performance of `scipy.ndimage.filters.generic_filter`, which this function uses, and came across https://github.com/scipy/scipy/issues/8916.

By replacing `np.mean` and `np.std` inside `find_peaks_stat()` with the following:
```
from numba import njit

@njit(cache=True)
def mean_thresholding(window):
    return np.mean(window)

@njit(cache=True)
def std_thresholding(window):
    return np.std(window)
```
There is a significant improvement for large images:

[image]

I don't know enough about how often these two functions are used by people compared to either the Laplacian-of-Gaussian or Difference-of-Gaussian functions, but are these improvements worth pursuing?

The downside of the second change is of course the introduction of a dependency on Numba, which may or may not be desirable.

dnjohnstone, 2020-04-03:
Thanks @tjof2 - awesome first contribution. It's almost like you've done this before elsewhere... ;)

In short, yes, please.

I think this is definitely worth pursuing. The stat method has appealingly few parameters and I think would be used much more often if it weren't so slow!

A few things to note, mostly just for info:

1. We actually already have a weak dependency on numba from our diffraction simulation package (https://github.com/pyxem/diffsims)  - so I'm quite relaxed about that.

2. We definitely need to speed up quite a few things in pyxem as I think we are approaching the point where speed is more of a problem than functionality for several things. I think it's likely that we'll try to get most of that speed-up from numba + dask (see #569 for where our initial dask stuff will come from). - So again I think we're just going to end up going there with numba eventually.

3. center_direct_beam() is another method that needs looking at in this regard (https://github.com/pyxem/pyxem/blob/master/pyxem/signals/diffraction2d.py) it's sometimes doing some stupid things with CPU useage at the moment.

tjof2, 2020-04-03:
Thanks @dnjohnstone - I can create a PR with these two changes pretty quickly in that case if you're happy with numba.

I'll take a look around as I work my way through the code to see if there's anything else that jumps out, including point (3).

dnjohnstone, 2020-04-03:
I suppose I should say that I'm open to other suggestions (paging @pc494 and @magnunor) on ways to get significant speed enhancements but I think - numba + dask seems like a good way to go that will almost certainly be well supported by the python community in the long run...

dnjohnstone, 2020-04-03:
> Thanks @dnjohnstone - I can create a PR with these two changes pretty quickly in that case if you're happy with numba.

Do you see any significant issues with introducing numba code? I feel like it's going to be the way things go for getting speed with relative ease in python - but always open to alternatives...

I think in this case the stat method as it stands is just too slow for anyone to use it, so it's basically useless if we don't do something like this.

dnjohnstone, 2020-04-04:
PS... If you're looking for places where performance improvements would have a significant impact - the correlate() and index_vectors() methods here (https://github.com/pyxem/pyxem/blob/master/pyxem/generators/indexation_generator.py) are also problematically slow at the moment.

pc494, 2020-04-04:
Getting numba in at this stage seems like a good idea to me. Would be really cool to have the 'stat' method working at a speed that people could actually use it.

tjof2, 2020-04-04:
I'd be happy to take a look at those other functions after this one @dnjohnstone - if you have a particularly good code example to show them being slow, that'd be great!

tjof2, 2020-04-06:
Ok so one problem with Numba I noticed while finishing #582 is that coveralls won't measure line-by-line test coverage for Numba functions (e.g. https://github.com/numba/numba/issues/4268).

A few options:

- If `f(x)` is a Numba-compiled function, then in the unit tests, test the result of `f.py_func(x)` which will run the underlying Python code without compilation - and the test coverage is then good.
- When measuring test coverage, disable JIT by setting the `NUMBA_DISABLE_JIT` environment variable to a non-zero value.

dnjohnstone, 2020-04-06:
Ah yes, I forgot about that issue - but I don't think we should let this stop us and your fix in this case is good.

I vaguely recall thinking that codecov somehow overcame this issue? Does anyone know if that's true, I can't find a handy stack overflow or whatever to corroborate myself...

tjof2, 2020-04-06:
I think you may be right - though note that anyone using `pytest-cov` locally will also encounter this issue.

tjof2, 2020-04-07:
Closing in favour of #585

closed 2020-04-07 by tjof2