pyxem/pyxem #583: Unused convergence in find peaks stat method

dnjohnstone, 2020-04-06:
In #583 @tjof2 pointed out an unused bit of code:

    def stat_peak_finder(image):
        """Find peaks in image. Algorithm stages in comments."""
        image = normalize(image)  # 1
        image = stat_binarise(image)  # 2, 3
        n_peaks = np.infty  # Initial number of peaks
        image, peaks = _peak_find_once(image)  # 4-6
        m_peaks = len(peaks)  # Actual number of peaks
        """
        #XXX
        while (n_peaks - m_peaks) / n_peaks > convergence_ratio:  # 8
            n_peaks = m_peaks
            image, peaks = _peak_find_once(image)
            m_peaks = len(peaks)
        """

        # Algorithm branch not currently used
        # while (n_peaks - m_peaks) / n_peaks > convergence_ratio:  # 8
        #     n_peaks = m_peaks
        #     image, peaks = _peak_find_once(image)
        #     m_peaks = len(peaks)

We should use it or remove it!

tjof2, 2020-04-06:
Perhaps the speed of the code was an issue? Maybe it's better now. I don't have the reference available though so can't comment on how necessary it is

pc494, 2020-04-07:
I think it was getting it covered that was an issue. Would probably put it back in with a pragma no cover, as I'm pretty sure it it is a meaningful step in the algorithm

dnjohnstone, 2020-04-08:
> I think it was getting it covered that was an issue. Would probably put it back in with a pragma no cover, as I'm pretty sure it it is a meaningful step in the algorithm

Could you do that for v0.11.0 please?

dnjohnstone, 2020-04-13:
Ok - I've started looking at this in #581 - conveniently I took Tom White's thesis home with me pre-shutdown so I've written out the algorithm as described there in the docstring of the find_peaks_stat method - which is probably a useful addition to the code in any case.

Looking back through the code it seems to me that we've never had this quite right for some reason. I had a very brief bash at fixing it but didn't quite get there and ran out of time - @pc494 please could you have a look at it in #581 ?

closed 2020-05-06 by dnjohnstone