pyxem/pyxem #585: Code performance improvements

tjof2, 2020-04-07:
In #580, the performance of peak_finders was discussed and addressed in #582 through some small Python changes and the introduction of Numba into pyxem.

The purpose of this issue is to collect examples of performance bottlenecks in pyxem.

- `center_direct_beam()` is another method that needs looking at in this regard - [see here](https://github.com/pyxem/pyxem/blob/master/pyxem/signals/diffraction2d.py). It's sometimes doing some stupid things with CPU usage at the moment.
- the `correlate()` and `index_vectors()` methods [here](https://github.com/pyxem/pyxem/blob/master/pyxem/generators/indexation_generator.py) are also problematically slow at the moment.

dnjohnstone, 2020-04-07:
Thanks for this @tjof2 - you can see examples of this functionality [here](https://github.com/pyxem/pyxem-demos) in demo1 and demo4 respectively.

In demo4 - the correlate method doesn't appear to take that long, but this is primarily because we made the template library very small to illustrate behaviour. If we include all symmetry inequivalent orientations it's a couple of orders of magnitude more templates and that's what causes the issue.

pc494, 2020-04-07:
More details on `correlate`. The problem is that even though it's a tiny bit of code, in normal usage it will be run

N x M times

where N is the number of patterns in a scan, and M is the number of templates needed in a library, both of these numbers can easily reach the order of 1,000s.

tjof2, 2020-04-07:
Indeed, thanks @pc494. Probably still something to gain despite the nested for-loops, e.g. if the whole thing can be put inside `@numba.jit(nopython=True)`

CSSFrancis, 2020-04-08:
@dnjohnstone  and @pc494  I am not sure exactly sure how correlate works but my understanding is that a large part of the pattern matching is setting up the same diffraction pattern at different orientations.

A major speed up might be to azimuthally integrated2D sample and then cross correlate along the angular axis with a simulated azimuthally integrated2d pattern.

For two polar signals this comparison would look like:

```
random_oriented_diffraction  # an on zone axis pattern randomly rotated
simulated_diffraction  # an on zone axis pattern

random_oriented_diffraction_fft =  np.fft.fft(random_oriented_diffraction, axis=1) #along angular axis
simulated_diffraction_fft =  np.fft.fft(random_oriented_diffraction, axis=1) #along angular axis
cross_correlation = np.fft.ifft( random_oriented_diffraction_fft *
                                                   np.conjugate(simulated_diffraction_fft),
                                                   axis=1).real
correlations_vs_angle = np.sum(cross_correlation, axis=0)
```

Where you gain a lot of speed using FFT's to do the correlation.

dnjohnstone, 2020-04-09:
Thanks for that @CSSFrancis - the current correlate function minimises the number of pixels evaluated to just those where a Bragg reflection is expected in the pre-calculated templates. There are lots of problems with that, but it's a standard method.

What you're describing, I think, is "full frame correlation" actually simulating the whole diffraction pattern and comparing all the pixels somehow. It's absolutely something we want to have in pyxem. We'd been nudged in that direction by: https://www.sciencedirect.com/science/article/abs/pii/S0304399118302626

In fact, this was the reason why I'd implemented the PolarDiffraction2D class in that branch as a platform for some student projects.

It's something I'd discussed with @EirikOpheim, @norskie and @tinabe - but I'm not sure if any of them are actively working on it at the moment.

Anyway it would be great if you're able to work on that - and also if we can make sure that there's not too much duplication of efforts with the various people I've tagged in this comment.

tjof2, 2020-04-09:
@dnjohnstone  did some more digging into this:

> `center_direct_beam()` is another method that needs looking at in this regard. It's sometimes doing some stupid things with CPU usage at the moment.

This looks like all the work is in `.align2D()` which comes from HyperSpy correct? I [see in there](https://github.com/hyperspy/hyperspy/blob/dbb9821d5a9d41dd6a78af646a73c499cc0ceaa6/hyperspy/_signals/signal2d.py#L650) that it relies on `_map_iterate()`, which when `parallel=True` spawns threads, but as @thomasaarholt and I discovered [in this issue](https://github.com/hyperspy/hyperspy/issues/2320#issuecomment-611433876), that won't necessarily speed it up due to the GIL. It requires proper multiprocessing rather than multithreading. 

Note that if `parallel` is not specified, then:
```
if parallel is None:
    parallel = preferences.General.parallel
if parallel is True:
    from os import cpu_count
    parallel = cpu_count() or 1
```
So even if you're not passing this parameter, it's quite possible it is spawning multiple threads based on your preferences, and this might cause the CPU usage you see. It would be worth trying to set `parallel=False` explicitly to see if it changes.

hakonanes, 2020-04-09:
Just want to say that I would be interested in looking into doing full frame correlation in some form similar to what's done in the mentioned paper in KikuchiPy in the future. If there are plans to implement this in pyxem, I would be very delighted if it was done in a general way, with dask, so that it could be done with EBSD patterns as well.

dnjohnstone, 2020-04-09:
> This looks like all the work is in `.align2D()` which comes from HyperSpy correct? I [see in there](https://github.com/hyperspy/hyperspy/blob/dbb9821d5a9d41dd6a78af646a73c499cc0ceaa6/hyperspy/_signals/signal2d.py#L650) that it relies on `_map_iterate()`, which when `parallel=True` spawns threads, but as @thomasaarholt and I discovered [in this issue](https://github.com/hyperspy/hyperspy/issues/2320#issuecomment-611433876), that won't necessarily speed it up due to the GIL. It really requires proper multiprocessing rather than multithreading.

Ah - yes that's exactly an issue that recently came up when we tried to use center_direct_beam() on the Diamond Light Source cluster... The massive thread spawning played horrifically with the job managers on the cluster and it led to this method taking many hours to complete rather than being accelerated by the additional computing resource.

Great that the source is becoming a little more clear!

Paging @M0hsend, @ptim0626 and @mcduta (Note this issue was Ticket=DLS-22584)

tjof2, 2020-04-09:
Yes that sounds exactly like what would happen. Exposing the parallel kwarg in pyxem might be useful as a first step.

CSSFrancis, 2020-04-09:
> What you're describing, I think, is "full frame correlation" actually simulating the whole diffraction pattern and comparing all the pixels somehow. It's absolutely something we want to have in [...]

Cool, I haven't looked much into correlations with crystalline materials but I think there is a good degree of potential for this.  @hakonanes I have a general correlation method that I wrote for doing self correlations along some axis but I don't think it would be too hard to make a 2 dimensional version and cross correlations. I can see about writing a more comprehensive correlation method with the angular correlations pull request #572 

As far as speed goes, the numpy fft module is fairly optimized and runs pretty fast.  Eventually, though I think that a GPU enabled implementation is the answer if you want to do many many correlations.  This might be something to discuss as a potential add-on.

tjof2, 2020-04-09:
> As far as speed goes, the numpy fft module is fairly optimized and runs pretty fast.

There may be instances where `scipy.fftpack.fft` is faster than numpy, though it needs testing to be sure, and is dependent on the use case with complex dtypes vs separate real+imaginary floats. I certainly came across a couple of instances in that linked issue for HyperSpy where numpy was slower by a factor of 2x compared to scipy.

CSSFrancis, 2020-04-09:
> There may be instances where `scipy.fftpack.fft` is faster than numpy, though it needs testing to be sure, and is dependent on the use case with complex dtypes vs separate real+imaginary floats. I [...]

Huh, interesting, thanks for the tip.  I think both packages use [FFTPack] (https://www.netlib.org/fftpack/)  which is a Fortran program for the calculations but it seems like there are different implementations of the set up and take down.

A quick search gave me this related to their speeds but I'll look into it further.

https://stackoverflow.com/questions/6363154/what-is-the-difference-between-numpy-fft-and-scipy-fftpack

tjof2, 2020-04-09:
FWIW @CSSFrancis I think the difference might be from things like scipy automatically using rfftn() for real-valued inputs rather than fftn(). Numpy probably has no such smart logic.

dnjohnstone, 2020-04-09:
> Cool, I haven't looked much into correlations with crystalline materials but I think there is a good degree of potential for this. @hakonanes I have a general correlation method that I wrote for [...]

Yeah, that was going to be my suggestion also.

I think we should turn #572 into a PR for general pattern-to-pattern FFT correlation of either Diffraction2D or PolarDiffraction2D.

Perhaps this would fit in to pyxem best as an FFTCorrelationGenerator that would take a reference pattern (or library of patterns) and a Diffraction2D or PolarDiffraction2D and the case where this is in fact an autocorrelation then just becomes a special case.

We can obviously work out the details in #572 - but I think something like this is going to be the best architecture in terms of code reuse in the medium/long term.

thomasaarholt, 2020-04-09:
What size arrays do you typically perform FFTs/correlation on? I noticed that a search for CuPy on pyxem (and hyperspy) didn't return any results, and that @tjof2's GPU comment was the only one mentioning GPUs. 
[CuPy](https://cupy.chainer.org/) is a GPU/CUDA drop-in replacement for NumPy that is *really* easy to use. If you're performing FFTs on either A) individual 2D arrays larger than 128x128 or B) basically any stack of 2D arrays, I can recommend trying CuPy. I've [modified a notebook](https://github.com/thomasaarholt/thomasaarholt.github.io/blob/master/fftspeedtest/CuPyVSNumPy.ipynb) I made earlier to compare NumPy vs CuPy for a variety of array sizes if anyone is interested in seeing what one achieves on a 1080TI.

dnjohnstone, 2020-04-09:
We don't have any large FFTs in pyxem at the moment - that bit of this conversation is all about future code.

In general, we have so far felt that we had more to gain from distributed computing on many CPUs, partly because of the computing resources that current developers happen to be using and partly because we'll need CPU implementations for everything in any case because we're not going to have GPU only code unless there is some exceptionally good reason. Although I appreciate that working between cupy and numpy is easy for that.

We do have some big FFTs in diffsims, which are implemented with GPU acceleration but using numba and not cupy (https://github.com/pyxem/diffsims/pull/33/) - I'd be open to rewriting that where possible with cupy.

EirikOpheim, 2020-04-10:
> It's something I'd discussed with @EirikOpheim, @norskie and @tinabe - but I'm not sure if any of them are actively working on it at the moment.

> Anyway it would be great if you're able to work on that - and also if we can make sure that there's not too much duplication of efforts with the various people I've tagged in this comment.

I was planning on starting to work on the full frame correlation next week. I have not looked into 2D polar correlation yet.

tjof2, 2020-04-13:
At the very least with numpy/scipy FFTs, I think it should utilise [`scipy.fft.next_fast_len`](https://docs.scipy.org/doc/scipy/reference/generated/scipy.fft.next_fast_len.html#scipy.fft.next_fast_len) to do better than just padding to the next power-of-two.

dnjohnstone, 2020-04-13:
> At the very least with numpy/scipy FFTs, I think it should utilise [`scipy.fft.next_fast_len`](https://docs.scipy.org/doc/scipy/reference/generated/scipy.fft.next_fast_len.html#scipy.fft.next_fast_len) to do better than just padding to the next power-of-two.

Was that a general comment or in reference to a particular bit of code (I agree, by the way)?

tjof2, 2020-04-13:
General really - it can make a huge impact looking at the scipy docs!

tjof2, 2020-04-13:
I don't see them being used much here so definitely a general comment, but it's one of the changes to go into Hyperspy for align2D

tjof2, 2020-04-15:
The changes to estimate_shift_2D are [merged to HyperSpy RELEASE_next_minor](https://github.com/hyperspy/hyperspy/pull/2358) now, so hopefully you should see some improvements in `center_direct_beam` going forwards.

pc494, 2021-01-13:
Closing this as it requires no action at present.

closed 2021-01-13 by pc494