pyxem/pyxem #591: Possible bug in plot_best_matching_results_on_signal() for template matching

norskie, 2020-04-15:
**Describe the bug**
The order in which I add the rotation list structure library governs the output of the template matching routine plot_best_matching_results_on_signal(). 

For example, if I put Aluminium first, then I will only obtain the best matching results for Aluminium, while if I put betadp first I will only obtain the best result for betadp. (See screenshots)

```
struc_lib = StructureLibrary(['Aluminium', 'Betadp'],
                            [structure_Aluminium, structure_Betadp )],
                            [rot_list_Al, rot_list_Betadp])
```

gives a different result than

```
struc_lib = StructureLibrary(['Betadp', 'Aluminium'],
                            [structure_Betadp, structure_Aluminium)],
                            [rot_list_Betadp, rot_list_Al])
```

**To Reproduce**
1. Create rotation lists  using get_local_grid and get_grid_around_beam_direction
2. Make two StructureLibraries where the two elements are swapped. 
3. Perform the usual template matching routine

```
rot_list_Al = get_local_grid(get_rotation_from_z_to_direction(structure_dictionary.get("Aluminium"), [0, 0, 1]), 2, 2)

beam_rotation_betadp = get_rotation_from_z_to_direction(structure_dictionary.get("Betadp"), [0, 1, 0])
rot_list_betadp = get_grid_around_beam_direction(beam_rotation_betadp, 1, angular_range=(0, 360))

struc_lib = StructureLibrary(['Aluminium', 'Betadp'],
                            [diffpy.structure.loadStructure('./Aluminium.cif'), diffpy.structure.loadStructure('./Betadp.cif' )],
                            [rot_list_Al, rot_list_betadp])

diff_gen = DiffractionGenerator(accelerating_voltage=accelarating_voltage,
                                max_excitation_error=max_excitation_error)

lib_gen = DiffractionLibraryGenerator(diff_gen)

target_pattern_dimension_pixels = dp.axes_manager.signal_shape[0]

half_size = target_pattern_dimension_pixels // 2

reciprocal_radius = diffraction_calibration*(half_size - 1)

diff_lib = lib_gen.get_diffraction_library(struc_lib,
                                           calibration=diffraction_calibration,
                                           reciprocal_radius=reciprocal_radius,
                                           half_shape=(half_size, half_size),
                                           with_direct_beam=False)

indexer = IndexationGenerator(dp, diff_lib)
indexation_results = indexer.correlate(n_largest = 3)
crystal_map = indexation_results.get_crystallographic_map()

indexation_results.plot_best_matching_results_on_signal(
    dp, diff_lib, diff_gen, reciprocal_radius)

```

**Expected behavior**
The order in which the phases are added into the structure library should not matter. 
Also, the phase map shows that there should be two different phases in the area investigated, however there is no change when moving from one phase to another. I suspect this is an issue with plot_best_matching_results_on_signal(). Before Christmas this worked as intended, the changes I have made is how the rotation lists are made. 

**Screenshots**
Case 1.  Aluminium first, pictures on the right show the phase map and location.
```
struc_lib = StructureLibrary(['Aluminium', 'Betadp'],
                            [structure_Aluminium, structure_Betadp )],
                            [rot_list_Al, rot_list_Betadp])
```
[image]

Case 2.  Betadp first, pictures on the right show the phase map and location.
[image]

dnjohnstone, 2020-04-15:
Which version of the code are you using? I think this is fixed in the master

norskie, 2020-04-15:
Thank you for the quick reply. I was at version 0.10.1, I have now updated to version 0.11.0 and will attempt to run the code on this version.  I see that there has been quite a few changes while I have been in Japan! Good work! I will update this bug report when I am able to run my code again.

norskie, 2020-04-15:
@dnjohnstone Thank you again for the quick reply, after switching to v0.11.0 this was fixed. Closing the issue.

closed 2020-04-15 by norskie