pyxem/pyxem #597: .mib files fail to load

JoonatanL, 2020-05-11:
Loading .mib as suggested by the pyxem-demos fails due to a missing a 'chunks' parameter. I'd prefer to leave fixing this to someone more familiar with dask.

Relevant screenshot
[image]

hakonanes, 2020-05-11:
I would think `da.from_array(data_mem, chunks="auto")` would fix this. Or perhaps help dask find a better chunking to pass.

dnjohnstone, 2020-05-11:
Which version of pyxem and the demos are you using? The demos on the github only work with the current master (i.e. v0.11.0) not the current release version.

JoonatanL, 2020-05-11:
This is on 0.11.0 (should have included in the original error note)

`da.from_array(data_mem, chunks="auto")` does indeed seem to fix it. I can leave a pull request later.

closed 2020-05-18 by dnjohnstone