pyxem/pyxem #601: conda-forge packaging failing for pyxem-0.11.0 due to skimage deprecation

dnjohnstone, 2020-05-18:
The azure-pipeline on our conda-forge feedstock for 0.11.0 is failing https://github.com/conda-forge/pyxem-feedstock/pull/11

This (at the moment) is because skimage.feature.register_translation is deprecated in scikit-image-0.17.2 and we should use skimage.registration.phase_cross_correlation instead.

We missed this because travis is still on scikit-image-0.16.2 for some reason

I think that this is not the first time that we've had this kind of issue where we've not noticed a deprecation until conda-forge packaging, because travis has been passing, which is a real pain because we typically do it at the end of a release.

Given this I think we should do the following:

- [ ] pin the version of scikit-image to 0.16.2 for the pyxem-0.11.0 conda-forge release
- [ ] replace the deprecated skimage.feature.register_translation in the code
- [ ] following https://github.com/hyperspy/hyperspy/pull/2363 switch to azure pipelines for continuous integration on the pyxem repo
- [ ] release pyxem-0.11.1 with the above patches

@pc494 and @ericpre might have something to say about this - do you think this is the correct route?

pc494, 2020-05-18:
So I think some of this is `scikit-image` being difficult to work with, this feature was marked to be kept until there `0.19` version [1] so our chances of catching this were always low.

However, it would be a lot easier to find these kind of things if our test suite didn't spit so many warnings, so I think this should be bumped up the list of important things for (probably) `0.13`.

Happy to proceed as described for the patch release. Less excited by the prospect of rewriting the CI, but it's probably worth waiting for @ericpre's view here.

[1] https://github.com/scikit-image/scikit-image/blob/200ae1b41beb531beb529ac872314e120fbe6094/skimage/feature/__init__.py#L44

ericpre, 2020-05-20:
In case you want an another example of CI configuration: https://github.com/LumiSpy/lumispy/pull/35

In https://github.com/hyperspy/hyperspy, we end up using wheels packages from pypi for travis on linux only and conda packages on the 3 main platforms for various reasons.

tjof2, 2020-05-21:
>  it would be a lot easier to find these kind of things if our test suite didn't spit so many warnings, 

I looked at this not so long ago - a lot of the HyperSpy ones ought to go away with 1.6 when that is done.

dnjohnstone, 2020-07-21:
Ok - I think at this point we might skip a 0.11.1 release and jump straight to 0.12.0 since I think we want to get an initial release with the new pixSTEM stuff in done soon ish. @pc494 what do you think of that idea?

pc494, 2020-07-21:
Yep, moving straight to 0.12 is the correct move as far as I am concerned.

pc494, 2020-07-31:
_This (at the moment) is because skimage.feature.register_translation is deprecated in scikit-image-0.17.2 and we should use skimage.registration.phase_cross_correlation instead._ 

This is probably a 0.13 issue, with a minimum version requirement on skimage from then on?

dnjohnstone, 2020-08-13:
> _This (at the moment) is because skimage.feature.register_translation is deprecated in scikit-image-0.17.2 and we should use skimage.registration.phase_cross_correlation instead._
> 
> This is probably a 0.13 issue, with a minimum version requirement on skimage from then on?

Is this comment still relevant? If so, please can you make a new Issue about it as I think it is out of scope for this one, which I think is now resolved.

closed 2020-08-13 by dnjohnstone