pyxem/pyxem #606: pyxem-0.12.0

dnjohnstone, 2020-05-21:
pyxem-0.12.0 will be the first release of pyxem incorporating functionality previously in pixstem and introduced in #603 - the only aim of this release is to integrate functionality introduced in #603 to a useable level.

The following actions are required:

- [ ] Determine how we need to modify DiffractionVectors to accommodate dask peak finding (see #485 and discussion in #569)
- [ ] Integrate dask peak finding #604 
- [ ] Look at how to combine the radial integration functions currently duplicated.
- [ ] Tidy up the test suite, including reducing run time and returning coverage to 100%
- [ ] Check for unecessary duplication and look for opportunities to streamline code
- [ ] Update the demo notebooks to cover new functionality - most obviously DPC

Any comments, suggestions, contributions on this gladly received.

tjof2, 2020-05-22:
> Tidy up the test suite, including reducing run time and returning coverage to 100%

Some of the tests appear to launch plotting windows?

magnunor, 2020-06-04:
I already made some notebooks covering things like STEM-DPC for pixStem: https://gitlab.com/pixstem/pixstem_demos , they probably need some modifications.

I guess the issue is the use of a really big dataset?

I'll have some time from next week to contribute.

pc494, 2021-01-13:
Almost all of this was achieved, closing.

closed 2021-01-13 by pc494