pyxem/pyxem #623: Make use of hyperspy/ci-scripts

dnjohnstone, 2020-08-13:
HyperSpy now provides CI scripts - thanks to @ericpre 

https://github.com/hyperspy/ci-scripts

I think we should use them.  Does anyone object or have comments?

ericpre, 2020-08-13:
An alternative, worth considering, is to use github actions... It seems mature enough now, there are also nice "actions" to use from the github marketplace and it should be easier to setup, particularly the connection between CI provider and github. Also there are 20 concurrent simultaneous jobs for the open source free tier.

I have started to use it for https://github.com/hyperspy/hyperspy/pull/2473 and the hyperspy-bundle and it is less painful to use than others!

dnjohnstone, 2020-08-14:
@pc494 do you have an opinion here?

pc494, 2020-08-15:
I think we probably want both our CI and our Release to be github workflows. Then the pyxem version of the CI can be the "top level" one, and people can copy it across as they see fit (we can comment on the bits that need to be package specific)

pc494, 2021-01-13:
Closing this for now, possibly will revists later but after the migration to github-action in December 2020 that's going to be delayed

closed 2021-01-13 by pc494