pyxem/pyxem #628: Testing is really time consuming and needs to be fixed.

pc494, 2020-08-15:
We've noted this before, but having tried to do some development today it's clear that this is a real problem. I think cutting the size of the dummy datas used may be a good place to start. Then removing the figure openings. Will then profile and see what's left.

dnjohnstone, 2020-08-15:
See #606 - it's test_radial and test_dummy_data from the recent code merge that are the worst offenders.

pc494, 2020-08-15:
Okay, off the 900 seconds it takes to run the tests on my machine (with skips on the `test_dummy_data`) we have, as the top 3 offenders:

```
136.68s call     pyxem/tests/utils/test_radial.py::TestRadialModule::test_get_optimal_centre_position_off_centre_non_square
102.58s call     pyxem/tests/utils/test_radial.py::TestRadialModule::test_refine_signal_centre_position
43.30s call     pyxem/tests/utils/test_radial.py::TestFitEllipse::test_fit_single_ellipse_to_signal_rotation
```

Which is sufficient (at c.30%) for me to think that I shall be starting there.

magnunor, 2020-08-18:
Sorry about those very slow tests, they've been on my todo-list a long time. But unfortunately I haven't had the time to improve them.

Feel free to remove them for the time being, and for example copy-paste them into an Issue assigned to me. My aim is to rewrite parts of the radial module, utilizing the much nicer pyfai, which will include improving the tests.

pc494, 2020-08-18:
Thanks for this @magnunor, I'll do something sensible and open an appropriate issue with you assigned. 

I think you are right though, the radial bits of code seem to exist in 2 (maybe 3) forms, so it's likely some of these tests will be covering code that is going to be substantially changed.

dnjohnstone, 2020-08-18:
I think we can just remove the radial module altogether and reintroduce when integrating with the aximuthal integrator?

tjof2, 2020-08-19:
Why not mark the tests with a custom "slow" decorator for now, such that they don't run by default (e.g. local testing) but they can still run on (one of) the CI tests? https://docs.pytest.org/en/latest/example/simple.html#control-skipping-of-tests-according-to-command-line-option--

pc494, 2020-10-04:
This was dealt with in the short term by version 0.12.2 and in the long term will be dealt with by #632 ~ so I'm closing

closed 2020-10-04 by pc494