pyxem/pyxem #629: Documentation strategy for 1.0.0

pc494, 2020-08-18:
This is a broad catch all issue for deciding how best to document the code, with the hope of having something consistent agree and then implemented for 1.0.0 ~ it (obviously) relates to issue #624 - I'll leave my own thoughts to a comment.

pc494, 2020-08-18:
My order of importance runs:
1) accurate clear, docstrings on functions, methods and classes
2) up to date notebooks
3) a document (maybe a pdf dropped on the website) that tells users what kind of things they can do (without the need to run any code)

CSSFrancis, 2021-05-10:
I might have some time in the next couple of weeks to work on this (I assume that I am not fighting anyone for the job?)

Currently my goal is to:
- Change over the documentation to the Sphinx read the docs theme similar to hyperspy.
- Link the example notebooks using https://nbsphinx.readthedocs.io/en/0.7.1/ to provide easy working examples 
- Clean up some of the existing home page,
- Add in a page that automatically links any paper which cite the PyXEM DOI

The goal being that the documentation should require fairly minimal updating in the future as things change.  Most of the documentation should build automatically and with much of the recent development I think this is a good place to start.

hakonanes, 2021-05-10:
Just as a reference, we have this setup in kikuchipy with notebooks turned into reStructuredText by nbsphinx displayed via Readthedocs: https://github.com/pyxem/kikuchipy/doc and https://kikuchipy.org/en/latest/

pc494, 2021-05-13:
Thanks Carter, this seems like a good idea. If you need anything let me know.

ericpre, 2021-10-07:
#722 is adding the structure for the documentation and it is obviously a very good first step!
One thing which I find is really missing and is not being discussed here is a "narrative documentation" which could cover, for example:
- introduction/overview
- pyxem's design: what are the different type of pyxem's object, how they interact between each others, etc
- explanation about how to use pyxem

The notebook shows example of how to use pyxem but doesn't explain how to use it. A typical example is that it is difficult to find information about what is a pyxem `library` or a pyxem `generator`. Even if there was some explanation in the notebooks but it wouldn't be easy to find - even using the search function in the documentation.

ericpre, 2021-10-17:
> I agree, I would broadly call the existing pyxem-demos "examples", while a "narrative documentation" could be called "tutorials" or a "user guide". Examples of these distinctions are [scikit-image](https://scikit-image.org/docs/stable/auto_examples/index.html) and matplotlib's [examples](https://matplotlib.org/stable/gallery/index.html) and [tutorials](https://matplotlib.org/stable/tutorials/index.html).
> 
> We have such a narrative documentation in our [user guide](https://kikuchipy.org/en/stable/index.html) in kikuchipy. It takes a lot of work, so I would leave such work to other PRs after @CSSFrancis' #772 is in, which does the groundwork.

I suspect that using the wording "narrative documentation" is ambiguous and misleading. What I mean it is mostly a section which give an overview of the library and explanations - mostly conceptual and generic - without using code. This is more or less the opposite of examples or tutorials, which aims at applying and illustrating ideas or explanations given previously. See for example: https://matplotlib.org/3.5.0/users/explain.html.

> The important thing when writing such documentation is to keep it up to date. I think the fear of writing static documentation of which parts are outdated in the next release has put pyxem maintainers (@dnjohnstone, @pc494 ++) off of doing this before? The important thing then is to write the documentation as notebooks which can be automatically run and tested using tools like [nbsphinx](https://nbsphinx.readthedocs.io/en/0.8.7/) and [nbval](https://nbval.readthedocs.io/en/latest/). Any internal/external links in the documentation can also be checked using `make linkcheck` ([sphinx docs](https://www.sphinx-doc.org/en/master/usage/builders/index.html#sphinx.builders.linkcheck.CheckExternalLinksBuilder)) and updated manually.

The explanations I am referring above doesn't change very often and pyxem should have enough maturity so that its design and structure can be explained once and will not require significantly changes in a while. It will not be a "static documentation of which parts are outdated in the next release"
It doesn't need to be long and actually, I suspect that it would be a useful exercise for pyxem developer to *write down* pyxem design and structure. It would also help growing the contributor base by explaining the reasoning behind pyxem structure, etc. and how it integrates with other libraries.

> **But I honestly think the most important thing missing in the documentation is a complete and easily navigatable [API reference](https://kikuchipy.org/en/stable/reference.html) with all functionality covered.**

This is useful for people who are already familiar with the library or comfortable with python. For others, there is a need for more explanation, as done in https://kikuchipy.org/en/stable/index.html, for example. Typically, many users will not know what an "API reference" mean and focusing on an API reference as documentation strategy is not an inclusive approach in my opinion. Also, relying on demos is not good enough, because for most beginner users, it is not straightforward to re-use the "demos" to apply it to their experiments. Without additional source of explanation, it can be frustrating for the user to figure out why it doesn't work, etc.
In short: it is very difficult to learn something simply by reading definition and examples.  

An example of documentation, I have just came across, where you have a series of tutorials, examples along with some explanation sections: https://mne.tools/stable/overview/index.html

hakonanes, 2021-10-18:
> I suspect that using the wording "narrative documentation" is ambiguous and misleading.

I misunderstood you, sorry. I agree with all your comments above. I also still think that completing the module/function/class reference and displaying it via RtD is the easiest task to get done when documenting pyxem, and focus should be put there first. Then more fleshed out notebooks can point to this reference.

> An example of documentation, I have just came across, where you have a series of tutorials, examples along with some explanation sections: https://mne.tools/stable/overview/index.html

The MNE documentation is great: sections with a glossary, design philosophy, and implementation details are very useful, in addition to the examples, tutorials and API reference, of course. Thanks for the link.

pc494, 2021-11-10:
A boring, "easy" answer to this is that I will be writing a thesis sub-chapter that does almost exactly what @ericpre described (high level overview of the design of pyxem, answering questions like: what are generators?) that I can touch up for the website. There is a similar sub-chapter in Duncan's thesis which I would probably also link to

closed 2022-04-29 by pc494

hakonanes, 2022-04-29:
Now that a tagged release contains the Read the Docs documentation, I'm setting https://pyxem.readthedocs.io/en/stable to be the default when people visit https://pyxem.readthedocs.io (previously, `latest` was the only option).