pyxem/pyxem #634: Support for multi-file .MIB datasets

AndrewHerzing, 2020-08-28:
We typically save each frame of our 4D-STEM data in individual .MIB files.  There is currently no support for reading this type of data into PyXEM.

It should be fairly easy to do as the HDR/MIB file structures are practically the same as they are for the currently supported case of single MIB files.  Is anyone working on this?  If not, I can probably work up a rough version for a PR.

dnjohnstone, 2020-08-28:
I don't know of anyone working on this - but the key people are @M0hsend and @matkraj who had both been looking at this part of the code. Would be great if they could comment here.

AndrewHerzing, 2020-08-29:
Great.  I'll wait for their input.  In the meantime, we will typically do one of two things:

1.) Save a Digital Micrograph file with the Merlin data which provides the scan size, pixel size, and other metadata.  Use this DM3/DM4 file (which can be read directly by Hyperspy) to inform the reshaping of the Merlin data structure.

2.) In the absence of a DM3/DM4 file, we locate the pixels where the dwell time is greater than expected (usually by a factor of 2) in order to determine the width of the scan and then reshape the Merlin data based on this.

I have scripts which do all of this, but would be interested to hear from @M0hsend and @matkraj as to what they think the best path forward is.

pc494, 2020-11-16:
Hi @AndrewHerzing, this slightly passed me by when it originally came up. Can I be of any use, I've not written much io code yet, so it might be a code opportunity for me to get stuck in helping out.

AndrewHerzing, 2020-11-16:
I've written a standalone module for our data with the goal of making something ready for further processing with PyXEM.  If you want to have a look at what I have done and see if this might be of use to the PyXEM user-base as a whole, you can find it here:

https://github.com/AndrewHerzing/merlintools

If this does not look to be robust enough or is considered too much of a niche case, I am happy to keep this as a standalone module.

M0hsend, 2020-11-16:
Hi @AndrewHerzing , Sorry I just saw this conversation (not sure why I miss Github mentions...) The code we put together exactly does option 2 in your comment above. We read the exposure times [here](https://github.com/pyxem/pyxem/blob/71e7ca2b04f1543fc6baa7d560028f1ff9a75075/pyxem/utils/io_utils.py#L677) and then determine if we see an overexposed frame, i.e. flyback, using this [function](https://github.com/pyxem/pyxem/blob/71e7ca2b04f1543fc6baa7d560028f1ff9a75075/pyxem/utils/io_utils.py#L782). We moved away from using the HDR file, because it happens sometimes that you get the RAM throttle-back error and you do not reach the pre-set number of frames. So we use the header info of each frame within the MIB file and use the file size to determine the total number of frames acquired - See `_get_mib_depth`, [here](https://github.com/pyxem/pyxem/blob/71e7ca2b04f1543fc6baa7d560028f1ff9a75075/pyxem/utils/io_utils.py#L535).

M0hsend, 2020-11-16:
There is a notebook also showing more functionality with larger MIB files: https://github.com/pyxem/pyxem-demos/blob/master/00%20Importing%20large%20mib%20datasets.ipynb

AndrewHerzing, 2020-11-16:
As far as I can tell, this only works for data that is stored in a single MIB file.  Correct?  I should think it could be fairly easily extended for the case where each frame is stored in a separate MIB file.

M0hsend, 2020-11-16:
Yes, correct. That was on my TO-DO list at some point! Have not visited this because we see no reason to save multiple MIB files. Have you seen significant speed difference? Maybe you could open an h5 file and dump all individual MIB frames (or X number of frames per MIB) into it and once done reshape it (like what is now implemented for large scan arrays).

AndrewHerzing, 2020-11-16:
No, we haven't seen a speed difference.  The original motivation was to avoid having all of our data in a single large file, but I think this is less of a concern at this point.  I'll look into the h5 file option.  Unfortunately, the data coming off our system has some particularities that also make the load_mib function in pyXEM return a "Something has gone wrong with the triggering" message.  I'm sure this is a unique case for our data, so I'm not sure it warrants addressing withing pyXEM.

M0hsend, 2020-11-16:
This is coming from [here](https://github.com/pyxem/pyxem/blob/7ae40b7ef751438adf02a77bccf475a9734de3df/pyxem/utils/io_utils.py#L829) that we check that majority of the frame exposure times (>90%) corresponds to the triggered dwell time. Have you checked the histogram in `exp_times_list`in your case? We set this threshold - rather arbitrarily -  just to rule out any cases that triggering has failed for any reason.

AndrewHerzing, 2020-11-16:
Yes, this is definitely the problem and it is related to our somewhat flaky triggering signal.  If I run:

```
>>> exp_times_list = pxm.utils.io_utils._read_exposures(datafile, 100)
>>> times_set = set(exp_times_list)
>>> print(times_set)
{0.010268, 0.326011, 0.0, 0.001801, 0.001799, 0.001797, 0.000903, 0.000901, 0.000899, 0.000897, 0.00086, 0.000914, 0.003161, 0.0018, 0.000904, 0.0009, 0.000871, 0.000898, 0.000896, 0.000892}
```
The nominal frame time in this case is 0.0009 seconds, but a bunch of the exposure times are just slightly longer or slightly shorter than this.  Plus, in addition to the flyback pixels where the frame time is 2X the actual frame time, there are also some extra frames at the beginning and end of the scan.  So now, I get this:

```
>>> exp_time = max(times_set, key=exp_times_list.count)
>>> print(exp_time)
0.0009
>>> exp_times_list.count(exp_time)
163
>>> int(0.9 * len(exp_times_list))
367
>>> if exp_times_list.count(exp_time) < int(0.9 * len(exp_times_list)):
>>>      print("Something has gone wrong with the triggering!")
Something has gone wrong with the triggering!

```
I suppose a way around this would be to round the frame times to the nearest tenth of a millisecond like this:

```
>>> exp_times_list = pxm.utils.io_utils._read_exposures(datafile, 100)
>>> exp_times_list = np.round(exp_times_list, 4).tolist()
>>> times_set = set(exp_times_list)
>>> print(times_set)
{0.326, 0.0, 0.0009, 0.0018, 0.0032, 0.0103}

```
However, I'm not sure if this would cause issues for other users.

M0hsend, 2020-11-16:
That is interesting / weird... 
Here is what I get in a typical 256*256 dataset:
`times_set = set(exp_times)`
`print(times_set)`
{0.000416, 0.0455, 0.0338, 0.00065, 0.001704, 0.000425, 0.011707, 0.03445, 0.011855}
`for t in times_set:
    print('number of', t, 'cases:', test.count(t))`
for t in times_set:

    print('number of', t, 'cases:', test.count(t))

number of 0.000416 cases: 209
number of 0.0455 cases: 1
number of 0.0338 cases: 204
number of 0.00065 cases: 65019
number of 0.001704 cases: 52
number of 0.000425 cases: 1
number of 0.011707 cases: 1
number of 0.03445 cases: 48
number of 0.011855 cases: 1

The set exposure time was 0.00065 sec. The lower exposure times you see happen at the start and are skipped during reshaping. Given the large variation you see in your trigger line, I would raise this issue with QD. If people are after small intensity variation in their diffraction data this might cause issues. Happy to discuss this via e-mail if you think what we have set up could help you there in your lab.

pc494, 2020-11-17:
@AndrewHerzing 

The code in merlintools looks "clean" enough to be in pyxem as is, I think for now I'm going to mark this as "reconsider when pyxem is close to a release" (could easily be another 6 months) at which point we might have more of a sense of how common this workflow is likely to be.

AndrewHerzing, 2020-11-18:
That seems like a reasonable approach to me

pc494, 2023-01-16:
This is io (and old) so closing.

closed 2023-01-16 by pc494

AZangiabadi, 2024-03-12:
Hi @M0hsend 
I was going through tutorial 00 importing .mib:
`dp = pxm.load_mib(data_path)` 
and I receive this error. 

---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[3], line 1
----> 1 dp = pxm.load_mib(data_path)
AttributeError: module 'pyxem' has no attribute 'load_mib'

did something change?

CSSFrancis, 2024-03-12:
@AZangiabadi let me fix that tutorial! You should be able to use `hs.load()` to load the data now! 

Do you have a small quad dataset that you can maybe give to provide as an example?

M0hsend, 2024-03-13:
Hi @AZangiabadi all the file loading methods have now been moved to https://hyperspy.org/rosettasciio/ so we will need to update that example notebook!
@CSSFrancis I have uploaded an example quad MerlinEM mib file here: https://zenodo.org/records/10813049

AZangiabadi, 2024-03-14:
Thank you very much@CSSFrancis and @M0hsend. 
I did another test on a smaller area of polycrystalline gold sample and worked well. Larger files (>10 GB) was giving me insufficient Juypter memory error during curve fitting. 

Here is my raw data on Merlin detector (recorded on 10x10 area both at 256x256 pix and at 512x512 pix). 

Please feel free to upload this raw data to your tutorial. This can be used by anyone to learn and calibrate their diffraction pattern. 
https://we.tl/t-qAVbtc2U6L (valid for 7 days)
It was recorded on TFS Talos at camera length 160mm, 200kV, SA aperture 40um.

matkraj, 2024-03-14:
Is there any reason hyperspy/rosettasciio doesn't load Merlin data lazily as a default?
Merlin data is easily memory mapped so it should not be problem and in my opinion a preferred behaviour. The only one I see the memory mapping will not work is if the data was binary and RAW - 1-bit packed.

CSSFrancis, 2024-03-14:
> Hi all, Is there any reason hyperspy/rosettasciio doesn't load Merlin data lazily as a default? Merlin data is easily memory mapped so it should not be problem and in my opinion a preferred behaviour. The only one I see the memory mapping will not work is if the data was binary and RAW - 1-bit packed.

@matkraj Do we want to move this discussion to rosettasciio?