pyxem/pyxem #638: Alternative pattern matching metrics & approaches

dnjohnstone, 2020-09-04:
We want to explore alternative metrics and methods for pattern/template matching. There was an initial attempt at this in #608 but we have closed that for now since it didn't reach a conclusion. We may also consider pulling this out into a separate library that can serve both `kikuchipy` and `pyxem` following discussions with @pc494 and @hakonanes. This issue is a place to discuss what we should do and remember that it is on the roadmap.

tjof2, 2020-09-11:
Whatever the consensus is, it should support passing custom metrics/distance functions! 

Perhaps inspired by sklearns metrics https://scikit-learn.org/stable/modules/generated/sklearn.metrics.make_scorer.html as one idea.

hakonanes, 2020-09-22:
We're (@onatlandsmyr) currently looking into implementing comparison of EBSD patterns to simulations in kikuchipy, however, the circumstances are such that the extra work of setting up a new package doesn't fit into our plans. It's going to be implemented in its own module, like `kikuchipy.indexing`, so that those parts can be easily moved to another package when the time is right. The plan is also to use dask for all heavy lifting, which is the reason for not building off of the all-numpy stuff already in pyxem.

The scipy distance metric module looks very nice! Seems like the normalized cross-correlation is already there as the `1 - "cosine"` metric, which is what we're most interested in. One draw back, or at least something that needs to be handled with care when specifying dask chunks, is the `scipy.spatial.distance.cdist/pdist` requirement for patterns and templates to have a 1D signal axis. The IO functions reading these should be able to return 1D signal axes when they're to be used for this purpose.

I'm not familiar with sklearns `make_scorer`, so I couldn't tell how difficult it would be to use that with dask arrays, but the idea of an easy way to create a custom metric is intriguing.

tjof2, 2020-09-27:
> I'm not familiar with sklearns make_scorer

I mean it as a way of **standardizing** the way to add/define your own metrics. e.g. it has to meet certain requirements, follow a certain pattern, and has the easy "greater is better"/"smaller is better" flag.

pc494, 2020-10-08:
On a related note, I'm going to drop the content of #612 into this thread.

"Currently we have no method for comparing a simulated list of vectors and an experimental measured list of vectors in a manner consistent with conventional template matching apporaches (ie- not vector based indexation), such functionality would be a meaningful improvement to the our orientation/phase mapping workflows."

onatlandsmyr, 2020-10-08:
I have started on a framework, under kikuchipy.indexation in [my fork](https://github.com/onatlandsmyr/kikuchipy/tree/template-matching), for comparing similarity between 2D images of equal size(is this called full-reference?). My framework allows for user-defined metrics with a function _make_similarity_metric_ which takes the following parameters:
- _metric_func_: computes similarity/distance matrix between (experimental) pattern(s) and (simulated) template(s)
- _greater_is_better_: True/False used for choosing n largest metric results
- _flat:_ True/False, 2D images is flatten before sent to metric_func
- _scope_: describes how metric_func's input is intended to be, many templates to many patterns, one to many, many to one, any to one, and so on.
- _make_compatible_to_lower_scopes_: If True pads the inputs with single dimensions to match the given scope.

I have implemented ZNCC and NDP(only difference is to not subtract the mean as far as I know) using this framework with inputs being dask arrays.

pc494, 2020-10-08:
Hi @onatlandsmyr, this sounds like an excellent start. My only thought is that in general you will always want to flatten 2D images at some stage (for speed reasons)

hakonanes, 2020-10-08:
> Hi @onatlandsmyr, this sounds like an excellent start. My only thought is that in general you will always want to flatten 2D images at some stage (for speed reasons)

I'm not so sure about this, e.g. when they are already in memory as 2D images, I think Dask might prefer not reshaping before processing?

hakonanes, 2020-10-24:
So based on discussions (off GitHub) yesterday, we agreed to not set up this "matching" package now, but wait for when other work has progressed in pyxem and the design of the `kikuchipy.indexing` module has "settled" a bit.

hakonanes, 2020-11-22:
the paper referenced in this comment https://github.com/pyxem/diffsims/pull/137#issuecomment-731808228 is relevant for the potential "matching/indexing" package, i think, since, as morawiec emphasises in the paper, the indexing procedure is "technique agnostic". note that the code for the software morawiec have developed/improved as part of the publication is not available (kikoch2), however the command line program is (.exe file).

pc494, 2020-11-24:
As a plug and play template matching option, rank refinement seems like an excellent candidate:

https://doi.org/10.1107/S0108767312007234

pc494, 2023-01-16:
To open-ended (and old), closing.

closed 2023-01-16 by pc494