pyxem/pyxem #641: Error message when Diffraction2D.ai is not set

magnunor, 2020-09-21:
When running `s.get_azimuthal_integral1d` for a `Diffraction2D` signal without having set the Azimuthal Integrator property, the error message is a bit confusing:

```python
pyxem/signals/diffraction2d.py in get_azimuthal_integral1d(self, npt, mask, radial_range, azimuth_range, wavelength, unit, inplace, method, sum, **kwargs)
    400         sig_shape = self.axes_manager.signal_shape
    401         if radial_range is None:
--> 402             radial_range = _get_radial_extent(ai=self.ai, shape=sig_shape, unit=unit)
    403             radial_range[0] = 0
    404 

pyxem/utils/pyfai_utils.py in _get_radial_extent(ai, shape, unit)
     75         The unit to calculate the radial extent with.
     76     """
---> 77     postions = ai.array_from_unit(shape=shape, unit=unit, typ="center")
     78     return [np.min(postions), np.max(postions)]
     79 

AttributeError: 'NoneType' object has no attribute 'array_from_unit'
```

There should rather be an error message saying the `s.ai` can be set using the `s.set_ai` method.

pc494, 2020-09-21:
Think this happens in a couple of places, I would probably add a check at the top of the functions which spits out a good error message if `self.ai` isn't set

pc494, 2020-12-13:
Actually, turns out this is an easy fix, which I can do:

```python

 @property
    def ai(self):
        try:
            return self.metadata.Signal["ai"]
        except (AttributeError):
            return None
```
Just needs to raise a clear `ValueError`

PR #676 (2020-12-15): Tidying up the class arrangement for "ai"
---
name: Tidying up the class arrangement for "ai"
about: Adapt the `@property` of ai to be more pythonic

---

**Checklist**
- [x] Updated CHANGELOG.md
- [ ] (if finished) Requested a review (from pc494 if you are unsure who is most suitable)

**What does this PR do? Please describe and/or link to an open issue.**
- fixes #641 by changing the `except` block for `dp.ai` when `ai` has not been set 
- tackles #642 by removing the `unit` kwarg

**Describe alternatives you've considered**
This section of the code still needs quiet a lot of work, but this  [... cut]
closed 2020-12-15 by dnjohnstone