pyxem/pyxem #642: Diffraction2D.get_azimuthal_integral1d: unused unit parameter?

magnunor, 2020-09-21:
When using `Diffraction2D.get_azimuthal_integral1d`, the `unit` parameter is not utilized, as it is overwritten in line 399 (https://github.com/pyxem/pyxem/blob/master/pyxem/signals/diffraction2d.py#L399).

This leads to an error in the processing, if the signal's units haven't been set.

pc494, 2020-09-21:
Do you think we should keep the argument and overwrite `self.unit`?

CSSFrancis, 2020-09-21:
We could just get rid of self.unit but it might lead to the case where we have the units in 2 different places. 1 in the axes manager and 2 in the parameters.  It's really just trying to balance the axes_manager set up from hyperspy and the .poni calibration for pyfai.

In doing so I might have made everything overly complicated.  I'm all for removing self.unit or just overwriting it.

pc494, 2020-09-22:
I'm in favour of overwritting with a warning telling the user that this is happening

pc494, 2020-12-13:
I'll address this for 0.13.0

pc494, 2020-12-16:
#676 fixed this by making unit no longer a kwarg

closed 2020-12-16 by pc494