pyxem/pyxem #643: Lazy processing of azimuthal integration

magnunor, 2020-09-21:
A common use case for my large datasets, is doing azimuthal integration (get intensity as function of scattering angle). This consist of two steps:

1. Find the centre of the diffraction pattern using (for example) centre of mass, since the centre can move around quite a bit as a function of probe position
2. Do the azimuthal integration around these centre positions

This must be done without the whole dataset being loaded into memory at the same time, since it can be much bigger than the available computer memory. This does currently not work with `Diffraction2D.get_azimuthal_integral1d`, as i) it does not support processing of lazy signals, ii) `Diffraction2D.center_direct_beam` does not support lazy processing.

It would be fairly easy to add a dask version of both these functions directly to pyxem, however, another possibility could be to improve HyperSpy's lazy processing functionality.

- pyxem dask implementation: would be much quicker, due to it being more specific
- HyperSpy lazy improvement: would take a lot more time, as it needs to be implemented in a fairly "general" fashion. However, it would be useful for a much greater range of functionalities in pyxem (and elsewhere).

Any thoughts?

hakonanes, 2020-09-21:
I haven't looked at or used the azimuthal integration. However, we're using dask for all EBSD pattern processing in kikuchipy, so I'm interested in this discussion.

An argument for implementing dask routines in pyxem, is that chunk alignment can be tailored to the operations pyxem want to do, instead of having to rely on some upstream HyperSpy routine which tries to guess the best chunking?

A (rather difficult?) question: How would HyperSpy's chunk iterating functionality (https://github.com/hyperspy/hyperspy/blob/RELEASE_next_minor/hyperspy/_signals/lazy.py#L615 and similar) be improved?

pc494, 2020-09-21:
From a functionality point of view, the centering + radial integrating is so prevalent in our workflows that having a solution for 0.13.0 would be great. This probably means writing it ourselves in the short term.

In general, scoping out a more comprehensive lazy signal process in hyperspy seems like a good idea, even if it turns out to not be feasible.

CSSFrancis, 2020-09-21:
@magnunor I realize that my work with lazy data sets is rather naive, but I don't really know why the current implementation doesn't work from a lazy standpoint.  My understanding is that hyperspy.map function should handle the dask implementation.  I've tried running small data sets through the integration and it seems fine.

Now the hyperspy.map function could use a little bit of work.  In the past I've tried to written a function that allows an "axes" keyword to allow for iterate over the navigation axes rather than the signal ones.  I've also written a map function that automatically changes the axes labels but it's kind of rough around the edges.

I'd also argue that the map function doesn't operate in parallel well mostly because it spends way too much time on the overhead of distributing the tasks rather than doing it efficiently. 

For the most part I would vote for trying to rewrite the map function, but it might be good to first outline exactly the issues with the function.

magnunor, 2020-09-21:
@CSSFrancis, the current main limitation is the ability to get a `dask` array as a return object, and the lack of control on exactly how the different dask chunks are handled. For example, if our (256, 256, 256, 256) dataset is stored with chunking (32, 32, 32, 32), we ideally want to retain as much of this chunk structure as possible.

Lets say we couple of pixels shifting of the diffraction pattern, we want to fix dead pixels, centre it the diffraction pattern and extract the intensity from one specific diffraction spot. First we could do dead pixel correction, while retaining the chunks (while getting the border of the neighbouring chunk). With this, we could run centre of mass on the central spot. Lets say it is contained within the (64, 64) central pixels, we would then only process the 6.25 % of the total dataset size to get this shift. This could then be followed by using dask.array.roll to shift the diffraction pattern to the centre. This would lead to a small breakup of the chunks, but only to its neighbouring chunks. Then we could then extract the intensity from one of the diffraction spots, which would require accessing only a (64, 64) "signal" chunk.

I'm not sure if it is possible to replicate such a workflow with the `map` functionality.

@hakonanes, yeah, I think the biggest reason for doing a pyxem specific implementation of this, is the ability to keep the "chunks" under control. It is possible to disable HyperSpy's rechunking though, so that specific issue can be avoided.

@pc494, I'm tempted to start with the pyxem specific one, and try to make it fairly general. Then in the future see if the more general parts can be "upstreamed" to HyperSpy. I think I'll start with the azimuthal integration since I already have some dask-ified code to do that.

magnunor, 2021-01-07:
For future reference, this was implemented in a number of pull requests directly in pyXem in `dask_tools`:

- `get_direct_beam_position`: https://github.com/pyxem/pyxem/pull/648
- `center_direct_beam`: https://github.com/pyxem/pyxem/pull/658
- `get_azimuthal_integral1d`: https://github.com/pyxem/pyxem/pull/668

This was done with a new function in `utils.dask_tools`: https://github.com/pyxem/pyxem/blob/master/pyxem/utils/dask_tools.py#L196

The next step will be to extend this to the other `get_azimuthal_...` and `get_radial_...` methods in `Diffraction2D`.

pc494, 2021-04-15:
I think #759 is going to sweep this up too? Can you confirm @magnunor?

magnunor, 2021-04-15:
@pc494, it will mostly be covered in https://github.com/pyxem/pyxem/issues/759. However, it will require some minor changes to many of the methods in, for example adding a `lazy_result=False/True` flag.

That being said, feel free to close this one, as the scope of these changes will be much smaller compared to the original plan.

pc494, 2021-04-15:
Thanks, I think I'll close this but bear it in mind when we start tidying up those changes for 0.13.4

closed 2021-04-15 by pc494