pyxem/pyxem #650: python 3.9 is here! We need to test for it

pc494, 2020-10-09:
I think in the short term we will keep testing for version `3.7` as well, as it's only 13 months since it was the most recent version of python.

pc494, 2020-12-02:
skimage has been dealt with now, updating title accordingly.

hakonanes, 2020-12-02:
Thanks for keeping track of 3.9-compliant packages. kikuchipy has more or less the same dependencies as pyxem, so if it's okay with you I'll follow your progress here and adapt that package accordingly when it's time.

pc494, 2021-01-04:
numpy has now got 3.9 all sorted, so I'm going to try and make that work for 0.13.0

ericpre, 2021-01-05:
numba doesn't support python 3.9.

pc494, 2021-01-05:
> numba doesn't support python 3.9.

Okay, so back to the other plan, 3.8 and 3.7 for our 0.13.0

closed 2021-01-05 by pc494