pyxem/pyxem #660: Moving beyond #654's improvement to the orientation mapping workflow

pc494, 2020-11-20:
#654 leaves quite a lot of things open, but the solution is basically just one operation. The current vector matching spits out an object with all the right things but in a slightly inconsistent shape. To fix this one needs to 

(a) pull out so that only phase,alpha,beta,gamma,score are stored in the `.data` attribute of the `MatchingResult`
(b) push all the other properties that are calculated into a new `.prop` attribute of the `MatchingResult`

pc494, 2020-12-21:
This actually looks like it could be really difficult, and I don't really trust much of the code that underlies it. Given that we are expecting to rewrite almost every component of Vector Based matching within the next year I would like to either

a) Pull everything out so that we can restart
b) Put suitable labels on the code so that users know that it's a worse state than the average bit of pyxem and plan to rebuild soon-ish (with some reversions to fix the testing etc)

@dnjohnstone thoughts?

dnjohnstone, 2020-12-22:
@pc494 I think we all agree that the vector matching bit of code needs a "hard stare" and to be totally re-written. One reason it hasn't happened yet is because it needs a fairly fundamental re-think of how vectors are handled, as started in #485 .

I suspect that few people, if any, are actually using that code as it is and it sounds like it's going to be broken so I'm not opposed to pulling it out in principle. But I'd like to find a way to record what has been taken out and a rough plan for putting it back e.g. a new draft PR - perhaps one for us to pick up in a call after the break for the holidays?

pc494, 2020-12-23:
I am going to revisit this in the new year.

pc494, 2021-01-05:
I'm closing this as #699 addresses it adequately for the time being.

closed 2021-01-05 by pc494