pyxem/pyxem #663: Progress on ASTAR-like indexation

din14970, 2020-11-24:
Not sure if this is the best place for it, but I thought I would share an update on ASTAR like template matching as I discuss here: 
* https://github.com/pyxem/diffsims/issues/124. 
* https://github.com/pyxem/diffsims/pull/137

The idea is as follows: 
1. I convert the image and template to polar images. 
2. I calculate the cross correlation on these images via the fourrier transform, after the method in scikit image
3. I get a radial fit profile
4. I get the maximum in this profile. This gives me simultaneously the best in-plane angle for that template and a measure of correlation.
5. I repeat for all templates, which gives me a measure of correlation for each
6. I can plot "quality maps"

To demonstrate, here we start with this diffraction pattern. Note, intensities in the image are plot on a logarithmic scale to enhance weak reflections.
[image]
I calculate where the direct beam is. Currently I just do this simply with the maximum; I am aware there is a function for this in pyxem. Using the `warp_polar` method from sci-kit image centered on the direct beam I get
[image]
I create an empty image of the same dimensions as the polar image and fill in the intensities in the correct pixels (may be hard to see but there are a couple bright pixels in there)
[image]
I calculate the cross correlation with
```
fft_im = fft.fftn(polar_image)
fft_temp = fft.fftn(polar_template_image)
cross_correlation = np.abs(fft.ifftn(fft_im*fft_temp.conj()))
```
This should give me a measure of how well the template and image line up in polar space. We only care about the y (=theta) direction, so I would guess `cross_correlation[:, 0]` should give me a "fitness" profile describing the match at each angle:
[image]
The maximum along this line should yield a best fit angle, and some measure for correlation fitness. Applying this angle on this particular pattern (blue is unrotated, orange is rotated): 
[image]
I can run this over all templates in a library for one diffraction pattern, which gives me a best angle and correlation value for each template. I can plot this correlation value in a stereo plot:
[image]
or using cubic interpolation:
[image]
Comparing one diffraction pattern to all templates this way takes 4 seconds on my average laptop. That is unfortunately not too great, for my 256x256 scan it would take 72 hours or so to index. 
I can find the maximum correlation value and use that template and angle:
[image]

Things I still need to look into: 
* theres a couple sign errors and places where I've gotten confused about x, y, rows, columns, ...
* I need to think a bit more about the math of this cross correlation and how to properly normalize the templates. The "correlation plot" is definitely not looking correct. But if implemented correctly I think this should yield something equivalent to the way the correlation is now calculated between templates and image, except the best alignment angle is also found.
* how can this be made faster and more efficient? My guess is the cross-correlation/fft step is the slowest step, and since we only need to shift the image along the theta direction maybe a custom function accelerated with numba could help. Maybe the sampling in the theta and/or r direction can be reduced, and then an optimal angle found with interpolation. 

That's about the progress so far. Currently most things are in loose files and jupyter notebooks. The way I was thinking about adding it to pyxem was adding an additional argument in `IndexationGenerator.correlate`, something like `optimize_in_plane_rotation`. The thing is that with this method, the `method` and `n_largest` don't really make so much sense anymore. But perhaps this warrants a more detailed discussion.

troubadour-j, 2020-11-24:
Very nice work!! Thanks a lot. As a user, I have some questions: 1) Can we use 4k images for reconstruction of orientation map? 2) How can we calibrate the camera length? 3) Can we use more fine grid for template matching?

din14970, 2020-11-24:
@troubadour-j 
1) Yes, in principle there is no limitation on the size of the images or their depth, but taking FFTs of those images will take longer. 
2) You don't calibrate camera length as this is quite meaningless, you calibrate the pixel scale. There are examples here https://github.com/pyxem/pyxem-demos, see especially https://github.com/pyxem/pyxem-demos/blob/master/02%20GaAs%20Nanowire%20-%20Phase%20Mapping%20-%20Orientation%20Mapping.ipynb. You do have to have some idea of this scale a-priori and set it, as far as I know there is no "camera length optimization" function.
3) Yes, you can make the grid as fine as you want. With the function `get_beam_directions_grid` you can generate a grid of orientations and specify the maximum angle between orientations.

din14970, 2020-11-25:
Minor progress update. Dividing the correlation with the norm of the intensity vector gives me a reasonable looking correlation map:
[image]

I've been experimenting with the way to make the in-plane rotation step faster, but have not beat the cross-correlation method so far. I wrote a completely "manual" function whereby the template is rotated stepwise in a loop, and I only extract the relevant values from the image. Accelerated with numba this gave me again about 4 seconds (to calculate the image above) when using a step of 1 degree. Without numba it takes 30 seconds.

din14970, 2020-11-26:
I was able to reduce the time required to index one pattern to 0.5 seconds using numba. This is still not fast by any means but is getting into acceptable range territory.

I threw together an example here:
https://github.com/din14970/indexing_demo

For the moment it can only be run when using my diffsims and orix branches that contain extra bits of code.

pc494, 2020-11-27:
Sorry @din14970, I've been a bit absent on this one. It looks like the code you've got has all the right ideas. My instinct would be that FFT based cross correlation is overkill (in terms of the accuracy of our results) and if "manual" is quicker we should do that.

If you agree, I think the next step then boils down to optimising the code under "**Helper functions for transformations**" in the notebook? Would you want some input on that from me?

din14970, 2020-11-27:
@pc494 Yep FFT is overkill and unnecessary for the super sparse templates. I thought minimal code with canned scipy and scikit-image functions would work nicely but it's just slower unfortunately. 

Note that the first bit of the notebook I made as a demo for someone else who would like to use diffsims.

The "manual" functions I've implemented in the notebook look super ugly and unpythonic but after some iteration I surrendered to what numba demanded - that is try to put as much as possible into numpy arrays, totally avoid lists, prealocate variables. I'm pretty new to numba so maybe you or others can still find more optimizations, I'm sure there's still things that can be done. So for sure, would like to see your input on it. I'm now trying to see whether I can cut down using e.g. Wu's method of comparing integrated radial profiles. 

The downside of the numba thing is that it doesn't like the diffsims object abstractions. So I doubt this implementation will play nicely with the existing template indexers, especially if I want to then parallelize it to index multiple experimental patterns simultaneously.

pc494, 2020-11-27:
I will have a look into, my biggest takeaway from working on similar problems in this space is to do everything you can before you combine template and image.

For example, you can radially integrate the image before you expose it to the templates and you can get radial values for templates in advance of the images. 

For a Wu like workflow this would probably mean 

(1) creating a radial profile a library generation time (which should be quick)
(2) integrating images (which might be slow) only once (and not every-time you have a new template)

At a first approximation this is the difference between doing M+N and M*N operation (M templates, N images).

Some of this boils down to how you nest lists, some of it requires some higher level design considerations.

Also, once this work is all merged in, that notebook would be great to have as a diffsims demo.

pc494, 2020-11-27:
Also, yes, my experience with numba also includes realising you have to play the game it wants you to play, for better or worse.

pc494, 2020-11-30:
So I've been looking at this:

```python
@njit 
def match_polar_to_template(polar_image, xy_coordinates, intensity, d_r):
    d_theta = 360/polar_image.shape[0]
    polar_coords = _cartesian_to_polar(xy_coordinates)
    r_template = np.rint(polar_coords[:, 0]/d_r)
    theta_template = np.rint(polar_coords[:, 1]/d_theta)
    condition = r_template < polar_image.shape[1]
    r_template = r_template[condition].astype(np.uint16)
    theta_template = theta_template[condition]
    intensity = intensity[condition]
    correlation = np.zeros(polar_image.shape[0])
    for i in prange(polar_image.shape[0]):
        theta_compare = np.mod(theta_template + i, polar_image.shape[0]).astype(np.uint16)
        image_intensities = np.zeros_like(theta_compare, dtype=np.float64)
        for j in range(theta_compare.shape[0]):
            yy = theta_compare[j]
            xx = r_template[j]
            image_intensities[j] = polar_image[yy, xx]
        correlation[i] = np.sum(image_intensities * intensity)
    return correlation
```

It seem like a lot of this could be pulled outside the loop, plus, I'm not sure you get a speed saving by converting to radial coordinates. I think in an ideal workflow you would go:

```
create sphere mesh library and simulate, radial integrate those patterns
radially integrate experimental data (probably under the hood)
match these, discard all but the best 10 or similar
find the theta coordinates for those (use the same matching process and rotation matricies)
```

If this saved us a a factor of 100 in speed (with no messing about with numba) would the per pattern speed be quick enough?

din14970, 2020-11-30:
Thanks for having a look.

> I'm not sure you get a speed saving by converting to radial coordinates to then match pixel by pixel, you could (and I would) just spin x,y with a rotation matrix.

I implemented it first in a similar way as you suggest and this was significantly slower. Rotating the x-y coordinates I need on every iteration in the main loop to:

1. center the template peaks to 0
2. construct a rotation matrix
3. multiply all coordinates by the matrix
4. shift the peaks again to the central beam
5. round coordinates and convert to integers
6. extract image intensities

With this implementation I can do most things outside the loop an all I have to do inside the loop is increment the theta coordinate. The polar transforms have to happen only once and could be done even before the entire indexation procedure. I don't have to do any rounding because I am only adding integers. So the loop is reduced to:

1. add scalar int to 1D array (theta)
2. extract image intensities

However, now that I want to index large datasets of multiple images I am running into the problem of where to best fit in these polar transforms. The function doesn't work well with numba and it complicates things with lazy signals. So I might give it another go without the polar transform and see if it can be further optimized if this simplifies the rest of the process.

> At present I think that my ideal workflow you would go:
> create sphere mesh library and simulate, extract r values with np.sqrt(x**2+y**2) 
> radially integrate experimental data (probably under the hood) creating a flat array
> match these, discard all but the best 10 or similar, probably by indexing the dictionary
> find the theta coordinates for those (use the same matching process and rotation matricies)

I actually implemented this but haven't pushed to the example repo. The function basically looks like:

```python
@njit
def match_library_to_polar_fast(polar_image, coordinates, intensities):
    polar_sum = np.sum(polar_image, axis=0).astype(np.float64)
    polar_sum = polar_sum / np.linalg.norm(polar_sum)
    coors = np.zeros(intensities.shape[1], dtype=np.float64)
    N = polar_sum.shape[0]
    for i in range(intensities.shape[1]):
        intensity = intensities[:, i]
        xy = coordinates[:, :, i]
        r_template = np.sqrt(xy[:,0]**2 + xy[:, 1]**2)
        condition = r_template < polar_image.shape[1]
        r_template = r_template[condition].astype(np.uint16)
        intensity = intensity[condition]
        image_row = np.zeros(polar_image.shape[1], dtype=np.float64)
        for j in range(intensity.shape[0]):
            inten = intensity[j]
            r_p = r_template[j]
            image_row[r_p]+=inten
        sumx = np.sum(image_row)
        sumy = np.sum(polar_sum)
        coors[i] = correlation_function(image_row, polar_sum)
    return coors
```
This works fine and is super fast, 4 ms to index a pattern, compared to 500 ms in the full indexation. 
Unfortunately, I came to the conclusion that radially integrated matching does not guarantee a good match, at least for the dataset I'm looking at. The pole figure looks like:
[image]
compared to the full indexation:
[image]
You see that the maximum is in a totally different place, so the top 10 candidates will not suffice for the full indexation. After a bit of experimentation I found that I needed to keep about 30% of the templates to retain the optimal match from the full indexation in the dataset:
[image]
When I filter these out and then apply the full matching, I get a speedup in total to about 250 ms for 1 DP, so a factor of 2. Everything is numba jitted.

So I'm starting to think there will not be free lunch. We can add multiple types of indexation and let the user decide, but at the moment I'm inclined to go with the numba mess unless I find a better alternative with the same performance and accuracy. The speed should then come from parallelizing over the dataset, although I still believe time can be shaved off by using e.g. `guvectorize`.

My main challenge at the moment is how I should best apply the computation across the dataset and build in parallelization. My dataset does not fit into memory so I have to use dask and lazy signals, but I'm not so familiar with these.

pc494, 2020-12-01:
> Thanks for having a look.
>
>
> I implemented it first in a similar way as you suggest and this was significantly slower. Rotating the x-y coordinates I need on every iteration in the main loop to: [...]
Could you explain this step to me? 
> 
>     2. construct a rotation matrix
> 
>     3. multiply all coordinates by the matrix
> 
>     4. shift the peaks again to the central beam
And this one?
> 5. round coordinates and convert to integers
>
>     6. extract image intensities
>
>
> With this implementation I can do most things outside the loop an all I have to do inside the loop is increment the [...]

I suppose this is why one is taught to profile code rather than guess what will be faster. Fair enough.
 > This works fine and is super fast, 4 ms to index a pattern, compared to 500 ms in the full indexation.
> Unfortunately, I came to the conclusion that radially integrated matching does not guarantee a good match, at least for the dataset I'm looking at. The pole figure looks like:
> [image]
> compared to [...]

So this is a science question as far as I am concerned. If we can make the radial match accurate enough, we've got our speed up, and if we can't, I think this routine is going to be slow.

> When I filter these out and then apply the full matching, I get a speedup in total to about 250 ms for 1 DP, so a factor of 2. Everything is numba jitted.
>
> So I'm starting to think there will not be [...]

I think my main summary thought would be that if we think the time taken for a complete indexation is 

~ M x N x T

where M is the number of patterns, N is the number of templates and T is the time to compare one template to one pattern, we are going to really struggle to reduce T, in the current implementation it's two lines of numpy operation, and numpy is kinda fast.

Which is why I've been pushing cutting N down, but if that gives us "bad" answers, it's not really worth it. That said, if we could push it down towards 25ms (~16 patterns a second), we could then index something like a 200 by 200 in under half an hour, which I would deem "good enough".

din14970, 2020-12-01:
> Could you explain this step to me?

Forget this step, I mean the central beam of the template should be centered on (0, 0) before rotation. However, I'm using the `calibrated_coordinates` stored in the simulation objects, which already have this property.

> And this one?

To extract the right pixels, the central beam of the template should coincide with the central beam of the pattern.

> I think my main summary thought would be that if we think the time taken for a complete indexation is
> ~ M x N x T

Yes, if T also includes the rotation step (or shift in polar coords), otherwise it would be rather M x N x R x T. The radial integration takes out R basically and adds a bit of overhead, but it's essentially 100 times faster. The question is if one thinks holistically, whether the (N x R x T) ensemble can be sped up. I think there's still some wiggle room by using numba in a smarter way. The combined approach of radial integration + full indexation cuts N by 70% but adds some overhead; with this approach I would still be looking at 4.5 hours to index my 250x250 dataset on my laptop. 

However, one should also consider parallelism, which if we simply create threads for indexing each DP, we could still on a regular laptop with 4 virtual cores cut to maybe 1-1.5 hour. Indexing on a workstation with multiple cores, say 16-32, would become very doable. One might also consider revisiting (N x R x T) at some point and scale these operations to the GPU. This is where I could see a massive speed-up as all these comparisons can be perfectly parallelized but don't warrant separate CPU threads IMO.

My feeling is that Astar does the combined approach and indeed selects only the top 10 or so candidates for a full indexation. This may be (comparatively) blazingly fast, but does not ensure that the "best" indexation is in the dataset. Astar also provides a "full indexation" option, which I guess is the approach without radial integration - my colleague tried this and it took a full day plus some to index his dataset (granted he also used a very large template library, so it wasn't really a fair comparison). So what is fast may not be exactly right, and what is right will probably always be quite slow. Unless we make full use of parallelization.

I would go with implementing both slow and fast and let users select what they want. In the combined approach I have a parameter to tune which fraction of templates you want to keep for a full indexation. Then I would try to make sure to parallelize as much as the available hardware allows.

pc494, 2020-12-01:
> Forget this step, I mean the central beam of the template should be centered on (0, 0) before rotation. However, I'm using the `calibrated_coordinates` stored in the simulation objects, which [...]

Just closing this bit of conversation quickly, although it's poorly documented, the template matching workflow so far has always assumed that the input diffraction patterns have been centred in advance. Will look over the rest of the discussion shortly.

pc494, 2020-12-01:
Also, before you rebuild too much, be aware that `.map()` from hyperspy has a `parallel=True` kwarg that might speed things up in a similar fashion to the one you have planned.

din14970, 2020-12-01:
> Also, before you rebuild too much, be aware that .map() from hyperspy has a parallel=True kwarg that might speed things up in a similar fashion to the one you have planned.

Yes initially I was looking into whether I could use this one. However, with the experiments I've tried with lazy data I get an extremely complicated execution graph that takes forever - i was not able to get a solution and I also don't see any progress. I'm experimenting with the lower level dask stuff to see whether fine-grained control can improve it. As long as the user facing functions take in and produce hyperspy/pyxem objects it should be fine.

hakonanes, 2020-12-01:
> I'm experimenting with the lower level dask stuff to see whether fine-grained control can improve it. As long as the user facing functions take in and produce hyperspy/pyxem objects it should be fine.

For this I assume you're looking a bit at @magnunor's [dask tools in pyxem](https://github.com/pyxem/pyxem/blob/master/pyxem/utils/dask_tools.py#L97)? We also don't use HyperSpy's `map()` function in kikuchipy, but dask's `map_blocks()` to operate on chunks (e.g. https://github.com/pyxem/kikuchipy/blob/master/kikuchipy/signals/ebsd.py#L824 which calls https://github.com/pyxem/kikuchipy/blob/master/kikuchipy/pattern/chunk.py#L344 per chunk which calls https://github.com/pyxem/kikuchipy/blob/master/kikuchipy/pattern/_pattern.py#L304 per EBSD pattern).

magnunor, 2020-12-01:
With regards to the `dask` and `lazy` processing, there are some functions for processing large datasets in https://github.com/pyxem/pyxem/blob/master/pyxem/utils/dask_tools.py#L138. However, it hasn't been extended to template matching. Me and @hakonanes experimented a little with this a couple of weeks ago, but I haven't had time to work in it. The focus has been on getting the azimuthal integration pipeline working lazily.

The main issue will be how to pass the library to each diffraction chunk... How large is typically the library?

din14970, 2020-12-01:
@hakonanes thanks for the links, will look into it for inspiration

@magnunor the way I currently do it (open to criticism) for compatibility with my numba functions is I convert the templates to coordinate and intensity arrays of shapes [R, 2, N] and [R, N] respectively, with R the maximum number of reflections in all templates (in templates with fewer reflections I pad with 0's) and N is the number of templates, typically ~1500.

pc494, 2020-12-01:
Just as an extra thought, I wrote (I'll admit a very dumb) module https://github.com/pyxem/pyxem/blob/master/pyxem/utils/big_data_utils.py that exploits the fact that template matching reduces the size of an object a lot (from a diffraction pattern to an orientation triplet or similar) and is probably the simplest to integrate into existing code. 

I still use it for strain mapping, but everyone (rightly) makes fun of it.

din14970, 2020-12-03:
A bit more progress.

Using the helpful examples from @hakonanes, @magnunor and @pc494  and the dask documentation, I got the indexation to work on the whole lazy dataset. Thanks for that, i think I have a better grasp on dask now. The IPF map:

[image]

A stereo IPF plot:
[image]

@hakonanes these are MTEX plots, I couldn't figure out what the good way is to make these kind of images with orix. Do you have a suggestion on that?

I have to look into why some regions were indexed as blue and that small region is red, even though I know everything should be green.

I used the "mixed" approach for indexing each pattern whereby X top candidates are selected from the radially integrated profiles, and then a full rotation calculation is performed on these. The time to index one pattern scales almost linearly with X, in this case i used X=100 which gives about 60 ms per pattern. I use the `map_blocks` approach to iterate over the dataset. I use intermediate functions to iterate over chunks, which call the respective functions for individual patterns. First I map the polar conversion to the dataset, which adds an overhead of about 5 min for the dataset when using a single thread. Then I map the indexation procedure onto the dataset, which returns an index for the best fitting template, the best fit angle, and the correlation value. I use the angle and index to get a (Y, X, 3) euler angle array. 

Unfortunately I am not getting the speed up I had hoped for using multiple workers. When I call `compute` at the end I use the `threads` scheduler and 4 workers, so I was thinking indexation would take 52000 patterns x 60 ms/pattern / 4 workers = 13 minutes + overhead. However, the indexing took 48 minutes. Checking the workers using the dask diagnostics dashboard suggested that workers were mostly idling. I figured then it might be IO limited, given that the data must come from disk. However, if I changed X to 200, indexation took 1 h 25 minutes, almost two times as long as with X=100, indicating the expensive step is the full indexation. So it is not effectively using the workers for some reason. Using a single worker I seem to be getting a very similar performance, though the calculation is not yet complete. So it seems this related to the GIL, since the "index chunk" function is a python loop. This is pointed out here: https://blog.dask.org/2020/07/30/beginners-config. Might try the process scheduler to check. However, processes present more overhead, i'm wondering whether there is a way to get multithread to work instead. Any ideas?

When I get this working I want to try the masking, and then indexation with the second phase.

din14970, 2020-12-03:
Ok I think I found a quick way around the GIL. I modified whatever I had so that my `index_chunk` function could be compiled with numba with the `nogil=True` flag. Then, `map_block` can be parallelized across threads. Looks like it will take about 20-25 minutes now. The main bottleneck now is the polar transform of the images, which can not be compiled with numba, and so looping over a block still locks the interpreter. When it's busy with this step the progress bar clearly slows. Still working on a solution for this.

hakonanes, 2020-12-03:
> @hakonanes these are MTEX plots, I couldn't figure out what the good way is to make these kind of images with orix. Do you have a suggestion on that?

There's no IPF map colouring or IPF plotting in orix yet, unfortunately. So exporting to MTEX is for now necessary to get those. Regarding the colour: you expect all patterns to have their [011] plane normals parallel to z?

> I used the "mixed" approach for indexing each pattern whereby X top candidates are selected from the radially integrated profiles

I'm curious, how do you select these during computation, do you use dask's `argtopk` (indices of top values) or `topk` (top values)? We use this [in kikuchipy](https://github.com/pyxem/kikuchipy/blob/master/kikuchipy/indexing/_pattern_matching.py#L278) to get the indices of the best matching simulated patterns in a dictionary of simulated patterns with satisfactory performance.

> When I call compute at the end I use the threads scheduler and 4 workers

Although we use dask for all processing in kikuchipy, we don't configure dask at all, we just specify the chunks. As mentioned in [the introduction to the blog post you reference](https://blog.dask.org/2020/07/30/beginners-config): "Possibly for a long time you can get away with not configuring anything special at all." So I cannot say anything meaningful about configuring dask...

din14970, 2020-12-03:
> I'm curious, how do you select these during computation, do you use dask's argtopk (indices of top values) or topk (top values)? We use this in kikuchipy to get the indices of the best matching simulated patterns in a dictionary of simulated patterns with satisfactory performance.

At the moment I just do the most brute force approach possible. I have all my templates converted to polar coordinates, stored in arrays. Calculating the azimuthal integral of these templates is simply making an appropriate sum. I compare this to the sum over the azimuthal axis of the pattern. This entire calculation only takes 10 ms/pattern or so. I get out a 1D vector with correlation values for each template. I find the pattern indices with a correlation within the top `x` percentile. I then do brute force matching with those templates. If I use ~100 patterns this takes an additional 50 ms/pattern.

Once I've got the notebook a bit cleaned up you can take a look at the mess it has become. But at least it went from 72 hours to 25 minutes on my laptop.

hakonanes, 2020-12-04:
> I get out a 1D vector with correlation values for each template. I find the pattern indices with a correlation within the top x percentile.

It's this part I'm curious of, i.e. do you have this 1D vector in memory, or is it something you tell dask to calculate only when you call compute at some point. I'll wait for the notebook!

din14970, 2020-12-11:
@hakonanes @pc494 I've updated the notebook and started 3 PR. Please disregard the second half of the notebook, it's a bit of a sandbox, but definitely have a look at the rest. Some things in the notebook might be nice to add into code at some point, but I didn't want to bother at this stage. I got the IPF color mapping to work to visualize the result of the indexation. I do all the calculation in the notebook, you can check it out there. However, the way I do it only works for cubic, color maps for other crystal structures will be totally different.

din14970, 2020-12-11:
fixed some bugs and now the map looks like
[image]
I'm fairly confident about this one

din14970, 2021-01-13:
with #673, the indexing with in-plane rotation angle optimization should be solved. The template libraries will need a small update in Pyxem.

closed 2021-01-13 by din14970