pyxem/pyxem #664: Upcoming new version: pyxem 0.12.3

pc494, 2020-11-28:
It looks like hyperspy is about to release `1.6.1`, and I am going to backport this change to `0.12.x`, as the `1.6` series has a couple of really nice features that we want our users to have access to + it will close #620 (a regression from 1.5--->1.6).

This will be a little bit difficult without CI, but I'm hoping I can also tackle that on `0.12.x` and then port it up to `0.13`. Furthermore, it gives us a change to announce further changes directly with `warnings` from the code.

That would then make
0.12.2 a hyperspy 1.5 code
0.12.3 a hyperspy 1.6.1 code

Furthermore, we can start thinking about upstream peakfinding for 0.13 (currently PR #617). 

Thanks are due to @ericpre who seems to have been the one to give this hyperspy release the final push it needed.

dnjohnstone, 2020-11-28:
Sounds like a good plan to me - shall we try to work towards a v0.13.0 to close out the year?

pc494, 2020-11-28:
Softly, yes, I am hoping we can tidy enough loose ends up to make a 0.13.0 by the end of the year.

pc494, 2020-11-28:
Okay, doing some of these things locally it seems almost everything will pass with hyperspy `1.6.1`, which is good news. Once the test suite is back (see #661 for theat conversation) I will get to work on this.

pc494, 2020-12-02:
This is now at the conda-feedstock stage:

https://github.com/conda-forge/pyxem-feedstock/pull/16

pc494, 2020-12-03:
Finished the feedstock.

closed 2020-12-03 by pc494