pyxem/pyxem #666: Planning a pyxem 0.13.0 for the turn of the year

pc494, 2020-11-30:
While we've missed the target timeline laid out in #606, I think we have made good progress, and should think about releasing the code again. I have tried to divide up what needs to be done by theme (and thus likely contributors). This issue is marked discussion for now, I would hope to finalise an exact plan in the next fortnight or so.

-----------------------------------------------------------------------------------------

### Template and vector matching for orientation mapping 

This is mainly me and @din14970, and involves
(a) The new template matching workflow (#663, will close #656)
(b) Adjusting the output of vector matching workflow so that it can be exported to orix (#660) 

### Radial stuff, DPC, FEM and assorted other things I am less knowledgeable about

I think (though all help/contributions are welcome) that we have the following on the list
(a) A couple of small usability bugs, #641 and #642 that could be fixed for 0.13.0 or could be patched in later
(b) A waiting for review (sorry, my bad) FEM redesign (#618 and closing or at least reducing #555) from @CSSFrancis 
(c) The content of issue #643, which I think is a big selling point of this version, that is lazy centring (completed) and azimuthal integration (to come) @magnunor 

### Broader housekeeping (mainly me)

(a) We should revist if we can support Python 3.9, and if we want we should force users to use 3.8 (#650) 
(b) If possible, now would be a good time to upstream the peak finding code to hyperspy
(c) It may now be the time think about sweeping the code for potential deletions, deprecations and docstrings
(d) Testing clean ups and duplication removal (#632) 

----------------------------------------------------------------------------------------------
Not on the agenda (yet).

We currently don't have the resources to rebuild `DiffractionVectors`, while this is disappointing, it's a huge task, and getting new code out to users is the priority of this release. This means we will not be dealing with #469 just yet (despite it being low hanging fruit) and some other, similar bits of the code base.

magnunor, 2020-11-30:
Sounds good! I should be able to at least get the lazy azimuthal integration1d sorted (https://github.com/pyxem/pyxem/pull/668), and probably the other `_integration` functions.

With regards to improving handling for lazy signals, my pull request in HyperSpy should greatly improve the plotting performance. However, that won't be in before HyperSpy 1.7 at the earliest... So we shouldn't wait for that.

pc494, 2020-12-06:
Giving this a deadline of Friday the 18th.

dnjohnstone, 2020-12-06:
> Seem that this is in a good place, so I'm setting a deadline of Friday the 18th.

Are we going to try and get testing and coverage back to 100% for this? I can accept not doing that, but would like to have a plan in place for which version we're going to restore that in :)

pc494, 2020-12-06:
Nope, you're quiet right @dnjohnstone, we should do that for this release, no time like the present etc. etc. :D

magnunor, 2020-12-08:
I'm not sure how much time I'll have to contribute for the 18th of December deadline (need to prepare an exam for 19th of December). I'll most likely have time to finish the azimuthal integration in https://github.com/pyxem/pyxem/pull/668, but probably not much else.

How much of the missing coverage is from the tests from pixstem?

pc494, 2020-12-08:
If we get #668 in that'll be great, I don't think we need anything else from you for 0.13.0! 

For example, I'm happy to deal with the missing coverage (I think it's a ~10 single lines from pixstem code) - most of our lost coverage is code that I wrote (and consciously uncovered by deleting some bad bits of the testing suite)

magnunor, 2020-12-11:
One thing which should be checked, is the that the new version of `center_direct_beam` does not affect the template matching processing.

The results should be similar, but there might be some small differences. I don't do phase mapping using template matching myself, so someone with some experience in doing this should give it a go.

pc494, 2020-12-15:
We're going to miss Friday the 18th as a deadline. The duplication tidy up is taking me a fair bit longer than expected, however, it is making the code much much better as I'm able to select the better of pairs of implementations in a lot of places.

If people are keen to help, LGTM's on the code tagged "needs review" would help (I'll self merge the smaller ones) as my reviewers are currently kinda busy. (2) If you have open PR's, work on them as if we were going to hit the original deadline.

magnunor, 2020-12-17:
Unfortunately my exam-related tasks ate up a bit too much of this week, but I have some time dedicated tomorrow for pyXem programming.

pc494, 2020-12-17:
That's great @magnunor, although we're going to be late it's going in the right direction. I'll be opening an issue later today that I would really appreciate your input on (I will tag you).

pc494, 2020-12-23:
Revised deadline is January 8th, which I think we should be able to hit fairly comfortably.

pc494, 2021-01-04:
This is making good progress, I'll be working on the diffsims integration and returning coverage to 100% tomorrow.

pc494, 2021-01-06:
#704 has changed the namespace a lot, which means it's very likely all the demos are broken. I'm going to make a release candidate (once I've got the coverage back to 100%) so that I can start fixing the demos (so that we can bundle them and release as we've yet to automate that process)

magnunor, 2021-01-07:
Working on getting the coverage to 100% now, should have it finished in an hour or so.

pc494, 2021-01-07:
heads up, I just fixed a typo on directly on `master` (sorry) so you might have a simple merge conflict to fix.

Either way, I anticipate a release candidate coming today.

magnunor, 2021-01-07:
Still working on this, taking a bit longer than expected as the missing lines are sometimes a bit hard to get tested.

pc494, 2021-01-07:
No worries, some of them looked very deep, I'm not that opposed to `pragma: no cover` for the most difficult ones

pc494, 2021-01-08:
We've got to the "stuff that the demos doesn't like" stage of tidying up here, so I'm putting in a hard deadline of Monday night, anything not in then can be patched/small-minor-released when it's completed.

pc494, 2021-01-08:
We're at demo tidying up stage, for the time being I'm going to keep working on them (and iterating pyxem accordingly) and once that's done we can release.

pc494, 2021-01-11:
Will be able to finish this tomorrow morning.

dnjohnstone, 2021-01-12:
Just to flag the need for coverage on the `AcceleratedIndexationGenerator`

pc494, 2021-01-12:
> Just to flag the need for coverage on the `AcceleratedIndexationGenerator` and please can the pyxem-demos notebooks have a "restart and clear output" on all of them before release of those for aesthetics?

Thanks, I had noticed and forgotten the missed coverage.

Expecting to run each of the demos locally to check for changes etc. Do 01 and 05 (that I was able to run last week) look okay to you?

pc494, 2021-01-13:
It's out, closing this issue. Thanks to all those that have contributed.

closed 2021-01-13 by pc494

hakonanes, 2021-01-13:
> It's out, closing this issue. Thanks to all those that have contributed.

Congratulations, good work!