pyxem/pyxem #675: Direct beam finding on lazy dataset throws chunking error

din14970, 2020-12-13:
**Describe the bug**
My dataset is of shape (200, 260, 256, 256) and I want to process this lazily. I wanted to try out `center_direct_beam` but I get the error
```
ValueError: iter_array nav shape (201, 260) must be same as dask_array nav shape (200, 260)
```
For some very strange reason, the data is stored with chunk size (3, 2, 256, 256), so I suppose the error may result from the fact that 3 does not divide 200.

If I rechunk the data into (20, 26, 256, 256) I don't get the error.

magnunor, 2020-12-15:
Thanks for reporting this! I'll see if I can track down the cause.

> For some very strange reason, the data is stored with chunk size (3, 2, 256, 256), so I suppose the error may result from the fact that 3 does not divide 200.

This is probably due to a `rebin` or `change_dtype` earlier? HyperSpy automatically rechunks lazy signals in an attempt to optimize the chunk size. I don't like this as a default behavior for STEM-diffraction datasets, so I think the default behavior for at least `Signal2D` should be changed to not automatically rechunk. But I haven't had time to make an Issue about it on the HyperSpy issue tracker. This rechunking can be disabled by doing `rebin(..., rechunk=False)`.

din14970, 2020-12-15:
> This is probably due to a rebin or change_dtype earlier?

No, this is the automatic chunking as determined by hyperspy for saving data to the hspy format. In the creation of the hspy file I don't actually use dask at all.

magnunor, 2020-12-15:
Makes sense. The chunk size can be set with `chunks` parameter: `s.save("...", chunks=(32, 32, 32, 32))`.

Note: this if you have any NumPy arrays or Signals in the metadata, this will lead to errors.

magnunor, 2020-12-18:
This should be fixed in https://github.com/pyxem/pyxem/pull/686

closed 2020-12-21 by pc494