pyxem/pyxem #678: Code cleaning in diffraction2d

pc494, 2020-12-15:
FAO: @magnunor (it's a quick one I promise!)

In diffraction2d a lot of the methods that are written for lazy which start with the following

```python
if self._lazy:
      dask_array = self.data
else:
      sig_chunks = list(self.axes_manager.signal_shape)[::-1]
      chunks = [8] * len(self.axes_manager.navigation_shape)
      chunks.extend(sig_chunks)
      dask_array = da.from_array(self.data, chunks=chunks)
```

I imagine this will be the same in all methods and we could factorise it out? Is there an obvious reason that isn't the case?

magnunor, 2020-12-15:
I agree, they should be calling a common function for "converting" a NumPy array to a dask array.

I made such a function for the newer "lazy" functions I've made, for example in `center_direct_beam`: https://github.com/pyxem/pyxem/blob/master/pyxem/signals/diffraction2d.py#L1103. The function is in `dask_tools`, https://github.com/pyxem/pyxem/blob/d668cce7a2a332ea4e22767bf7ead5196c248187/pyxem/utils/dask_tools.py#L86

pc494, 2020-12-15:
Brilliant, problem solved :)

closed 2020-12-15 by pc494