pyxem/pyxem #684: Merging pixstem and pyxem Diffraction2D methods

pc494, 2020-12-17:
As described elsewhere (eg. #632 and #678) we're in the process of combining these functionalities, as well as reducing the testing so that most things are covered as much as necessary (and not more). In general methods come in one of three flavours

1) Implemented in both pyxem "non-lazy" and pixstem "lazy"
These should probably be merged together, but on occasion this makes the docstrings very complicated, as some methods will be available in only one or other framework

2) Implemented only as dask methods
In general these are easy enough, as in-memory can be converted with a util at the top of the method, however some of these functions should probably include a lazy prefix

3) Implement only for in-memory work
These will be most natural to the average user, so can probably be left, but see (2)

-------------------------------------------------------------------------------------------

Below is a list of the current methods, with the exceptions of `correct_bad_pixel` (which has of type (1) and has been addressed in a #681) and hot/dead pixel finding (of type 2, cleaned up in #683)

`get_direct_beam_mask` - in memory, needs no action (IMNNA)
`apply_affine_transform`  - IMNNA
`apply_gain_normalization` - IMNNA

5 methods based off PyFAI, to be dealt with elsewhere.

`get_direct_beam_position`
Currently a type (2), but in memory code also exists, this really should be refactored to become type (1), but could probably wait for a patch release if time does not allow

`center_direct_beam` 
As above, and similar levels of refactoring needed, but potentially is working out of the box at present

`remove_background` - in memory, duplicated by lazy method `subtract_diffraction_background` - should be combined for 0.13.0

`shift_diffraction` - lazy, could be wrapped over `apply_affine_transform` to offer an in-memory choice 
`threshold_and_mask` - non-lazy, needs no action
`rotate_diffraction` - currently non-lazy + should be running over the top of `apply_affine_transformation` (patch later?)

`flip_diffraction_x` and `flip_diffraction_y` - IMMNA, but could be merged into a single method? Also should this have an inplace option?

`center_of_mass` - currently lazy only, have mixed feelings, but probably okay to leave for now

`add..._as_markers` - no strong feelings. Will leave.

`virtual_..._field`, these conflict in naming with the rest of the repo. Current thinking is to merge into a single lazy prefixed function with naming that is consistent with the rest of the code.

Another radial function, will deal with elsewhere.

`template_match_disk`, this can now be done with find_peaks, will delete
`template_match_ring` seems fine
`template_match_with_binary_image` - I don't think I understand what this is about

`peak_position_refinement_com` - should be lazy only, in-memory has a different route with `subpixelrefinementGenerator`
`intensity_peaks` - probably okay as is, will refactor the opening bit to use the util (patch fix)
`angular_mask` - think this is lazy only, will rename as such

`angular_slice_radial_average` - no idea what this is doing, help invited

`get_variance` - another different conversation
`make_probe_navigation` - good method, no changes needed
`plot`- sensible overriding, no changes

-----------------------------------------------------------------------------------------------------

I will start marking these as with PR numbers as I get to them.

pc494, 2020-12-17:
@magnunor this was the issue I was talking about earlier on in #666

magnunor, 2020-12-18:
Cleaning up the methods sounds like a good idea!

I'm not sure if I understand all the details though. What do you mean with:

> pyxem "non-lazy" and pixstem "lazy"

The way I see it, every method could ideally utilize `dask` for the underlying processing. With this, we get automatic parallel processing, and easy handling of very large datasets. So the only difference between processing "lazy" or "non-lazy", so that `.compute` is called inside the method or not. "Converting" a NumPy array to dask array inside a method is straightforward using `da.from_array`. And doing in-place processing into an already existing NumPy array is easy using `da.store`. If the returned object should be lazy or non-lazy can be handled inside the method with a parameter, like `lazy_result=None`. If the method is used on a `LazyDiffraction2D` object, by default a lazy signal is returned. If run on a `Diffraction2D`, a non-lazy signal is returned.

For example, `center_direct_beam` (should) currently work nicely with both "lazy" and "non-lazy" signals. The only difference is really if the user wants the calculation to run now, or later.

Another example is `center_of_mass`, this could simply get the same treatment as `center_direct_beam`, where `lazy_result=None` is used: non-lazy returns non-lazy, lazy return lazy.

pc494, 2020-12-18:
(sorry this is very short, about to jump into a meeting but wanted to get this out ASAP)

> The way I see it, every method could ideally utilize dask for the underlying processing.

I'm not that convinced by this; see for example https://docs.dask.org/en/latest/best-practices.html which seems to lean very heavily towards getting out of dask as soon as one can.

magnunor, 2020-12-18:
Yeah, I probably phrased that a bit poorly. I think that processing which is expected to handle very large datasets, like `center_direct_beam` should have one implementation in `dask`. With the "non-lazy" version simply calling `dask` function. I don't think having a separate non-dask function will improve processing times. (But it might! I haven't tried.)

And I totally agree with the dask best practices page: `dask` really isn't useful for every purpose, so it needs to be used with some care.

For other functions where the datasize will always be very small, (for example https://github.com/pyxem/pyxem/blob/master/pyxem/signals/differential_phase_contrast.py#L179), there is no need for `dask`, since the datasizes will always be tiny compared to the available memory.

pc494, 2020-12-18:
I think I'm going to run some performance tests for dask vs non-dask code in some of the cases in which we have both and see if there is a pattern. I do agree that if the differences are insubstantial there's an elegance to just converting everything into dask and having a single route.

magnunor, 2020-12-18:
One thing to keep in mind is that the chunk sizes is very important, and that by default HyperSpy does automatic rechunking in quite a few functions like `change_dtype` and `rebin`. This can drastically reduce the processing performance. The way to avoid this is using `rechunk=False`.

@hakonanes, might have some input on this, as he has been playing around with `dask.array` a bit.

magnunor, 2020-12-19:
I spent some time checking the processing times for `center_direct_beam(method='blur', sigma=20)`, on one of my experimental files. 

**Short version:** In the dev branch, the current default chunk size for non-lazy signals is not optimal for smaller datasets. This can easily be fixed by having a smarter way of setting the chunk size in `_get_dask_array`, instead of just having a single number.

**Long version:**
I loaded a large (512x512x256x256) file lazily, sliced out a region using `inav`, and loaded it into memory using `compute()`. With this, I made several different dataset sizes. Different parts of the dataset was used, to avoid caching between the calculations (expect for the full 512x512 probe size dataset). (See the full script at https://gist.github.com/magnunor/f66e3346efa7a0e6c24cb5bf5e9496e1). This was run on a 32 core (64 threads) computer with 128 GB of memory.

Probe positions is the size of the dataset, so 16 x 16 means the signal was (16 x 16 x 256 x 256). `Old` is the old implementation of `center_direct_beam` (pyXem 0.12.2). `New` is the implementation in the master branch of pyXem, NxN is the chunk size. So (32x32) means a chunk size of (32 x 32 x 256 x 256), which is the current default chunking. The results:

|Probe pos  |  Old      |New (8x8)  |New (16x16)|New (32x32)|New(64x64) |
|-----------|-----------|-----------|-----------|-----------|-----------|
| 16 x 16   |  0.18 s   |  0.77 s   |  2.27 s   |  2.27 s   | 2.27 s    |
| 32 x 32   |  0.52 s   |  0.88 s   |  2.43 s   |  8.54 s   | 8.64 s    |
| 64 x 64   |  2 s      |  1.9 s    |  2.98 s   |  9.1 s    | 34 s      |
| 128 x 128 |  7.7 s    |  5.6 s    |  7.45 s   |  11.3 s   | 36.4 s    |
| 256 x 256 |  30 s     |  31 s     |  28.9 s   |  30 s     | 44.9 s    |
| 512 x 512 |  137 s    |  121 s    |  115 s    |  112.8 s  | 123.8 s   |

- Compared to the current master branch implementation, the old implemenation is faster for smaller datasets (less than 256 x 256 probe positions)
- Lowering the chunk size to (8x8), the new implementation is on par with the old one, except for the smaller datasets (16 x 16) and (32 x 32).

So the worse processing times is caused by the calculations effectively becoming non-parallel when the chunks sizes become too large. However, having too small chunks, we bump into the problem of having too many tasks. This can drastically slow things down.

I think the best solution to this is when non-lazy signals are "converted" into dask arrays, the chunk size should be set in a smarter way to utilize the parallel processing aspects of `dask`.

For lazy signals the chunk size should be preserved as much as possible, from the HDF5 file to the last processing step. This due to the actual loading of the datasett from the harddrive being a big contributor to the processing time.

pc494, 2020-12-21:
Thanks for this @magnunor , it seems I had misunderstood quiet a few things/ideas, I'll try and tidy up this issue into something more coherent today. I will probably also run your experiment (or a scaled back version) on one of my machines, as I think 128GB is probably a fair bit more RAM than the median pyxem user has access too and I'm curious to see what happens.

pc494, 2020-12-21:
Okay, I've got this issue down to a bullet point list for the next release (in an edit to the original post)

magnunor, 2020-12-21:
> template_match_disk is going to be merged into find_peaks_lazy

I'm a bit hesitant with this change, as this is quite an important pre-processing step for much of the data processing we're doing, before doing peak finding. Finding the best `disk_r` can require a bit of fine-tuning, so having easy access to the disk-template matched image before doing the peak finding is useful.

I guess the name is a bit suboptimal, since it can cause some confusion with regards to the template matching often utilized in SPED phase map processing.

> The two background subtractions methods are going to be merged

Sounds good!

>  Refactor so the dask conversion is done by the util

I'm guessing this refers to moving some of the code for setting up the dask processing from the class methods to the utils module? If yes, that sounds good. Ideally most of the dask related "complications" should be moved away from the class methods, to the utils.

> virtual_imaging will be renamed to be "lazy" as we should use the generic method for in-memory work

Sounds good, at some point I think we should be able to get good plotting performance even with lazy signals, given that the signals are chunked approporately. But that is for a release beyond 0.13.

> I will probably also run your experiment (or a scaled back version) on one of my machines, as I think 128GB is probably a fair bit more RAM than the median pyxem user has access too and I'm curious to see what happens.

Good idea. There are a bunch of different effects of the chunking size, so finding the optimal one for most datasets is tricky. For example, having a bigger chunk size can use more memory, if we're not constrained by the number of cores. However, having small chunks can lead to generation of the task graph taking a long time.

pc494, 2020-12-22:
> I'm a bit hesitant with this change, as this is quite an important pre-processing step for much of the data processing we're doing, before doing peak finding. Finding the best `disk_r` can require a [...]

I was unaware of that workflow (as in-memory you can use the GUI to tune peakfinding), and so will leave the code as it is (+ I don't think the confusion between the two methods should be severe) 

> I'm guessing this refers to moving some of the code for setting up the dask processing from the class methods to the utils module? If yes, that sounds good. Ideally most of the dask related [...]

Honestly, it was mainly just the bit in #678, and any other bits of code that appear multiple times

pc494, 2021-01-04:
The two recently opened pull requests finish this up, closing.

closed 2021-01-04 by pc494