pyxem/pyxem #685: demos errors signal2D has no attribute

winston-song, 2020-12-17:
dp = pxm.load_hspy('./data/01/twinned_nanowire.hdf5')
dp.set_microscope_parameters(beam_energy=300.0,
                               camera_length=21.0,
                               scan_rotation=277.0,
                               convergence_angle=0.7,
                               exposure_time=10.0)
--------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-11-301c0fb46ce0> in <module>
----> 1 dp.set_microscope_parameters(beam_energy=300.0,
      2                                camera_length=21.0,
      3                                scan_rotation=277.0,
      4                                convergence_angle=0.7,
      5                                exposure_time=10.0)

AttributeError: 'Signal2D' object has no attribute 'set_microscope_parameters'

data = pxm.load_hspy("./data/09/PdNiP_test.hspy")
data.set_signal_type("electron_diffraction")
data.beam_energy=200
data.unit = "k_nm^-1"
rad = data.get_azimuthal_integral2d(npt_rad=100, center=(31.77734804, 31.23438963))
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-6-f9fdd1ef8e89> in <module>
      2 data.beam_energy=200
      3 data.unit = "k_nm^-1"
----> 4 rad = data.get_azimuthal_integral2d(npt_rad=100, center=(31.77734804, 31.23438963))

AttributeError: 'Signal2D' object has no attribute 'get_azimuthal_integral2d'

hakonanes, 2020-12-17:
Apparently your signal became a `Signal2D` at some point, which doesn't have that class method? From [notebook 1](https://nbviewer.jupyter.org/github/pyxem/pyxem-demos/blob/master/01%20GaAs%20Nanowire%20-%20Data%20Inspection%20-%20Preprocessing%20-%20Unsupervised%20Machine%20Learning.ipynb) it seems that you must use `dp.set_signal_type("electron_diffraction")`? I guess more active pyxem users can give you the answer if this doesn't work.

pc494, 2020-12-17:
Hi there,

In the short term these can be fixed by add the line

`data = pxm.ElectronDiffraction2D(data)`

after loading. The 0.12 series (which I presume you are running) has some issues with `io` that should be fixed in our next release.