pyxem/pyxem #692: Add deprecation to Diffraction2D.radial_average

magnunor, 2021-01-01:
Currently there are several implementations for doing azimuthal integration (or sometimes called radial average or radial integration).

Since an improved version (`get_azimuthal_integral1d`) has been implemented in the current development version, the old ones should be removed at some point. One example of this is `radial_average`. Thus, there should be a `DeprecationWarning` for `radial_average` function, saying it will be removed in a future version, possibly the next one?

This might also extend to other functions.

pc494, 2021-01-01:
I was planning to make these deletions for the 0.13.0 version, but am happy to mark as deprecated (and for deletion in 0.14.0). Either way it's on my list for next week.

@magnunor am I right in thinking you got PyFAI to play nicely with dask then (in #668)?

magnunor, 2021-01-04:
@pc494, dask  + pyfai seems to work nicely together. Testing the implementation from #668 with my fairly underpowered laptop on a experimental dataset, it worked really nicely!

The one thing which should be done is adding a "smarter" `_get_dask_array`, so that in-memory (non-lazy) datasets gets a more optimal chunk size. Especially for datasets smaller than (32 x 32) navigation positions, as they're effectively computed non-parallel at the moment. I'm not sure how best to handle it, but we can probably add functionality so that the data at least uses 4 cores?

pc494, 2021-01-13:
This was closed by #710

closed 2021-01-13 by pc494