pyxem/pyxem #706: Further Optimizing Chunking and Lazy Operations

CSSFrancis, 2021-01-05:
**Is your feature request related to a problem? Please describe.**

This is something that @magnunor might have already figured out (and is working on it) but I think that we should provide a better interface for controlling how we chunk our data. Hyperspy is particularly obtuse when it defines chunks, often times defining chunks of kb when they should be 100's of megabytes to gigbytes if [possible.](https://docs.dask.org/en/latest/array-chunks.html)  In many cases working with a dataset it will spawn 100,000's of tasks which completely overwhelms the task manager and freezes the operation.  This is why the .map function doesn't seem to work properly with large files as it spawns a task for each individual frame.

**Describe the solution you'd like**
- The way that I see it I think that something like a rechunk function for all of our lazy signals might be useful where it either does the rechunking for the user automatically giving them chunks of around 100 mb or allow the user to set the size of the chunks without having to rechunk the data.  
- I also think that something along the lines of a warning if the chunks are very small and the overhead is going to be large
- Finally the hyperspy.load function needs an option to define your chunks as you are loading the dataset so you don't have to rechunk the whole dataset.

**Describe alternatives you've considered**
- Most of this could be accomplished in hyperspy which is probably the better solution.  The loading should be handled there which might not be the hardest thing to do.  
- Rewriting the map function might be the best thing to do as it seems to be fairly broken...

@magnunor you probably have a lot better idea of what is going on with regard to this, I was just playing around with using lazy signals and realized that there is probably a fair bit that should be done both in hyperspy and in pyxem and I wanted to have a place for discussions about how to move forward using lazy operations.

magnunor, 2021-01-06:
I very much agree! How to properly do chunking is pretty complicated and a very big topic. But I think we can get some large performance improvements through some smaller changes in how HyperSpy and pyxem handles data chunking. One issue is that finding an "optimal" chunking strategy is really tricky, as the optimal chunking heavily depends on what type of processing people are doing.

I do think one of the issues is HyperSpy's automatic rechunking, for example in `change_dtype` and `rebin`, which I always disable using `rechunk=False`. This should probably be disabled by default for LazySignal2D, as keeping the same chunks through the whole processing is typically the better option.

Lets take one example, where we want to calculate a simple bright field image from a lazy dataset.

```python
from time import time
import pyxem as pxm

##############################################
s_lazy = pxm.load("000_dataset.hspy", lazy=True) # 512 x 512 dataset
s0_256 = s_lazy.inav[0:256, 0:256]
s1_256 = s_lazy.inav[0:256, 256:512]

##############################################
s0_256.change_dtype('float32', rechunk=False)
t0 = time()
s_bf = s0_256.isig[128, 128]
s_bf.compute()
print("Rechunk=False: ", time()-t0)

##############################################
s1_256.change_dtype('float32', rechunk=True)
t0 = time()
s_bf = s1_256.isig[128, 128]
s_bf.compute()
print("Rechunk=True: ", time()-t0)
```

* `Rechunk=False:  1.2349793910980225`
* `Rechunk=True:  46.22017526626587`

The reason for this is that the automatic rechunking makes the whole signal dimension into one chunk. Thus, for the `rechunk=True` the full dataset needs to be loaded into memory. While for the `rechunk=False` only the chunk in the center of the diffraction pattern needs to be loaded into memory.

One big thing I'd like to do at some point, would be to improve HyperSpy's `map` functionality, for example by "upstreaming" the `_process_dask_array` implementation. However, the `map` implementation is a lot more "general" with additional features, making it a bit more difficult...

ericpre, 2021-01-06:
Some of the points mentioned above should be easy to fix in hyperspy. Even if the implementation of the `map` is not the most pleasant to read, it is actually not that complicated, and if the problem is well understood and there is a known solution, its implementation shouldn't be difficult.

din14970, 2021-01-13:
Sorry to piggyback off of this discussion but I'm interested to follow up and maybe also get some feedback on the thing I implementated. In https://github.com/pyxem/pyxem/blob/e85ec2458faf3a3227696dd35c79403cd4ae7082/pyxem/utils/indexation_utils.py#L1437 I dealt with similar issues in trying to index a dataset lazily, and since hyperspy seems to load the dataset with the same chunk size as the hdf5 dataset and the map function wasn't really doing what I wanted, I just decide to work with raw dask and apply operations via `map_blocks`. The default setting I have in the function is to rechunk the incoming data with:
```
data = data.rechunk({0: "auto", 1: "auto", 2: None, 3: None})
```
I feel like this makes sense for 4D-STEM datasets? But the user could also set chunks manually via a kwarg.

@ericpre I was actually wondering why the map function does not simply use the functionality of dask by default? The part with the threadpool if `parallel=True` and loop if `False` seems quite complicated. Could this entire thing not be condensed into `map_blocks` both for lazy and non-lazy datasets? Or am I missing something critical.

CSSFrancis, 2021-01-13:
@din14970 I think that you bring up a good point.  I think @magnunor has dealt with these in the past as well.  Maybe something that should be created as two new issues.

First, parallel operation isn't very useful.
-   I would really want parallel operation with large lazy datasets but right now parallel operations don't work with lazy implementations.  Moving everything to Dask might be better in the long run.  
- Secondly, having used hyperspy's parallel processing it seems to have a long set up time and suffers from the same problems that the old map_iterate function did (I think it operates on individual signals rather than chunks, causing it to spend a lot of time creating tasks for workers to complete)

Secondly, the loading and saving of lazy datasets could use some work.  
- There might be some clever ways to speed up saving time
- When loading a lazy signal, it should allow a chunks kwarg when loading.

ericpre, 2021-01-13:
It looks to me that the issue is with the main lazy version of `map` is that it most likely hasn't changed since its initial implementation and nothing has happened to try to improve the situation. Things has changed since then, and knowing what are the issues, it should possible to fix them! https://github.com/hyperspy/hyperspy/pull/2617 may be a good start.

pc494, 2021-04-15:
This is being addressed via hyperspy and #759

closed 2021-04-15 by pc494