pyxem/pyxem #711: Should 'load' be in the pyxem namespace?

pc494, 2021-01-08:
In 0.12.2 we could run 

```python
pxm.load('some_filepath')
```

because we have hyperspy's load in the `__init__`, is there a good reason not to do this? @ericpre you seem like the person most likely to have a good answer to this.

pc494, 2021-01-08:
Similarly, we used to have the binding to do `pxm.roi`

ericpre, 2021-01-08:
Putting some of the API as part of another library API is not common, as far as I know and I don't know why we would do this. I think we should ask the question the other way round: is there a good reason to do this?

In the case of the `load` function, I guess, initially, the idea was to overload the hyperspy `load` function with functionalities to load mib file, but since then there is the `load_mib`?

My view on this is that it would be more transparent to the user, if `pxm.roi` and `pxm.load` were directly imported from `hyperspy`. If users want to know about these, without any doubt they will be looking at the hyperspy documentation, same applies for reporting bug, asking questions, etc.

pc494, 2021-01-08:
A strong case well put. 

I think you are correct that are original motivations depended on the fact that we overloaded `load`, but as we are no longer doing that I don't see any reason to stick to old (bad) habits. Many thanks @ericpre, closing this now.

closed 2021-01-08 by pc494