pyxem/pyxem #718: add_peak_array_as_markers() - Usage and possible bugs

edwinsupple, 2021-01-28:
I am attempting to find diffraction peaks on some 4dstem data and then plot the peaks onto the dataset.

I load the dataset (FEI .emi file) as `'electron_diffraction'`, use `.set_diffraction_calibration()` on the dataset, label a sum of a few DPs as my disc image, use 'xc' method to find peaks and save them as a variable. I then use `.add_peak_array_as_markers()` to add the peaks to the dataset, and plot the dataset. I've put a snippet of my code below.

I have two issues: first, the points are clustered too closely together and in the top left corner. I did see a previous comment about this being related to properly calibrating. As a workaround I'm multiplying the `DiffractionVectors` data by 13.75 and adding 254 to properly scale and center the data. When the dataset loads the signal offset x and y is -18.8E10. after calibration it is -18.8. I've also tried setting the offset to 0 and that appears to make no difference in the plotted location of the markers. When I use the interactive peak finder it puts the markers in the proper location.

Second, the points are turned 90 degrees from their correct position. In marker_tools.py on lines 101 and 102 the points are assigned values: I switched vx to peak[0] and vy to peak[1] and this turned them correctly. But I'm not sure if this is a bug or if I'm not passing the function the correct data.

```
                        vx = _pixel_to_scaled_value(signal_axes[0], peak[1])
                        vy = _pixel_to_scaled_value(signal_axes[1], peak[0])
```
As an aside, I expected to be able to pass `add_peak_array_as_markers()` the `DiffractionVectors` directly but instead is seems to want `DiffractionVectors.data`. And so I wonder if this is part of my issue.

Can someone please comment on these points and maybe give some example code of using `add_peak_array_as_markers()` if that would be helpful? 

```
dp4 = pxm.load('%filename%.emi', signal_type='electron_diffraction')

dp4_vacuum = dp4.inav[34:40,2:7].sum((0,1))
dp4.set_diffraction_calibration(0.0734577)

dp4_peaks = dp4.find_peaks(method='xc', disc_image=dp4_vacuum.data, min_distance=5, peak_threshold=0.01)

dp4.add_peak_array_as_markers(dp4_peaks.data*13.75+254.)
dp4.plot()
```
marker locations before scaling:
[image]
after scaling:
[image]

pc494, 2021-01-29:
Hi @edwinsupple,

This is a good issue, and I'm sorry to have to have an annoying follow up question as my first response; but it appears your using an older version of pyxem (`'xc'` ran in `0.12.x`, it's now called `'template_matching'` after we upstreamed it to hyperspy). Would it be possible for you to install the latest version and see if you run into the same error?

As you say, the "top left hand corner markers" is characteristic of an uncalibrated signal, but it seems you've done that part of the workflow correctly (as the scalebars are not in pixels). I think it's very likely a bug, but I would like to be able to test it on the current version.

magnunor, 2021-01-29:
@edwinsupple, thanks for reporting this!

This is due to the merger between `pixstem` and `pyxem`, which has lead to some inconsistencies with regards to coordinate systems, rotations, calibrations, +++. We're working towards resolving this, to make the data processing a bit more streamlined. The latest release was mostly focused on getting the azimuthal integration processing "pipeline" working with very big datasets. The peak finding and visualization of these results will probably be the next to improve.

An example of using this:

```python
import pyxem.dummy_data.dummy_data as dd
import pyxem as pxm
s = dd.get_cbed_signal()
st = s.template_match_disk(disk_r=6, lazy_result=True)
peak_array = st.find_peaks_lazy(lazy_result=False)
s.add_peak_array_as_markers(peak_array)
s.plot()
```

edwinsupple, 2021-01-29:
Phillip and Magnus,

Thanks for your quick responses on this. I upgraded to pyxem 0.13 and tried your suggestions. I adapted the code from Magnus and it found my peaks and properly placed them on the dataset. I think I will have to spend more time with `skimage.feature.blob_dog` / `.blob_log` to get the parameters set nicely to not pick up spurious peaks. But basically my questions about marking peaks have been resolved.

Not to go too far off topic, but while installing 0.13 I did have a few issues, some/all of which might not be pyxem's problem. The jupyter installation couldn't find the right kernel so I had to change my kernel.json reference. I also had to install PyQt5, pyopencl, and hyperspy_gui_ipywidgets via pip after using conda to install pyxem. Easy enough to do but not sure if that's standard behavior or something going wrong with the conda install.

Thanks again,

Edwin

pc494, 2021-02-18:
Closing this, as the original issue has been solved. See #725 for the follow up.

closed 2021-02-18 by pc494