pyxem/pyxem #719: load broken

maclariz, 2021-01-29:
pxm.load('filename.extension') does not work in 13.0

I downgraded to 12.3 and it works fine.  

Your sheet 01 GaAs Nanowire - Data Inspection - Preprocessing - Unsupervised Machine Learning.ipynb

does not run under 13.0 as a result

Ian

pc494, 2021-01-29:
Hi Ian,

I think this is a feature and not a bug; after discussion #711 we decided to keep it that way.

(fingers crossed) `hs.load` should work just fine

maclariz, 2021-01-29:
OK

Just new to this.  If you are happy with loading as hs objects, then we are all good.  Just python numpties up here just making our first halting steps into the world of pyxem since we want to do some SVD on our noisy data and then index it…

Best wishes

Ian

maclariz, 2021-02-01:
Phillip,
Revisiting that tutorial notebook at some point might be a good plan

Ian

On 29 Jan 2021, at 17:45, Ian MacLaren <Ian.MacLaren@glasgow.ac.uk<mailto:Ian.MacLaren@glasgow.ac.uk>> wrote:

OK

Just new to this.  If you are happy with loading as hs objects, then we are all good.  Just python numpties up here just making our first halting steps into the world of pyxem since we want to do some SVD on our noisy data and then index it…

Best wishes

Ian