pyxem/pyxem #720: Adding in a Counting Filter

CSSFrancis, 2021-02-03:
**Is your feature request related to a problem? Please describe.**
With faster and faster detectors coming onto the market it seems like applying a counting filter is going to become more and more common as a way to reduce readout noise by taking low dose images, applying a counting filter, and creating a binary image (1 for an electron and 0 for no electron). 

**Describe the solution you'd like**
- Adding a counting filter method to the electron_diffraction2d class that can be applied pixel by pixel or with sub pixel refinement
- Finding the center of mass for each electron strike
- Fast and applicable to lazy datasets

- Possibly including options to apply counting mode to certain regions of an image and not apply counting mode to other regions.

**Additional context**
Most of the counting mode previously done is behind the scenes.  It might be good to at least have a simple counting filter (or more advanced) counting filter publicly developed which is fairly fast.

pc494, 2021-02-04:
@CSSFrancis,

I have a couple of questions.

1) Is a counting filter more complicated (in terms of its logic) than:  `x_array > count_threshold`
2) How would one perform a subpixel refinement on this kind of operation?

Other than that this seems like a good idea + I think some bits of the functionality may already be in the code.

CSSFrancis, 2021-02-04:
@pc494 
> 1) Is a counting filter more complicated (in terms of its logic) than:  `x_array > count_threshold`

It can be. I know that some of the detector companies have algorithms they have developed but not released. I think most of those are for sub pixel refinement.

In general though there are a couple of cases:

1 - An electron hits the detector, excites only one pixel and depending on the gain of the detector some value is recorded. This case just applying the inequality works.

2 - An electron hits the detector and multiple pixels are excited. In this case, you still have much higher signal from the two excited pixels so we have been applying an inequality x_array> (gain/10) or something like that and the finding the center of mass for those particles.

3 - You have regions of both high and low dose in the sample (zero beam and diffraction spots at higher k) so you might want to operate where the center beam is just divided by the gain and a counting filter is applied to any additional pixels. 
> 2) How would one perform a subpixel refinement on this kind of operation?

For the sub pixel refinement I will need to talk to my advisor just to make sure I have everything correct. But I think the idea here is that the center of mass calculation gives you a float value that you could use to effectively increase your resolution. There is a limit to how much you can increase this based on the detector (pixel size, and thickness I think) but there isn't much literature out there. 

Sorry I'm not very concrete. I haven't found much literature on how counting is being applied currently, this is only how I would do it. Counting filters are mostly being used by the cryoEM community and are just developed by detector makers who are guarding probably 20 lines of code. 
> Other than that this seems like a good idea + I think some bits of the functionality may already be in the code.

Yea! I think there is a lot of the functionality already existing in the code, but it would be worth it to bring it all together.

pc494, 2021-02-04:
Seems like an excellent candidate for the next minor release :)