pyxem/pyxem #723: Diffraction2D.correct_bad_pixels: lazy vs non-lazy

magnunor, 2021-02-12:
In the most recent version of pyxem (0.13.0), and the current master branch, lazy and non-lazy version of `correct_bad_pixels` works differently.

For example:
```import numpy as np
import dask.array as da
import pyxem as pxm

data = np.random.randint(100, 120, size=(3, 3, 20, 20))
data[:, :, 12, 14] = 0
data[:, :, 5, 2] = 0

# Lazy
s = pxm.signals.diffraction2d.LazyDiffraction2D(da.from_array(data, chunks=(1, 1, 10, 10)))
s_dead_pixels = s.find_dead_pixels()
s_corr = s.correct_bad_pixels(s_dead_pixels)

# Non-lazy
s = pxm.signals.diffraction2d.Diffraction2D(data)
s_dead_pixels = s.find_dead_pixels()
s_corr = s.correct_bad_pixels(s_dead_pixels) # Gives an error
```
The error:
```python
~/Documents/pyxem/pyxem/pyxem/utils/expt_utils.py in remove_dead(z, deadpixels)
    331     """
    332     z_bar = np.copy(z)
--> 333     for (i, j) in deadpixels:
    334         z_bar[i, j] = (z[i - 1, j] + z[i + 1, j] + z[i, j - 1] + z[i, j + 1]) / 4
    335 

ValueError: too many values to unpack (expected 2)
```

I don't have time currently to look at this myself, so putting this as an Issue for now.

pc494, 2021-02-12:
I'll have a look into it. In the short term I would trust the lazy version of the code.

pc494, 2021-02-26:
Will fix this for the forthcoming patch version

pc494, 2021-03-21:
fixed for version 0.13.1 (and #745 on master)

PR #736 (2021-03-18): 0.13.1 patch release
---
name: Further additions for patch release
about: This whacks of the rest of the content in #731, which will close #730 and #716. Previous PR's (that this branch builds upon) will then close #723. Once the tests pass I'll make the bumps to this can be released and then request a review.

---

**Checklist**
- [x] Updated CHANGELOG.md
- [ ] (if finished) Requested a review (from pc494 if you are unsure who is most suitable)
closed 2021-03-21 by pc494