pyxem/pyxem #725: Dependancies may not be correctly spelled out

pc494, 2021-02-18:
copied over from: https://github.com/pyxem/pyxem/issues/718#issuecomment-770010029, thanks to @edwinsupple

> Not to go too far off topic, but while installing 0.13 I did have a few issues, some/all of which might not be pyxem's problem. The jupyter installation couldn't find the right kernel so I had to change my kernel.json reference. I also had to install PyQt5, pyopencl, and hyperspy_gui_ipywidgets via pip after using conda to install pyxem. Easy enough to do but not sure if that's standard behavior or something going wrong with the conda install.

hakonanes, 2022-02-01:
The mentioned dependencies aren't required to run pyxem, but can be installed if the user desires (which I would assume most do). I suggest to close this.

CSSFrancis, 2022-02-01:
+1 to closing this. We could add something about the hyperspy bundle as an easy way to get started. I think it includes all of these.

hakonanes, 2022-02-01:
Yes, that would be good, with a mild warning stating that users might have to update pyxem themselves after installation of the Bundle to get the latest release, since the Bundle release cycle follows the HyperSpy release cycle ([relevant kikuchipy installation docs](https://kikuchipy.org/en/stable/installation.html#with-the-hyperspy-bundle)).

pc494, 2022-02-02:
I'm happy with this.

closed 2022-02-02 by pc494