pyxem/pyxem #727: Convention for masks

pc494, 2021-02-23:
This came up in: https://github.com/pyxem/pyxem/pull/724#discussion_r581242587

For a `mask` array we should be consistent in if `True` is "this pixel should be included in the data" or "this pixel is masked out of the data".

I have written code in the past that uses `True` to mean this pixel is included in the data. @hakonanes you must have made a decision in orix for `CrystalMap`, what did you pick? Do you remember why you chose that?

hakonanes, 2021-02-23:
To decide whether a point in a crystal map is "in the data" or not, I simply added a boolean array in the attribute `CrystalMap.is_in_data` ([code]https://github.com/pyxem/orix/blob/master/orix/crystal_map/crystal_map.py#L197)). `True` are in the data, `False` are not, which I think is obvious because of the attribute name. I chose this way of doing it to avoid having to use "~" every time I want the points in the data, which is far more often than the times I want the points *not* in the data.

When it comes to a parameter named "mask", I would stick to [what NumPy does](https://numpy.org/doc/stable/reference/maskedarray.generic.html#what-is-a-masked-array), which is use `False` for points "in the data", i.e. the points not "masked away". This is also adopted in the Dask masked array module, which wraps NumPy's module.

We don't support only applying kikuchipy EBSD methods on certain patterns given by a mask yet. I think we will start to support it bit by bit after, or while, I/we create a feature (/grain) class for orix (https://github.com/pyxem/orix/issues/151). Personally, I would prefer the way you mention you've done it, @pc494, but I think we'll go with what NumPy/Dask/HyperSpy does.

pc494, 2021-02-24:
`CrystalMap.is_in_data` is pleasing unambiguous...

I checked LIBERTEM, they use `1` to mean "included in data" too.

CSSFrancis, 2021-03-02:
Sorry I am a little late to the party, but I would think that we should follow NumPy and Dask if we are essentially using those packages as the foundation for every signal.  Eventually, I would like to see mask allowed using the .ma class in numpy, but the MaskedArray class in NumPy is a little bit of a [headache at the moment](https://numpy.org/neps/roadmap.html?highlight=nep#maintenance).  I doubt that will ever get finished quickly as it seems to be a big sticking point with the code currently.

So we could just add in our own mask implementation as well.

hakonanes, 2021-03-03:
I haven't thought much about how to implement masking, but our plan is to start small and allow passing boolean masks of navigation shape to some EBSD methods. I don't think storing a mask with the signal is necessary at the moment.