pyxem/pyxem #739: Progress in accelerating and correcting template matching (and possibly other indexation approaches)

din14970, 2021-03-07:
As I wrote in https://github.com/pyxem/diffsims/issues/157 there is a bug in my "fast" indexation approach which I implemented in #673 , whereby I don't account for mirrored templates. The implication is I will need to check a template library twice the size of what I initially thought. I initially thought to fix it by just passing in twice the template library with a PR in diffsims, but now I think it would be best to implement the change at the low level during indexation. Given that a full indexation of a reasonable dataset is still relatively slow due to the brute force nature, I also want to explore some alternative approaches besides fixing the bug. This issue serves mainly to keep track of my experiments, but any suggestions and feedback are welcome.

Something worth exploring due to the embarrassingly parallel nature of the problem is utilization of a GPU. Currently my template matching algorithm should already make use of multiple cores via `njit(parallel=True)` and dask.
On a two-core laptop this makes a difference of a factor of 1.6 or so. So I started testing a few things out on our GPU cluster (each node has 40 real CPU cores and 2 V100 graphics cards).

In the most naive approach, indexing an image with template matching + simultaneously optimizing for the in-plane angle can be accomplished with:

```python
def correlate_library_cpu(polar_image, sim_r, sim_t, sim_i, image_norm, template_norms):
    correlation = np.empty((sim_r.shape[0], polar_image.shape[0]), dtype=np.float32)
    for template in range(sim_r.shape[0]):
        for shift in range(polar_image.shape[0]):
            tmp = 0
            for spot in range(sim_r.shape[1]):
                tmp += (polar_image[(sim_t[template, spot]+shift)%polar_image.shape[0],
                                                   sim_r[template, spot]] * sim_i[template, spot])
            correlation[template, shift] = tmp/(image_norm * template_norms[template])
    return correlation
```

where `polar_image` is the DP image converted to polar coordinates, `sim_r` `sim_t` `sim_i` encode the (r, theta, intensity) coordinates of all diffraction spots in the template library. It returns a correlation array of shape `(number_of_templates, in_plane_angle)`, which is all the information one needs for a single image - this can be further reduced to a single angle and correlation value via maximum. 

As-written with 3 nested python loops I wasn't able to time it in reasonable time, it took longer than a minute to index a single image.

By `njit` -ing this with numba, it took 670 ms to index a pattern on a single CPU thread with 4186 templates with each 67 spots (some are 0). On a reasonably sized dataset, e.g. 200x200 templates this would take 7 hours to index which is unacceptable. Granted I took the number of templates to be very large to test it out.

By `njit` -ing using `parallel=True` and replacing the first for-loop with `prange`, on the cluster this took 20 ms, not so far off from 670/40, so it is able to use the full core capacity on a node. For a full dataset this should take about 13 minutes. I suspect that if I use the single thread numba function and parallelize across cores with dask instead I should get similar performance.

I then transformed this same function into a pretty naive cuda kernel:

```python
@cuda.jit
def correlate_library(polar_image, sim_r, sim_t, sim_i, image_norm, template_norms, correlation):
    start_template, start_spot = cuda.grid(2)
    stride_template, stride_spot = cuda.gridsize(2)
    for template in range(start_template, sim_r.shape[0], stride_template):
        for shift in range(polar_image.shape[0]):
            tmp = 0
            for spot in range(start_spot, sim_r.shape[1], stride_spot):
                tmp += (polar_image[(sim_t[template, spot]+shift)%polar_image.shape[0],
                                                    sim_r[template, spot]] * sim_i[template, spot])
            correlation[template, shift] = tmp / (image_norm * template_norms[template])
```

And ran it with
```
import cupy as cp
test_cor = cp.zeros((r_cp.shape[0], im_cp.shape[0]), dtype=np.float32)

threadsperblock = 128
blockspergrid = 50
correlate_library[blockspergrid, threadsperblock](image_polar, r_cp, t_cp, i_cp, 1., template_norms, test_cor)
```

Where the inputs are `cupy` arrays. This also took about 20 ms for a single image, which was slightly disappointing as I was hoping to shave of another order of magnitude. I'm wondering how much of this time is transfer time from GPU -> CPU, as I was only able to time it if I sent the result back. I experimented a bit with blocksize and threadsperblock but this didn't seem to have any major impact unless I made the blockspergrid too small.

One other thing I experimented with was changing the number of templates in the library to 25000. In this case the parallelized CPU implementation took 138 ms, with the GPU and a large number of blocks I was able to get this down to 78 ms. So it's really when the arrays start getting very large that the GPU shows its relative advantage over CPU. This means the GPU should be processing multiple images simultaneously to make it worth it.

Some remaining questions:

* can the cuda kernel be made more efficient using less naive approaches using e.g cached memory as in http://numba.pydata.org/numba-doc/dev/cuda/examples.html#matrix-multiplication ? How would one go about doing this?
* if the GPU is only partially occupied with a task, can it receive additional work from e.g. dask to process multiple images at once? This can be tested by running the indexation on the entire dataset.

hakonanes, 2021-03-07:
I know little about getting the most out of computing resources, but in the dictionary indexing implementation in EMsoft, which finds the best matching simulated patterns to experimental patterns via a normalized dot product, the only thing the GPU does is to compute the dot products for large-ish arrays of typically 1024 (multiples of 16) simulated and experimental patterns. (Part of the thinking behind the Fortran implementation in EMsoft is explained [here](https://github.com/EMsoft-org/EMsoft/blob/7f6b7a9d1ca0680462dc0e7351fc92084f2ac6cb/Source/EMOpenCLLib/Indexingmod.f90#L891)).

Python is of course not Fortran, we don't have the same control here. And it seems to me that what you're doing in your `tmp += (polar_image...` computation is just the bare comparisons, right? I don't have any suggestions for possible improvements, other than perhaps computing the polar image shape only once, unless it can change during the computation?

din14970, 2021-03-07:
> polar image shape only once, unless it can change during the computation?

The conversion to polar coordinates, both for the images and templates, happens only once prior to indexation on CPU. At this point this doesn't take too long, but if this becomes a bottleneck it might become interesting to also do the image polar conversion on GPU.

> And it seems to me that what you're doing in your tmp += (polar_image... computation is just the bare comparisons, right?

Yes this is actually kind of like calculating the dot product but without having all the values in memory. Basically I am extracting the pixel corresponding to the (r, theta) coordinate of a diffraction spot, and multiplying it by the associated spot intensity in the template. I sum over all the spots in the template (innermost loop) (= dot product) and repeat for all possible in-plane angles (shift, second nested loop). This must be repeated for all templates (outermost loop). Then, one would have to do this again for the mirrored templates.

Since everything is happening on the gpu, the slow-ness must come from repeated access to shared memory on the gpu. I'm wondering whether I could make use of cached memory, which can significantly speed up calculations, at least for matrix multiplication this is the case.

> Python is of course not Fortran, we don't have the same control here.

I must admit I don't know any Fortran, but I feel numba can really get you quite close!

din14970, 2021-03-07:
Alright a minor fix to the "naive" approach turned out quite effective, something I can live with:

```python
@cuda.jit
def correlate_library_gpu(polar_image, sim_r, sim_t, sim_i, image_norm, template_norms, correlation):
    start_template, start_shift = cuda.grid(2)
    stride_template, stride_shift = cuda.gridsize(2)
    if start_template >= sim_r.shape[0] or start_shift >= polar_image.shape[0]:
        return
    for template in range(start_template, sim_r.shape[0], stride_template):
        for shift in range(start_shift, polar_image.shape[0], stride_shift):
            tmp = 0
            for spot in range(sim_r.shape[1]):
                tmp += (polar_image[(sim_t[template, spot]+shift)%polar_image.shape[0],
                                    sim_r[template, spot]] * sim_i[template, spot] )
            correlation[template, shift] = tmp/(image_norm * template_norms[template])
```
Calling the function with a bit more attention to the grids/blocks/threads

```python
test_cor = cp.zeros((r_cp.shape[0], im_cp.shape[0]), dtype=np.float32)

threadsperblock = (16, 16)
blockspergrid_x = math.ceil(test_cor.shape[0] / threadsperblock[0])
blockspergrid_y = math.ceil(test_cor.shape[1] / threadsperblock[1])
blockspergrid = (blockspergrid_x, blockspergrid_y)
correlate_library_gpu[blockspergrid, threadsperblock](image_polar, r_cp, t_cp, i_cp, 1., template_norms, test_cor)
```

For a library with 1081 templates this takes 1.6 ms per pattern, for an entire 4D-STEM dataset this would take 1.3 minutes (+ overhead from transfering data / converting to polar coordinates etc. ). While I'm pretty sure there are still optizations to be made by being smarter with the GPU memory, this is already quite a satisfying result and I might leave it at that. ASTAR typically takes 15-20 min to index an 8-bit dataset, here we should be able to do an order of magnitude better on a 16-bit dataset. At these speeds, the polar transform becomes the bottleneck.

din14970, 2021-03-07:
Also note that ASTAR does not use a "full" indexation; it selects part of the templates based on the match between radially integrated templates. This is also how I have implemented it in Pyxem. Typically it only does a full indexation on 100 templates (10%). For a full indexation using all templates I was told this could take multiple hours. Using the GPU, which I'm not sure ASTAR utilizes, we can do orders of 2 orders of magnitude better. That's pretty exciting for me.

din14970, 2021-03-08:
The workflow should now be corrected on the CPU with mirrored templates, though I haven't tested it yet over an entire dataset. Once this works, a CUDA enabled implementation should follow, so that the indexation could also be run on a GPU.

As it turns out, the negative euler angle patterns correspond to another zone than I expected: the one mirrored across the z-x plane:
[image]
I verified that indeed mirrored patterns on both sides are mirrored and give a correct in-plane angle. For a random one (red dots in the image above). For the best fit solution:
[image]

The mirrored patterns may be an issue for non-cubic systems. It may well be that in the case of the hexagonal system, the same patterns are twice represented this way due to the choice of fundamental zone. 
[image]
The same issue may arise for the mono clinic case. A wiser choice of fundamental zones is something that may need to be looked at.

din14970, 2021-03-14:
The polar transform was starting to become a bottleneck and would be a bottleneck if the diffraction pattern images would be large. 256x256 diffraction patterns could be converted to polar coordinates with `skimage.transform.warp_polar` in 5 ms but 4000x4000 images required 180 ms. Since indexation on the GPU takes only 1 ms per pattern, the polar transform now takes longer. I re-implemented `warp_polar` for the GPU using `cupy`:

```
import cupy as cp
import cupyx.scipy.ndimage as ndigpu

def warp_polar_gpu(image, center=None, radius=None, output_shape=None, **kwargs):
    """
    Function to emulate warp_polar in skimage.transform on the GPU. Not all
    parameters are supported
    
    Parameters
    ----------
    image: cupy.ndarray
        Input image. Only 2-D arrays are accepted.         
    center: tuple (row, col), optional
        Point in image that represents the center of the transformation
        (i.e., the origin in cartesian space). Values can be of type float.
        If no value is given, the center is assumed to be the center point of the image.
    radius: float, optional
        Radius of the circle that bounds the area to be transformed.
    output_shape: tuple (row, col), optional

    Returns
    -------
    polar: cupy.ndarray
        polar image
    """
    if radius is None:
        radius = int(np.ceil(np.sqrt((image.shape[0] / 2)**2 + (image.shape[1] / 2)**2)))
    cx, cy = image.shape[1] // 2, image.shape[0] // 2
    if center is not None:
        cx, cy = center
    if output_shape is None:
        output_shape = (360, radius)
    delta_theta = 360 / output_shape[0]
    delta_r = radius / output_shape[1]
    t = cp.arange(output_shape[0])
    r = cp.arange(output_shape[1])
    R, T = np.meshgrid(r, t)
    X = R * delta_r * cp.cos(cp.deg2rad(T * delta_theta)) + cx
    Y = R * delta_r * cp.sin(cp.deg2rad(T * delta_theta)) + cy
    coordinates = cp.stack([Y, X])
    polar = ndigpu.map_coordinates(image, coordinates, order=1)
    return polar
```

Not counting the transfer time from CPU to GPU, the polar transform now basically happens in O(1) time. On a Tesla V100, both a 256x256 image and 4000x4000 image were converted to polar coordinates in 400 microseconds.

hakonanes, 2022-02-01:
What would you say the status is of this issue, @din14970?

din14970, 2022-02-01:
Would say this is done with the mentioned PRs and the latest #796

closed 2022-02-01 by din14970