pyxem/pyxem #747: Small bug in 0.13.1 peak finding

tiarnan-doherty, 2021-03-24:
**Describe the bug**
I think there's a minor naming bug in peak finding in pyxem 0.13.1. 

Running the line: 
peaks = dp.find_peaks(method='difference_of_gaussian',
                      min_sigma=0.001,
                      max_sigma=5.0,
                      sigma_ratio=1.6,
                      threshold=0.01,
                      overlap=0.5)

In pyxem 0.13.1 on both windows and Mac produces the screen shotted error

**To Reproduce**
Steps to reproduce the behavior:
1. Run the above code in pyxem 0.13.1

**Expected behavior**
Peaks found using the difference of gaussians method

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

pc494, 2021-03-24:
Yup, this is a bug in hyperspy (v1.6.1) that has been fixed by https://github.com/hyperspy/hyperspy/pull/2647

Running the code you have with the further argument `interactive=False` should deal with this for now while we await a patch release from hyperspy.

tiarnan-doherty, 2021-03-24:
Thanks Phillip, that's working now

closed 2021-03-24 by tiarnan-doherty