pyxem/pyxem #748: mib_to_h5stack in 0.13.1 does not support single-chip detectors.

emichr, 2021-04-08:
**Describe the bug**
Converting .mib data from single-chip detectors using `utils.io_utils.mib_to_h5stack()` as shown in part 2 of demo0 raises an UnboundLocalError. The error is raised in `io_utils._untangle_raw()`, as the return variable is not defined properly. Looking into the code, it seems that `utils.io_utils.mib_to_h5stack()` only supports quad detectors.

**To Reproduce**
Steps to reproduce the behavior:

Step 1: Download the data from: [https://studntnu-my.sharepoint.com/:u:/g/personal/emilc_ntnu_no/EUKu1WmcJ-xJgwpbLXnxA_IBjcirQwpIqII8lsaHlWusLw?e=CmdznR](url)

Step 2: Run the following code
```python
import pyxem as pxm
data_path = r'frame.mib'
h5_path = r'frame.h5'

pxm.utils.io_utils.mib_to_h5stack(data_path, h5_path)
dp = pxm.utils.io_utils.h5stack_to_pxm(h5_path, data_path)
```
The above code raises the screenshotted image in pyxem 0.13.1.

**Expected behavior**
mib_to_h5stack should be expected to convert single-chip data the same way as quad data, and the converted file should be readable through h5stack_to_pxm.

**Screenshots**
[image]

**Additional context**
The code was run in a fresh conda environment with only pyxem and its requirements installed.

emichr, 2021-04-08:
Actually working link to data: [https://studntnu-my.sharepoint.com/:u:/g/personal/emilc_ntnu_no/EUKu1WmcJ-xJgwpbLXnxA_IBjcirQwpIqII8lsaHlWusLw?e=CmdznR](https://studntnu-my.sharepoint.com/:u:/g/personal/emilc_ntnu_no/EUKu1WmcJ-xJgwpbLXnxA_IBjcirQwpIqII8lsaHlWusLw?e=CmdznR)

pc494, 2024-01-26:
io is now handled is rosettaio

closed 2024-01-26 by pc494