pyxem/pyxem #749: Loading issues with non-square scans (.emi files)

AmoghMG, 2021-04-08:
Thank you for making this useful package! This is the first time I'm reporting an issue - so please bear with me.

I am having issues loading 'non-square scans' - datasets such as the one shown in the attached Fig [a]. Ideally, a plot in the navigation space for a rectangular ROI scan should look like Fig [b] as seen in the TIA software, but instead looks like the one in Fig [c] with the pixels in the navigation axes jumbled up. 

Could you please help with this? 
[image]

pc494, 2021-04-08:
This may well actually be related to hyperpsy code (in which case I will raise a separate issue) in the meantime either of the reshape commands described above should work :)

pc494, 2023-01-19:
closed as both out of scope and (for now) solved.

closed 2023-01-19 by pc494