pyxem/pyxem #754: HyperSpy release 1.6.2, lazy s.map: minor issue

magnunor, 2021-04-12:
**Short version:** one of pyxem's tests will fail when using HyperSpy 1.6.2.

**Long version:**
Recently, the lazy version of `s.map` in HyperSpy was improved. These changes will be included in the 1.6.2 release, which should be release soon.

These changes seem to break some functionality in pyXem, specifically the test: `pyxem/tests/signals/test_pixelated_stem_class.py::TestDiffraction2DShiftDiffraction::test_lazy`. This does not extend to other functions in pyXem, since most of them utilize the non-lazy version of `s.map`.

I'm not 100% sure of the underlying issue yet, but I'm in the process implementing so that the lazy version of `s.map` is used for non-lazy signals as well (https://github.com/hyperspy/hyperspy/pull/2703). And I see exactly the same bug there, but extending to many more of pyxem's unit tests.

pc494, 2021-04-12:
The plan had been for us to make a new patch release for `hyperspy=1.6.2` soon as possible. I would hope to then sweep most of these issues up while doing that. In the meantime, pyxem release are hard attached to hyperspy versions, so when 1.6.2 ships it won't break a fresh pyxem 0.13.1 install

pc494, 2021-04-14:
This was patched in #762

closed 2021-04-14 by pc494