pyxem/pyxem #759: Switch to new, improved, version of HyperSpy map function

magnunor, 2021-04-13:
Currently, there is a mix of different ways of doing essentially the same thing: iterating over signals over the navigation dimensions, and doing some kind of processing on the signal dimension.

Recently, HyperSpy's `s.map` function got a big improvement for lazy signals (https://github.com/hyperspy/hyperspy/pull/2617), and I'm currently working on a pull request to extend this to the non-lazy signals as well (https://github.com/hyperspy/hyperspy/pull/2703).

When https://github.com/hyperspy/hyperspy/pull/2703 is complete, a number of pyxem's methods can be switched over to using `s.map`, rather than `_process_dask_array` and other customized functions doing the same thing.

hakonanes, 2021-04-13:
@magnunor, great! Looking forward to see how this goes, and do this switch in kikuchipy as well for some methods.

pc494, 2021-04-13:
Yes. This is great news and when that code is released it should be top priority.

pc494, 2021-10-14:
Patch release, yes, happy to be on the ball with this one to keep things smooth. As far as I can see the `0.13.x` branch is good to go.

magnunor, 2024-03-15:
This was implemented in pyxem 0.15

closed 2024-03-15 by magnunor