pyxem/pyxem #765: Next patch and minor releases

pc494, 2021-04-15:
After shipping 0.13.2 I'm starting to think about future releases. I think we have at least two more patch releases worth of code in the pipeline and am seeking thoughts on how much minor release code people have unwritten ready to go.

If the answer is "not very much" I think I will target the start of the academic year 2021-22 for 0.14 as the number of breaking changes is small (and we can filter in a lot of deprecations over the upcoming patch releases)

0.13.3
--------
- Future-proofing for skimage #764 
- further numpy 1.20 compliance #757 
- orientation mapping bugfixes (#752 and #753)

0.13.4
-------- 
This release will focus on making use of the new hyperspy map functionality, and thus will mainly involve those changes (#759) and deprecations (#758)

ericpre, 2021-04-24:
It looks like that some changes included in the 0.13.2 release [are not in the 0.13.x branch](https://github.com/pyxem/pyxem/compare/0.13.x...v0.13.2). Is it expected? Was the 0.13.2 not tagged from the 0.13.x branch?

pc494, 2021-04-25:
This was a set of mistakes by me. When I merged the` 0.13.1` changes into `master` I accidentality brought the master changes into the patch branch. To turn things around quickly for 0.13.2 I made a new branch (`0.13.2-branch`) and tagged from that.

If fixing 0.13.x is simple we should do that, but if it's not making each new patch branch from the old patch tagged release was pretty easy

ericpre, 2021-04-25:
If I understand correctly, one simple solution is to replace the current `0.13.x` by the `0.13.2-branch` and remove the latter. The easiest is possibly to delete `0.13.x` and rename `0.13.2-branch` to `0.13.x`. Then merge the new and up to date `0.13.x` can be merged into `master`.

pc494, 2021-04-25:
The renaming worked a charm, but I still think resolving the conflicts means that `0.13.x` now has `master` changes in it (this might be to with breaking vs non-breaking changes between releases?

(see #767)

ericpre, 2021-04-25:
Indeed, the last commit (https://github.com/pyxem/pyxem/commit/187a4a2cc372e26c4ee80d05343fc065a4815baa) of `0.13.x` is not necessary and it is what is wrong. Can you reset to commit https://github.com/pyxem/pyxem/commit/e7d45144c9f3625859dd15ef0bc2bc88c2da5e63 and then merge `0.13.x` into `master` instead of doing `master` into `0.13.x`?

ericpre, 2021-04-25:
In case, this helps to clarify, I have done it in my fork: https://github.com/ericpre/pyxem/commits/master and https://github.com/ericpre/pyxem/commits/0.13.x

pc494, 2021-04-25:
> Indeed, the last commit ([187a4a2](https://github.com/pyxem/pyxem/commit/187a4a2cc372e26c4ee80d05343fc065a4815baa)) of `0.13.x` is not necessary and it is what is wrong. Can you reset to commit [e7d4514](https://github.com/pyxem/pyxem/commit/e7d45144c9f3625859dd15ef0bc2bc88c2da5e63) and then merge `0.13.x` into `master` instead of doing `master` into `0.13.x`?

Yes, that makes more sense, will fix that up

ericpre, 2021-05-26:
I would have expected the bug fixes in `0.13.x` (or at least the changes in the `0.13.2` release) to be present in the `master` branch, but this is not the case: https://github.com/pyxem/pyxem/compare/master...0.13.x

How are the `master` and `0.13.x` branch managed?

pc494, 2021-05-27:
This is in an error, I've got a bit of time today so I should be able to fix it.

pc494, 2021-07-02:
I am now planning to make one final patch release (to include #775 and #771) and then move all development to the 0.14 branch

pc494, 2022-01-06:
Closing in favour of #788

closed 2022-01-06 by pc494