pyxem/pyxem #770: Demo07 incompatible with pyxem0.13: get_azimuthal_integral

zhang-siyuan, 2021-05-18:
Following the jupyter notebook until:
integration2d = dp.get_azimuthal_integral2d(npt_rad=100, unit="pyxem")
1. npt_rad is now npt
2. after the easy fix
integration2d = dp.get_azimuthal_integral2d(npt=100, unit="pyxem")
would lead to: 
ValueError: ai property is not currently set

Same problem for get_azimuthal_integral1d

Suspect it has something to do with the lazy input, but the problem persists after changing
dp.unit="2th_deg"
or giving wavelength=2.5e-12 for dp.unit = "k_A^-1" (the unit in Demo7)

CSSFrancis, 2021-05-19:
@zhang-siyuan thanks for bringing up this issue!  I think the problem is just that the updated notebook never got merged. 

You can use the notebook here https://github.com/pyxem/pyxem-demos/pull/63 in the meantime and I'll try to get it merged.

@pc494 any reason the demo can't be merged? I can look at it again tomorrow to make sure it still works.

pc494, 2021-05-19:
> @pc494 any reason the demo can't be merged? I can look at it again tomorrow to make sure it still works.

Nothing more than my own inefficiency sadly. I had hoped to tidy up the syntax a little, but given that I haven't if you do go over it again I'll merge.