pyxem/pyxem #783: Error message upon using plot_diffraction_vectors_on_signal

w-markus, 2021-11-24:
Dear all,

I'd like to use pyxem to work on some diffraction patterns.
I have a list of patterns from dm3-files which I load via ``` hs.load ....```

```
import hyperspy.api as hs
# some other imports, e.g. numpy, matplotlib, ...

# -------------------------------------
# code for loading dm3 files
# ....
# ------------------------------------

dp = hs.stack(dp_list)
peaks = dp.find_peaks(method='difference_of_gaussian',
                       min_sigma=0.5,
                       max_sigma=16.,
                       sigma_ratio=1.3,
                       threshold=0.02,
                       overlap=0.99,
                       interactive=False)

peaks = DiffractionVectors.from_peaks(peaks, center=(1023, 1023), calibration=1.0)
peaks.plot_diffraction_vectors_on_signal(dp)
```

I receive:
```
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
/tmp/ipykernel_2050251/2929976898.py in <module>
      9 
     10 peaks = DiffractionVectors.from_peaks(peaks, center=(1023, 1023), calibration=1.0)
---> 11 peaks.plot_diffraction_vectors_on_signal(dp)

~/Software/conda/miniconda/miniconda3/envs/denoising/lib/python3.8/site-packages/pyxem/signals/diffraction_vectors.py in plot_diffraction_vectors_on_signal(self, signal, *args, **kwargs)
    329             Keyword arguments passed to signal.plot()
    330         """
--> 331         mmx, mmy = generate_marker_inputs_from_peaks(self)
    332         signal.plot(*args, **kwargs)
    333         for mx, my in zip(mmx, mmy):

~/Software/conda/miniconda/miniconda3/envs/denoising/lib/python3.8/site-packages/pyxem/signals/diffraction_vectors.py in generate_marker_inputs_from_peaks(peaks)
    104 
    105     """
--> 106     max_peak_len = _find_max_length_peaks(peaks)
    107     pad = np.array(
    108         list(

~/Software/conda/miniconda/miniconda3/envs/denoising/lib/python3.8/site-packages/pyxem/signals/diffraction_vectors.py in _find_max_length_peaks(peaks)
     74     x_size, y_size = (
     75         peaks.axes_manager.navigation_shape[0],
---> 76         peaks.axes_manager.navigation_shape[1],
     77     )
     78     length_of_longest_peaks_list = 0

IndexError: tuple index out of range
```

No idea, what went wrong. Any hint is appreciated.

Best wishes and thank you very much for sharing pyxem!!
Markus

din14970, 2021-11-24:
it would seem that the axis manager of your signal does not recognize the fact that the scan should be two dimensional. Could you share the output of `dp.axes_manager`? My naive guess would be that the file was loaded incorrectly with hyperspy, see the issues related to DM3 http://hyperspy.org/hyperspy-doc/current/user_guide/io.html#gatan-digital-micrograph. Can you plot the data as-is with hyperspy?
If this really is a one dimensional line profile my guess is indeed that this is a bug here and a small modification needs to happen there to account for 1D navigation space.

w-markus, 2021-11-24:
Hi Niels,

ok, probably my use case is different, or not as the major use case of pyxem(?). I am working on single data files each containing a single diffraction pattern. No 4D-STEM data, just parallel illumination (but in some sense it's a scan, though: I moved the stage in between exposures).

I can plot the data, and "diffraction"ness is understood by specify the signal_type as "diffraction_experiment" as indicated by axes (reciprocal x and y, i.e. it's 2D data) units in 1/nm and also the camera length is correctly imported into the meta data.

Best wishes from Berlin!
Markus

din14970, 2021-11-24:
Ok I follow, yes so effectively you have a stack of diffraction patterns. This is equivalent to a line scan in terms of data representation. So in fact it should work and is an oversight here. If I find some time later this week or next week I might take a look.

din14970, 2021-11-27:
@w-markus Hi Markus,

I've implemented a small fix in #784. It should now work for your stack of images. Pyxem was initially created to treat 4D-STEM data, so the assumption of 4D data is often hard coded and 1D stacks of diffraction patterns may not have always been considered. Still I encourage you to give it a try and if things don't work as expected report it as a bug - unfortunately you may encounter many more issues like this. Then we can look at a way to generalize the functionality.

It may take a while to get this fix into a new release of Pyxem. If you want to quickly get this yourself you can clone the repo and install the development version and pull changes from my fork. If you are not so comfortable with git, a quick and dirty "fix" would be manually editing the file on your system `~/Software/conda/miniconda/miniconda3/envs/denoising/lib/python3.8/site-packages/pyxem/signals/diffraction_vectors.py`. Remove everything and replace it with the contents of the file here: https://github.com/din14970/pyxem/blob/fix_2d_assumption/pyxem/signals/diffraction_vectors.py.

Best, Niels

w-markus, 2021-11-30:
Thanks a lot Niels!

I wanted to take advantage of the build in functionalities of hyperspy/pyxem with respect to metadata handling and since I am aiming at streamlining scripts. However, colleagues are waiting for the results, therefore I have to go back to "quick-and-dirty". 

It's a pity. Sadly, I encountered similar situations with hyperspy which for me often ended in using hs only for it's dm3-loading capabilities. In the struggle to go to some unified metadata-handling this is a real obstacle (and off-topic here).

Best wishes
Markus

din14970, 2021-11-30:
Hi Markus,

There are many related problems here. Users should understand that open source software is a gift, not a contract or obligation. Hyperspy and Pyxem are used by probably hundreds of researchers, but there are really only 3-5 active developers working on these packages at any one moment mainly volunteering their free time. Bugs are bound to happen and manpower is just too limited to address them all quickly. There are too few developers in our space, and since software development gives you basically zero academic respect points there's limited incentive to participate unless you really need a feature your use case. Software development is rarely recognized as a proper activity for scientists, and hence most people who've made great contributions to these packages have moved on from academia. During my post-doc I also spent a lot of time developing in this ecosystem, which is time spent not writing papers. I've now also moved on to software industry as I did not see long term perspectives in academia for myself. The colleagues who do "quick-and-dirty" are actually the smart ones and properly aligned with the incentives, as it allows them to crank out output more quickly. The system is hard to change at present but I think improvements can be achieved if:

* more users of software invest the time and become co-developers. More eyes, brains and typing hands are what makes free software great.
* users cite the free software they use in their work. Visibility for these projects is key for making them sustainable.

So thanks for your issue and I hope you continue to make them. And if you are up for it we would also really welcome code contributions! In the end it is random volunteers like us who have to do it.

Best, Niels

magnunor, 2021-11-30:
To comment on the diffraction vector functionality in pyxem.

There are plans to make this a bit more "streamlined", as this type of analysis is a major focus for us nowadays. For example, the `ragged` Signal functionality recently added in HyperSpy (https://github.com/hyperspy/hyperspy/pull/2842) will help with this.

However, as @din14970 mentioned, there is a limited amount of developer time available. And my own dev-time is currently bound up in "backend" work in HyperSpy, which will be heavily used in pyxem for "streamlining" the functionality.

w-markus, 2021-11-30:
no need for justifications! :-)

My comment was more a complaining about the dilemma I feel and which Niels described in more detail subsequently... No criticism at all.

pc494, 2021-11-30:
Thanks for the fix Niels, I'll get that reviewed when time allows. As the person who wrote the original code I can confirm the description outlined above, this code was very much a case of "make something work for 4D, worry about other stuff as and when people raise bug reports"

closed 2021-12-01 by pc494